Seagate/openSeaChest #1: Suggestion to provide a bit of documentation on building

edrozenberg, 2018-01-11:
The Building section of the docs is blank - would be great to see a little bit of information there. Maybe it's as simple as "make; make install" but these utilities are low level and important and your users would appreciate making sure to do the right thing in compiling them. Thanks.

rudock1, 2018-01-26:
thanks @edrozenberg,  I appreciate the reminder.  We have added in the missing details now to this project and the associated library projects.  They do follow the typical instructions.  If you could check it out and let me know if it can still be improved that would be most welcome.  Thanks again!

elifarley, 2018-06-18:
Hi, what should I do in order to obtain those 3 directories:
```
opensea-common @ e89db1a	
opensea-operations @ d0d05e3
opensea-transport @ 53f5f00
```
Here's the error I get when I try to compile:

```bash
pwd
/tmp/openSeaChest/Make/gcc

make release
mkdir -p openseachest_exes
gcc -Wall -c -std=gnu99 -O3 -I../../opensea-common/include -I../../opensea-transport/include -I../../include -I../../opensea-operations/include -DDISABLE_TCG_SUPPORT ../../utils/C/openSeaChest//openSeaChest_Erase.c -o ../../utils/C/openSeaChest//openSeaChest_Erase.o
../../utils/C/openSeaChest//openSeaChest_Erase.c:27:10: fatal error: common.h: No such file or directory
 #include "common.h"
          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:231: ../../utils/C/openSeaChest//openSeaChest_Erase.o] Error 1
```

xahmad, 2018-06-18:
Hi @elifarley 

You must be cloning the repo without getting any of the submodules. 

Depending on the version of git you are using, you may need to add --recursive or --recurse-submodules to your clone command line. 

To get more information, just search 'git cloning with submodules' in your favorite search engine.

CreationTribe, 2020-08-07:
> Hi @elifarley
>
> You must be cloning the repo without getting any of the submodules.
>
> Depending on the version of git you are using, you may need to add --recursive or --recurse-submodules to your [...]

I'm having the same issue @elifarley is having; I'm giving your recommendation a shot right now ...
Ok, cool. It appears to be working ... and it finished without errors. Cool. Thanks.

Even though it explicitly mentions cloning for submodules in the README; there may be others like myself who completely missed that part and are trying to figure things out - so if you are one such person, here you go:

This is obviously the prefect reference: https://git-scm.com/book/en/v2/Git-Tools-Submodules#_cloning_submodules

> To also initialize, fetch and checkout any nested submodules, you can use the foolproof `git submodule update --init --recursive`

This is pretty great too: https://stackoverflow.com/questions/3796927/how-to-git-clone-including-submodules

> With version 2.13 of Git and later, --recurse-submodules can be used instead of --recursive:

> ```
> git clone --recurse-submodules -j8 git://github.com/foo/bar.git
> cd bar
> ```

> Editor’s note: -j8 is an optional performance optimization that became available in version 2.8, and fetches up to 8 submodules at a time in parallel — see man git-clone.
> 
> With version 1.9 of Git up until version 2.12 (-j flag only available in version 2.8+):
> 

> ```
> git clone --recursive -j8 git://github.com/foo/bar.git
> cd bar
> ```

> With version 1.6.5 of Git and later, you can use:
> 

> ```
> git clone --recursive git://github.com/foo/bar.git
> cd bar
> ```

> For already cloned repos, or older Git versions, use:
> 

> ```
> git clone git://github.com/foo/bar.git
> cd bar
> git submodule update --init --recursive
> ```

vonericsen, 2021-03-22:
I have consolidated all the build documentation into [BUILDING.md](https://github.com/Seagate/openSeaChest/blob/develop/BUILDING.md).

If you have any issues or need clarification, please do open a new issue and we'll take care of it.

closed 2021-03-22 by vonericsen