Seagate/openSeaChest #2: autotools or cmake support

sledz, 2018-01-29:
Especially for a "cross platform utility" support for cross compiling is an important issue. Pure makefiles aren't a good way to handle this.

Therefor this project should get support for GNU Autotools or CMake!

xahmad, 2018-01-31:
@sledz Issue duly noted. The support for autotools or cmake is on our roadmap. We will add your +vote to it & see if we can schedule it any sooner.

In the mean time, if you are not compiling this code on the architecture of choice (ARM, MIPS etc.) you can cross compile using the following make command:

make CC=/usr/bin/arm-linux-gnueabi-gcc AR=/usr/bin/arm-linux-gnueabi-ar STRIP=/usr/bin/arm-linux-gnueabi-strip release

You can replace the arm-linux-gnueabi-gcc with the toolchain that you need.

I updated the Makefile to use the correct archiver, so please pull/clone the latest develop & the above make command will work. 

Let me know if this solution works for you.

vonericsen, 2022-04-14:
I am closing this as we have accepted all the work from multiple pull requests for the meson build system which works very well and seems to be a good build system that satisfies the intent of this request. It works on Windows, Linux, FreeBSD, and Illumos.
Please reopen this for discussion if there is a need for another build system, but meson seems like a great solution to adopt and continue with at this time.

closed 2022-04-14 by vonericsen