Seagate/openSeaChest #3: "openSeaChest_Info --scan" fails with "Bad address"

sledz, 2018-02-02:
I've cross compiled the latest HEAD for use in an embedded linux device.

If I try to call the binaries there's always a "Bad address" error reported.

    # openSeaChest_Info --scan
    ==========================================================================================
     openSeaChest_Info - openSeaChest drive utilities
     Copyright (c) 2014-2017 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
     openSeaChest_Info Version: 1.3.0-1_17_8 ARM
     Build Date: Feb  1 2018
     Today: Fri Feb  2 10:54:53 2018
    ==========================================================================================
    open: Bad address
    open failure
    Error: 14 - Bad address

Here's the output of the related smartctl info

    # smartctl -i /dev/sda
    smartctl 6.5 2016-05-07 r4318 [armv7l-linux-4.1.15-2.0.0-ga+yocto+g719fb24] (local build)
    Copyright (C) 2002-16, Bruce Allen, Christian Franke, www.smartmontools.org

    === START OF INFORMATION SECTION ===
    Device Model:     ST2000LV000-2G2174
    Serial Number:    WDZ2WMSV
    LU WWN Device Id: 5 000c50 0a896080e
    Firmware Version: ZZZZ
    User Capacity:    2,000,398,934,016 bytes [2.00 TB]
    Sector Sizes:     512 bytes logical, 4096 bytes physical
    Rotation Rate:    5400 rpm
    Form Factor:      2.5 inches
    Device is:        Not in smartctl database [for details use: -P showall]
    ATA Version is:   ACS-3 T13/2161-D revision 3b
    SATA Version is:  SATA 3.1, 6.0 Gb/s (current: 3.0 Gb/s)
    Local Time is:    Fri Feb  2 10:54:46 2018 UTC
    SMART support is: Available - device has SMART capability.
    SMART support is: Enabled

xahmad, 2018-02-02:
Do the following two commands work? 

`openSeaChest_Info -i -d /dev/sg0
`and
`openSeaChest_Info -i -d /dev/sda`

sledz, 2018-02-02:
There is not /dev/sg0. If I create one with

    mknod /dev/sg0 c 21 0

I get:

    # openSeaChest_Info -i -d /dev/sg0
    ==========================================================================================
     openSeaChest_Info - openSeaChest drive utilities
     Copyright (c) 2014-2017 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
     openSeaChest_Info Version: 1.3.0-1_17_8 ARM
     Build Date: Feb  1 2018
     Today: Fri Feb  2 12:55:05 2018
    ==========================================================================================
    open: No such device or address
    open failure
    Error: 6 - No such device or address
    Error: Could not open handle to /dev/sg0

Using /dev/sda leads to

    # openSeaChest_Info -i -d /dev/sda
    ==========================================================================================
     openSeaChest_Info - openSeaChest drive utilities
     Copyright (c) 2014-2017 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
     openSeaChest_Info Version: 1.3.0-1_17_8 ARM
     Build Date: Feb  1 2018
     Today: Fri Feb  2 12:53:43 2018
    ==========================================================================================
    open: Bad address
    open failure
    Error: 14 - Bad address
    Error: Could not open handle to /dev/sda

xahmad, 2018-02-02:
Is your kernel compiled with the following flags:

CONFIG_CHR_DEV_SG 
or/and
CONFIG_BLK_DEV_BSG

sledz, 2018-02-02:
After enabling CONFIG_BLK_DEV_BSG it seems to work. :)

Thx for this hint!

jangrewe, 2019-02-02:
@xahmad I have the same issue, and `CONFIG_BLK_DEV_BSG` is enabled in my kernel:
```
root@backbox:~/openSeaChest/Make/gcc/openseachest_exes# ./openSeaChest_Basics -d /dev/sda -i
==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2018 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 2.8.0-1_19_0 ARM64
 Build Date: Feb  2 2019
 Today: Sat Feb  2 15:07:34 2019
==========================================================================================
open: Bad address
open failure
Error: 14 - Bad address
Error: Could not open handle to /dev/sda

```

This is with the current master branch.

```
Linux backbox 4.4.132-1075-rockchip-ayufan-ga83beded8524 #1 SMP Thu Jul 26 08:22:22 UTC 2018 aarch64 GNU/Linux
```

xahmad, 2019-02-02:
@jangrewe Usually the "Bad address" comes from permission issues or nodes not being created during boot. Since you are running as root, it has to be something to do with the device nodes. 

a couple of questions:
1. is there a dev node /dev/sg* that correspond to /dev/sda? if so what is it?

2. Is there something under /dev/bsg/*  that  corresponds to /dev/sda?

Lastly, I would recommend turning on the CONFIG_CHR_DEV_SG and see if that helps.

jangrewe, 2019-02-02:
Thanks @xahmad 

1) There are no /dev/sg* devices
2) There are `/dev/bsg/0:0:0:0` and  `/dev/bsg/1:0:0:0`

With the two devices from 2) i don't get an error, but i also don't get any real output:

```
root@backbox:~/openSeaChest/Make/gcc/openseachest_exes# ./openSeaChest_Basics -d /dev/bsg/0\:0\:0\:0 -i
==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2018 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 2.8.0-1_19_0 ARM64
 Build Date: Feb  2 2019
 Today: Sat Feb  2 16:49:00 2019
==========================================================================================

/dev/bsg/0:0:0:0 -  -  - ATA
        Model Number:
        Serial Number:
        Firmware Revision:
        World Wide Name: Not Supported
        Drive Capacity (B/B): 0.00/0.00
        Temperature Data:
                Current Temperature (C): Not Reported
                Highest Temperature (C): Not Reported
                Lowest Temperature (C): Not Reported
        Power On Time: Not Reported
        Power On Hours: Not Reported
        MaxLBA: 0
        Native MaxLBA: Not Reported
        Logical Sector Size (B): 0
        Physical Sector Size (B): 0
        Sector Alignment: 0
        Rotation Rate (RPM): Not Reported
        Form Factor (inch): Not Reported
        Last DST information:
                Not supported
        Long Drive Self Test Time: Not Supported
        Interface speed:
                Not Reported
        Annualized Workload Rate (TB/yr): Not Reported
        Total Bytes Read (B): Not Reported
        Total Bytes Written (B): Not Reported
        Encryption Support: Not Supported
        Cache Size (MiB): Not Reported
        Read Look-Ahead: Not Supported
        Write Cache: Not Supported
        SMART Status: Unknown or Not Supported
        ATA Security Information: Not Supported
        Firmware Download Support: Not Supported
        Specifications Supported:
                None reported by device.
        Features Supported:
                None reported or an error occurred while trying to determine
                the features.
```

As this is an embedded system, i don't really feel comfortable recompiling the kernel to enable `CONFIG_CHR_DEV_SG`. :-(

xahmad, 2019-02-03:
@jangrewe any chance you have [sg3_utils](http://sg.danny.cz/sg/sg3_utils.html) compiled on this system? 
If sg_sat_identify or any other sg_* utilities work on your system, we will debug it further and make sure openSeaChest works too.

jangrewe, 2019-02-04:
@xahmad Does this help?

```
root@backbox:~# sg_sat_identify /dev/sda
Response for IDENTIFY DEVICE ATA command:
 00     0c5a 3fff c837 0010 0000 0000 003f 0000     .Z ?. .7 .. .. .. .? ..
 08     0000 0000 2020 2020 2020 2020 2020 2020     .. ..
 10     5a41 3239 414c 4b36 0000 0000 0000 454e     ZA 29 AL K6 .. .. .. EN
 18     3031 2020 2020 5354 3830 3030 4e45 3030     01       ST 80 00 NE 00
 20     3034 2d31 5a46 3131 4720 2020 2020 2020     04 -1 ZF 11 G
 28     2020 2020 2020 2020 2020 2020 2020 8010                          ..
 30     4000 2f00 4000 0200 0200 0007 3fff 0010     @. /. @. .. .. .. ?. ..
 38     003f fc10 00fb 5d10 ffff 0fff 0000 0007     .? .. .. ]. .. .. .. ..
 40     0003 0078 0078 0078 0078 0008 0000 0000     .. .x .x .x .x .. .. ..
 48     0000 0000 0000 001f 8d0e 0044 08cc 0044     .. .. .. .. .. .D .. .D
 50     07f0 006d 306b 7561 6173 3069 b441 6173     .. .m 0k ua as 0i .A as
 58     407f 816b 816b 0000 fffe 0000 fe00 1000     @. .k .k .. .. .. .. ..
 60     0000 0000 2710 0000 2ab0 a381 0003 0000     .. .. '. .. *. .. .. ..
 68     0000 0000 6003 0000 5000 c500 b2fc 6240     .. .. `. .. P. .. .. b@
 70     0000 0000 0000 0000 0000 0000 0000 41de     .. .. .. .. .. .. .. A.
 78     409c 0000 0000 0000 0000 0000 0000 0000     @. .. .. .. .. .. .. ..
 80     0021 2ab0 a381 2ab0 a381 2020 0002 0140     .! *. .. *. ..    .. .@
 88     0100 5000 3c06 3c0a 0000 003c 0000 0008     .. P. <. <. .. .< .. ..
 90     0000 0000 fdff 0280 0000 0170 0008 0000     .. .. .. .. .. .p .. ..
 98     0000 0c33 0000 8008 0000 02f7 9700 8068     .. .3 .. .. .. .. .. .h
 a0     0000 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 a8     0002 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 b0     0000 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 b8     0000 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 c0     0000 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 c8     0000 0000 0000 0000 0000 0000 70bd 0000     .. .. .. .. .. .. p. ..
 d0     0000 4000 0000 0000 0000 0000 0000 0000     .. @. .. .. .. .. .. ..
 d8     0000 1c20 0000 0000 0000 0000 107f 0000     .. .  .. .. .. .. .. ..
 e0     0000 0000 0000 0000 0000 0000 2ab0 a381     .. .. .. .. .. .. *. ..
 e8     0003 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 f0     0000 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 f8     0000 0000 0000 0000 0000 0000 0000 1fa5     .. .. .. .. .. .. .. ..
root@backbox:~# sg_sat_identify /dev/sdb
Response for IDENTIFY DEVICE ATA command:
 00     0c5a 3fff c837 0010 0000 0000 003f 0000     .Z ?. .7 .. .. .. .? ..
 08     0000 0000 2020 2020 2020 2020 2020 2020     .. ..
 10     5a41 3239 4435 344c 0000 0000 0000 454e     ZA 29 D5 4L .. .. .. EN
 18     3031 2020 2020 5354 3830 3030 4e45 3030     01       ST 80 00 NE 00
 20     3034 2d31 5a46 3131 4720 2020 2020 2020     04 -1 ZF 11 G
 28     2020 2020 2020 2020 2020 2020 2020 8010                          ..
 30     4000 2f00 4000 0200 0200 0007 3fff 0010     @. /. @. .. .. .. ?. ..
 38     003f fc10 00fb 5d10 ffff 0fff 0000 0007     .? .. .. ]. .. .. .. ..
 40     0003 0078 0078 0078 0078 0008 0000 0000     .. .x .x .x .x .. .. ..
 48     0000 0000 0000 001f 8d0e 0044 08cc 0044     .. .. .. .. .. .D .. .D
 50     07f0 006d 306b 7561 6173 3069 b441 6173     .. .m 0k ua as 0i .A as
 58     407f 816e 816e 0000 fffe 0000 fe00 1000     @. .n .n .. .. .. .. ..
 60     0000 0000 2710 0000 2ab0 a381 0003 0000     .. .. '. .. *. .. .. ..
 68     0000 0000 6003 0000 5000 c500 b316 b2aa     .. .. `. .. P. .. .. ..
 70     0000 0000 0000 0000 0000 0000 0000 41de     .. .. .. .. .. .. .. A.
 78     409c 0000 0000 0000 0000 0000 0000 0000     @. .. .. .. .. .. .. ..
 80     0021 2ab0 a381 2ab0 a381 2020 0002 0140     .! *. .. *. ..    .. .@
 88     0100 5000 3c06 3c0a 0000 003c 0000 0008     .. P. <. <. .. .< .. ..
 90     0000 0000 fdff 0280 0000 0170 0008 0000     .. .. .. .. .. .p .. ..
 98     0000 0c33 0000 8008 0000 02f7 9700 8068     .. .3 .. .. .. .. .. .h
 a0     0000 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 a8     0002 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 b0     0000 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 b8     0000 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 c0     0000 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 c8     0000 0000 0000 0000 0000 0000 70bd 0000     .. .. .. .. .. .. p. ..
 d0     0000 4000 0000 0000 0000 0000 0000 0000     .. @. .. .. .. .. .. ..
 d8     0000 1c20 0000 0000 0000 0000 107f 0000     .. .  .. .. .. .. .. ..
 e0     0000 0000 0000 0000 0000 0000 2ab0 a381     .. .. .. .. .. .. *. ..
 e8     0003 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 f0     0000 0000 0000 0000 0000 0000 0000 0000     .. .. .. .. .. .. .. ..
 f8     0000 0000 0000 0000 0000 0000 0000 59a5     .. .. .. .. .. .. .. Y.
```

xahmad, 2019-02-05:
@jangrewe yes, thank you for this. 

We will have to setup a system with a similar kernel config but I will add it to our work queue and reopen this issue.

xahmad, 2019-02-22:
@jangrewe I just pushed some changes that might fix your issue. 
https://github.com/Seagate/opensea-transport/commit/8be2c0fc7400e4010b0da66988672e7675e8526a

Would you mind pulling the latest openSeaChest  & see if it works? Make sure your pull updates the opensea-transport submodule. 

If it doesn't work, please build the debug target (i.e. make debug) and run the _dbg binary. It will spew some debug information that you can copy/paste here so I get a better idea on what is going on.

jangrewe, 2019-02-23:
Success! Thank you 👍 

```
root@backbox:~/openSeaChest/Make/gcc# ./openseachest_exes/openSeaChest_Basics -d /dev/sda -i
==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2018 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 2.8.0-1_19_0 ARM64
 Build Date: Feb 19 2019
 Today: Sat Feb 23 23:03:31 2019
==========================================================================================

/dev/sda - 0NE0004-1ZF11G - LK6 - ATA
        Model Number: ST8000NE0004-1ZF11G
        Serial Number: ZA29ALK6
        Firmware Revision: EN01
        World Wide Name: 5000C500B2FC6240
        Drive Capacity (TB/TiB): 8.00/7.28
        Native Drive Capacity (TB/TiB): 8.00/7.28
        Temperature Data:
                Current Temperature (C): 43
                Highest Temperature (C): 54
                Lowest Temperature (C): 0
        Power On Time:  22 days 10 hours 17 minutes
        Power On Hours: 538.28
        MaxLBA: 15628053167
        Native MaxLBA: 15628053167
        Logical Sector Size (B): 512
        Physical Sector Size (B): 4096
        Sector Alignment: 0
        Rotation Rate (RPM): 7200
        Form Factor (inch): 3.5
        Last DST information:
                Time since last DST (hours): 24.28
                DST Status/Result: 0x0
                DST Test run: 0x1
        Long Drive Self Test Time:  12 hours 4 minutes
        Interface speed:
                Max Speed (Gb/s): 6.0
                Negotiated Speed (Gb/s): 3.0
        Annualized Workload Rate (TB/yr): 271.41
        Total Bytes Read (TB): 7.94
        Total Bytes Written (TB): 8.74
        Encryption Support: Not Supported
        Cache Size (MiB): 256.00
        Read Look-Ahead: Enabled
        Write Cache: Enabled
        Low Current Spinup: Enabled
        SMART Status: Good
        ATA Security Information: Supported
        Firmware Download Support: Full, Segmented, Deferred
        Specifications Supported:
                ACS-3
                ACS-2
                ATA8-ACS
                ATA/ATAPI-7
                ATA/ATAPI-6
                ATA/ATAPI-5
                ATA/ATAPI-4
                SATA 3.1
                SATA 3.0
                SATA 2.6
                SATA 2.5
                SATA II: Extensions
                SATA 1.0a
                ATA8-AST
        Features Supported:
                Sanitize
                SATA NCQ
                SATA Rebuild Assist
                SATA Software Settings Preservation [Enabled]
                SATA Device Initiated Power Management
                Power Management
                Security
                SMART [Enabled]
                48bit Address
                PUIS
                GPL
                Streaming
                SMART Self-Test
                SMART Error Logging
                Write-Read-Verify
                AMAC
                EPC [Enabled]
                Sense Data Reporting
                SCT Write Same
                SCT Error Recovery Control
                SCT Feature Control
                SCT Data Tables
                Host Logging
                Seagate In Drive Diagnostics (IDD)
                Seagate Remanufacture

```

jangrewe, 2019-02-23:
One more thing, though: I wanted to disable the spindown (as these are Pro drives in a NAS), but this doesn't seem to work:

```
root@backbox:~/openSeaChest/Make/gcc# ./openseachest_exes/openSeaChest_PowerControl -d /dev/sda --setAPMLevel 128
==========================================================================================
 openSeaChest_PowerControl - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2018 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_PowerControl Version: 1.10.0-1_19_0 ARM64
 Build Date: Feb 19 2019
 Today: Sat Feb 23 23:06:53 2019
==========================================================================================

/dev/sda - 0NE0004-1ZF11G - LK6 - ATA
Setting APM Level is not supported on this device.
```

What would be the correct command / argument combination to achieve this?

xahmad, 2019-02-24:
@jangrewe glad the compile now works for you. I will close this issue for now. 

Since your question regarding disabling spindown is not related to this particular issue, would you please create a new issue?

closed 2019-02-24 by xahmad