Seagate/openSeaChest #4: Can't build openSeaChest_Security

achelon5, 2018-06-23:
Hi,

I have been trying to compile this toolset on Solaris 11.3 and everything builds fine but openSeaChest_Security is never generated and the source files required to compile it don't seem to be included. Is this by design or in error?

Advice appreciated.

Regards,
Achelon

xahmad, 2018-06-25:
Hi @achelon5 

Yes, it is by design. The _Security utility depends on certain libraries that we are not ready to published yet. If you need it, please let me know and we can try to build SeaChest_Security for Solaris 11.3 and post it here: 
https://github.com/Seagate/ToolBin/tree/master/SeaChest/Security/v1.6.1

Are the other utilities that you built are working in your 11.3 system?

achelon5, 2018-06-25:
Hi,

I built using openSeaChest_18.04 source code and all other utilities work flawlessly. The Makefile isn't entirely Solaris friendly, e.g. by default it errors with 

```
strip -s openseachest_exes/openSeaChest_Erase_dbg
usage: strip [-lVx] file...
gmake: *** [openSeaChest_Erase] Error 1
```

but this is minor point and the reality is the files are built correctly and they work.

I would be immensely grateful if you could build me a Solaris 11.3 compatible Security utility using openSeaChest_18.04 code.

Achelon

xahmad, 2018-06-27:
The Solaris binaries are not posted under the ToolBin project. 
For example, the SeaChest_Security is posted here:
https://github.com/Seagate/ToolBin/tree/master/SeaChest/Security/v1.6.1

Let us know if this works for you and I will close this issue.

achelon5, 2018-06-27:
This works - Many thanks. Feel free to close this issue. Would be great if you could provide the sourcecode in the future though. As TCG operations are industry standard I'd assume there isn't a commercial reason not to disclose the code behind them.

closed 2018-07-03 by xahmad

achelon5, 2019-10-20:
Is it possible for you to rebuild this for me using the latest code so I can make use of the new features you have added regarding ATA Password management?