Seagate/openSeaChest #5: Errors on Solaris 11.4

achelon5, 2018-08-30:
Hi,

I have been testing this application on the newly released Solaris 11.4 and am receiving errors in the system log that did not occur in Solaris 11.3. While I appreciate this may indicate a regression within the Operating System itself, I'd appreciate it if someone could take a look at this and let me know if this is a bug in OpenSeaChest and what the implications of it is.

An example of the issue is as follows. I execute:

`sudo ./openSeaChest_Info --deviceInfo --device /dev/rdsk/c1t0d0`

The command completes successfully but the system log shows:
```
Aug 30 21:37:00 ahci: [ID 296163 kern.warning] WARNING: ahci0: ahci port 0 has task file error
Aug 30 21:37:00 ahci: [ID 687168 kern.warning] WARNING: ahci0: ahci port 0 is trying to do error recovery
Aug 30 21:37:00 ahci: [ID 693748 kern.warning] WARNING: ahci0: ahci port 0 task_file_status = 0x451
Aug 30 21:37:00 ahci: [ID 332577 kern.warning] WARNING: ahci0: the below command (s) on port 0 are aborted
Aug 30 21:37:00 ahci: [ID 117845 kern.warning] WARNING: satapkt 0xffffa1c01451f990: cmd_reg = 0x47 features_reg = 0x0 sec_count_msb = 0x0 lba_low_msb = 0x0 lba_mid_msb = 0x0 lba_high_msb = 0x0 sec_count_lsb = 0x1 lba_low_lsb = 0x30 lba_mid_lsb = 0x9 lba_high_lsb = 0x0 device_reg = 0xa0 addr_type = 0x4 cmd_flags = 0x52
Aug 30 21:37:01 ahci: [ID 657156 kern.warning] WARNING: ahci0: error recovery for port 0 succeed
Aug 30 21:37:01 ahci: [ID 811322 kern.info] NOTICE: ahci0: ahci_tran_reset_dport port 0 reset device
Aug 30 21:37:01 scsi: [ID 107833 kern.warning] WARNING: /pci@0,0/pci103c,1609@11/disk@0,0 (sd0):
Aug 30 21:37:01	disk not responding to selection
```

Please see attachment [seachest_log.txt](https://github.com/Seagate/openSeaChest/files/2337899/seachest_log.txt) for verbose log of data exchange with the Seagate drive.

Regards,
Achelon

vonericsen, 2018-09-04:
I looked through the attached log file and see the following command being issued. In the log no error is being returned to the application from the driver. This looks like a bug in the driver. I've attached the snippet of the output below that shows we aren't given anything back in the sense data about this failure.
We may be able to make some changes to when we read this log page so it is only read if it is supported by the drive but this looks like a deeper issue within Solaris. If we try to read certain logs from the drive that are supported, the drive should respond to that with an aborted command, but driver/OS should be able to be handle this and give us back the response without any problem. I suspect this error happened in earlier versions of Solaris as well, but it may not have been logged in a system log like it is in this version.
[Solaris_11.4_snippet.txt](https://github.com/Seagate/openSeaChest/files/2349424/Solaris_11.4_snippet.txt)

achelon5, 2018-09-05:
Hi,

Thank you for looking into this. So this error is simply cosmetic and does not adversely affect the functioning of the application?

Regards,
Achelon

vonericsen, 2018-09-06:
I don't see this causing any tool problems currently.
This page is being read to help identify whether the tool is talking to a host managed drive or a host aware drive.
I started a branch for this issue so that we read a list of supported pages, then only read the pages the drive reports. I'll also be making sure we're doing a little more to validate the data we're returned to make sure that this doesn't cause a problem.

vonericsen, 2018-09-06:
I've completed the changes I was working on and it is now pushed to the develop branch.
The tool will now read the list of supported pages for the identify device data log (30h) and the device statistics log (04h) before reading the subpages to prevent this error from showing up in a system log.
I also added some extra validation of the returned data to make sure there is valid data and it is from the correct page.

achelon5, 2018-09-06:
Hi,

Thanks for the update - I really appreciate you efforts. I cloned the latest code from the develop branch using:

`git clone --recurse-submodules -j8 https://github.com/Seagate/openSeaChest.git`

```
 openSeaChest_Basics Version: 2.7.3-1_18_1 X86_64
 Build Date: Sep  6 2018
 Today: Thu Sep  6 23:37:02 2018
```

and I just wished to provide some feedback:

**1)** This does not compile on Solaris with the included Makefiles (errors with Undefined symbol: .pow) when I run gmake. This is because lines 45-46 in Makefile need to read:
```
    LFLAGS +=  -lm -lrt
    TCGLFLAGS += -lm -lrt 
```
and line 39 in Makefile.openSeaChest_firmware needs to read:
`    FWDLLFLAGS += -lm -lrt`

**2)** The reported issue unfortunately still occurs on this latest build. Im still seeing errors in the Solaris system log:

```
ahci: [ID 296163 kern.warning] WARNING: ahci0: ahci port 0 has task file error
ahci: [ID 687168 kern.warning] WARNING: ahci0: ahci port 0 is trying to do error recovery
ahci: [ID 693748 kern.warning] WARNING: ahci0: ahci port 0 task_file_status = 0x451
ahci: [ID 332577 kern.warning] WARNING: ahci0: the below command (s) on port 0 are aborted
ahci: [ID 117845 kern.warning] WARNING: satapkt 0xffffa1c035adc1f0: cmd_reg = 0x47 features_reg = 0x0 sec_count_msb = 0x0 lba_low_msb = 0x0 lba_mid_msb = 0x0 lba_high_msb = 0x0 sec_count_lsb = 0x1 lba_low_lsb = 0x30 lba_mid_lsb = 0x9 lba_high_lsb = 0x0 device_reg = 0xa0 addr_type = 0x4 cmd_flags = 0x52
ahci: [ID 657156 kern.warning] WARNING: ahci0: error recovery for port 0 succeed
ahci: [ID 811322 kern.info] NOTICE: ahci0: ahci_tran_reset_dport port 0 reset device
scsi: [ID 107833 kern.warning] WARNING: /pci@0,0/pci17aa,30a5@1f,2/disk@0,0 (sd1):
disk not responding to selection 
```

I don't know if I did something wrong but your advice would be appreciated.

Regards,
Achelon

achelon5, 2018-09-07:
I can confirm that this issue is now resolved.

Many Thanks,
Achelon

closed 2018-09-09 by xahmad