Seagate/openSeaChest #6: release tarballs aren't buildable

asomers, 2018-09-26:
The release tarballs don't include the git submodules, which makes them unbuildable.  Tested on OpenSeaChest 18.04 with the release tarball posted on GitHub.

xahmad, 2018-10-04:
We are working on a 18.10 release which will be finished before the end of this month and it should have all the FreeBSD fixes.

thesamesam, 2023-01-02:
ping

vonericsen, 2023-01-06:
I added a new action that will create a `Source_With_Submodules.tar.xz` and `Source_With_Submodules.zip` when we push release tags.

It seems that the default for a release uses git archive which does not traverse submodules. I read up on this issue (multiple git repos on Github have similar issues filed) and it seems `git-archive-all` is the best solution to use, so this new Github action will use that to publish these at the releases going forward.

faithanalog, 2023-02-02:
I see the new tarball in the alpha release https://github.com/Seagate/openSeaChest/releases/tag/v23.02-alpha.1, and it looks to build just fine for me. I think it is a little weird that the folder extracts to `Source_With_Submodules.` and I'd maybe prefer it extract to `openSeaChest-23.02-alpha.1` to match the release version. But that's pretty minor. Thanks for adding this!

vonericsen, 2023-02-02:
@faithanalog I will investigate changing that to the version you mentioned. I think that would make a lot more sense.

vonericsen, 2023-03-01:
This has been fixed in https://github.com/Seagate/openSeaChest/releases/tag/v23.03
I even got the subfolder to have a nice name like @faithanalog mentioned previously! 😄 

If you have any trouble, please feel free to reopen this issue and I will look at this again.

closed 2023-03-01 by vonericsen