Seagate/openSeaChest #7: error: conflicting types for '__uintptr_t'

asomers, 2018-09-26:
The build fails on FreeBSD 11.2 because openSeaChest defines `__uintptr_t` differently than the system's C library.  Tested with git head @1d5cadd.

```
> gmake release
mkdir -p openseachest_exes
gmake -C ../../opensea-common/Make/gcc
gmake[1]: Entering directory '/usr/home/somers/src/github/seagate/openSeaChest/opensea-common/Make/gcc'
rm -f lib/libopensea-common.a lib/libopensea-common.so.1.17.5 *.o ../../src/*.o
rm -rf lib
mkdir -p lib
gcc -Wall -c -std=gnu99 -O3 -c -fPIC -I. -std=gnu99 -DDISABLE_TCG_SUPPORT  -I../../include ../../src/common.c -o ../../src/common.o
gcc -Wall -c -std=gnu99 -O3 -c -fPIC -I. -std=gnu99 -DDISABLE_TCG_SUPPORT  -I../../include ../../src/common_platform.c -o ../../src/common_platform.o
gcc -Wall -c -std=gnu99 -O3 -c -fPIC -I. -std=gnu99 -DDISABLE_TCG_SUPPORT  -I../../include ../../src/common_nix.c -o ../../src/common_nix.o
rm -f lib/libopensea-common.a
ar cq lib/libopensea-common.a ../../src/common.o ../../src/common_platform.o ../../src/common_nix.o
gmake[1]: Leaving directory '/usr/home/somers/src/github/seagate/openSeaChest/opensea-common/Make/gcc'
gmake -C ../../opensea-transport/Make/gcc
gmake[1]: Entering directory '/usr/home/somers/src/github/seagate/openSeaChest/opensea-transport/Make/gcc'
rm -f lib/libopensea-transport.a lib/libopensea-transport.so* *.o ../../src/*.o
rm -rf lib
mkdir -p lib
gcc -Wall -c -std=gnu99 -O3 -c -fPIC -I. -std=c99 -DDISABLE_TCG_SUPPORT -D_GNU_SOURCE  -I../../include -I../../../opensea-common/include   ../../src/ata_cmds.c -o ../../src/ata_cmds.o
gcc -Wall -c -std=gnu99 -O3 -c -fPIC -I. -std=c99 -DDISABLE_TCG_SUPPORT -D_GNU_SOURCE  -I../../include -I../../../opensea-common/include   ../../src/ata_legacy_cmds.c -o ../../src/ata_legacy_cmds.o
gcc -Wall -c -std=gnu99 -O3 -c -fPIC -I. -std=c99 -DDISABLE_TCG_SUPPORT -D_GNU_SOURCE  -I../../include -I../../../opensea-common/include   ../../src/ata_helper.c -o ../../src/ata_helper.o
gcc -Wall -c -std=gnu99 -O3 -c -fPIC -I. -std=c99 -DDISABLE_TCG_SUPPORT -D_GNU_SOURCE  -I../../include -I../../../opensea-common/include   ../../src/cmds.c -o ../../src/cmds.o
In file included from ../../include/nvme_helper_func.h:18:0,
                 from ../../src/cmds.c:18:
../../include/nvme_helper.h:18:23: error: conflicting types for '__uintptr_t'
 typedef unsigned int* __uintptr_t;
                       ^~~~~~~~~~~
In file included from /usr/include/machine/_types.h:6:0,
                 from /usr/include/sys/_types.h:33,
                 from /usr/include/stdio.h:41,
                 from ../../../opensea-common/include/common.h:34,
                 from ../../include/common_public.h:17,
                 from ../../include/cmds.h:16,
                 from ../../src/cmds.c:15:
/usr/include/x86/_types.h:114:20: note: previous declaration of '__uintptr_t' was here
 typedef __uint64_t __uintptr_t;
                    ^~~~~~~~~~~
gmake[1]: *** [Makefile:98: ../../src/cmds.o] Error 1
gmake[1]: Leaving directory '/usr/home/somers/src/github/seagate/openSeaChest/opensea-transport/Make/gcc'
gmake: *** [Makefile:181: opensea-libs] Error 2
```

sharma-nidhi, 2018-10-04:
The issue has been fixed. 
Can be verified with fix 
https://github.com/Seagate/opensea-transport/commit/42d2f03e45a9c86012d0cd5d2e52ac4b31639023.

asomers, 2018-10-04:
I can confirm that it builds now.

closed 2018-10-04 by asomers