Seagate/openSeaChest #8: Multiple undefined references on FreeBSD

asomers, 2018-09-26:
OpenSeaChest fails to build on FreeBSD 11.2 due to multiple undefined references.  It looks like it might be a Makefile error, because the symbols in question are defined within OpenSeaChest itself.

```
> gmake release
...
gcc ../../utils/C/openSeaChest/openSeaChest_Erase.o ../../src/EULA.o ../../src/openseachest_util_options.o -DDISABLE_TCG_SUPPORT -Wall ../../opensea-operations/Make/gcc/lib/libopensea-operations.a ../../opensea-transport/Make/gcc/lib/libopensea-transport.a ../../opensea-common/Make/gcc/lib/libopensea-common.a -o openseachest_exes/openSeaChest_Erase
../../utils/C/openSeaChest/openSeaChest_Erase.o: In function `main':
openSeaChest_Erase.c:(.text.startup+0x1ba1): undefined reference to `get_Device'
openSeaChest_Erase.c:(.text.startup+0x1e44): undefined reference to `get_Device_Count'
openSeaChest_Erase.c:(.text.startup+0x2f0d): undefined reference to `get_Device_List'
../../opensea-transport/Make/gcc/lib/libopensea-transport.a(cmds.o): In function `read_LBA':
cmds.c:(.text+0x1aa1): undefined reference to `os_Read'
../../opensea-transport/Make/gcc/lib/libopensea-transport.a(cmds.o): In function `write_LBA':
cmds.c:(.text+0x1ac1): undefined reference to `os_Write'
../../opensea-transport/Make/gcc/lib/libopensea-transport.a(cmds.o): In function `verify_LBA':
cmds.c:(.text+0x1d39): undefined reference to `os_Verify'
../../opensea-transport/Make/gcc/lib/libopensea-transport.a(cmds.o): In function `flush_Cache':
cmds.c:(.text+0x1dd9): undefined reference to `os_Flush'
../../opensea-transport/Make/gcc/lib/libopensea-transport.a(common_public.o): In function `scan_And_Print_Devs':
common_public.c:(.text+0x1f85): undefined reference to `get_Device_Count'
common_public.c:(.text+0x1fe0): undefined reference to `get_Device_List'
../../opensea-transport/Make/gcc/lib/libopensea-transport.a(sat_helper.o): In function `send_SAT_Passthrough_Command':
sat_helper.c:(.text+0x10b3): undefined reference to `send_IO'
../../opensea-transport/Make/gcc/lib/libopensea-transport.a(sat_helper.o): In function `translate_SCSI_ATA_Passthrough_Command':
sat_helper.c:(.text+0x5254): undefined reference to `send_IO'
../../opensea-transport/Make/gcc/lib/libopensea-transport.a(scsi_cmds.o): In function `scsi_Send_Cdb':
scsi_cmds.c:(.text+0x1f6): undefined reference to `send_IO'
../../opensea-transport/Make/gcc/lib/libopensea-transport.a(nvme_cmds.o): In function `nvme_Read_Ctrl_Reg':
nvme_cmds.c:(.text+0xb8b): undefined reference to `pci_Read_Bar_Reg'
../../opensea-transport/Make/gcc/lib/libopensea-transport.a(nvme_cmds.o): In function `nvme_Cmd':
nvme_cmds.c:(.text+0x1b7): undefined reference to `send_NVMe_IO'
collect2: error: ld returned 1 exit status
gmake: *** [Makefile:233: openSeaChest_Erase] Error 1
```

sharma-nidhi, 2018-10-04:
The issue has been fixed.
Can be verified in fix - 
https://github.com/Seagate/openSeaChest/commit/79b2c34726cf76ba54469c8825890af16754afb0

asomers, 2018-10-04:
I can confirm that it builds now.

closed 2018-10-04 by asomers