Seagate/openSeaChest #14: Not possible to see FreeFall values in IronWolf Disks

tuxd3v, 2019-02-03:
Hello,
I have compiled this set of tools, on Armv7( cortex A9 ), and I have been experimenting with it on linux..

First problem I encoutered:
Compiling the code..
`git clone --recursive --branch v19.01.31 https://github.com/Seagate/openSeaChest.git`

with make release -j2, I got errors:

`cd lib && ln -s libopensea-transport.so* libopensea-transport.so
ln: target 'libopensea-transport.so' is not a directory
Makefile:101: recipe for target 'libopensea-transport.a' failed
make[2]: *** [libopensea-transport.a] Error 1
make[2]: Leaving directory '/opt/openSeaChest/opensea-transport/Make/gcc'
Makefile.openSeaChest_firmware:99: recipe for target 'opensea-libs' failed
make[1]: *** [opensea-libs] Error 2
make[1]: Leaving directory '/opt/openSeaChest/Make/gcc'
Makefile:212: recipe for target 'openSeaChest_Firmware' failed
make: *** [openSeaChest_Firmware] Error 2
make: *** Waiting for unfinished jobs....`

Probably because of parallel compilation process..
Then I git cloned again, and issued only 'make release', since then it all compiled the last release version..

And all seems ok now.
But 'openSeaChest_Configure', tels me that FreeFal Aceleration values are not availlable.
`root@helios4:~# openSeaChest_Configure﻿ --device /dev/sda --freeFall info|grep -E "dev|Free"
/dev/sg0 - ST4000VN008-2DR166 - ZDH4MM9E - ATA
Free Fall control feature is not supported on this device.﻿`

Does FreeFall feature is not availlable( for IronWolfs ),
Or the tool is incapable of reporting the values, or they are even static, without possibility to query?
I would like to know, if the discs are capable to detect a FreeFall.

Thanks in Advance for your time, and also for the Seagate tools, its a needed project and a nice one!
Best Regards
tux

xahmad, 2019-02-03:
@tuxd3v would you please share the output of the following?

openSeaChest_Configure﻿ --device /dev/sda --freeFall info -v 4

tuxd3v, 2019-02-03:
Hello xahmad,
Fisrt of all thanks for your reply.
This is the output,( lots of stuff.. ):

`root@helios4:~# openSeaChest_Configure --device /dev/sda --freeFall info -v 4
==========================================================================================
 openSeaChest_Configure - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2018 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Configure Version: 1.13.0-1_19_0 ARM
 Build Date: Feb  3 2019
 Today: Sun Feb  3 20:38:12 2019
==========================================================================================
Sending ATA Identify command
Sending SAT ATA Pass-Through Command:
	Protocol: PIO
	Data Direction: Data In
	Task File Registers:
	[Feature] = 00h
	[Count] = 01h
	[LBA Lo] = 00h
	[LBA Mid] = 00h
	[LBA Hi] = 00h
	[DeviceHead] = A0h
	[Command] = ECh

  CDB:

        0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F  
  0x00 85 08 0E 00 00 00 01 00 00 00 00 00 00 A0 EC 00 

Sending command with send_IO

  Sense Data Buffer:

        0  1  2  3  4  5  6  7  
  0x00 00 00 00 00 00 00 00 00 

Sense Key: 0h = No Error
ASC & ASCQ: 0h - 0h = No Additional Sense Information
FRU: 0h = No Additional Information
	Return Task File Registers:
	[Error] = 00h
	[Count] = 00h
	[LBA Lo] = 00h
	[LBA Mid] = 00h
	[LBA Hi] = 00h
	[Device] = 00h
	[Status] = 00h

Command Time (ms): 4.01

	  Data Buffer being returned:

          0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F  
  0x0000 5A 0C FF 3F 8C 73 10 00 00 00 00 00 3F 00 00 00   Z..?.s......?...
  0x0010 00 00 00 00 20 20 20 20 20 20 20 20 20 20 20 20   ....            
  0x0020 44 5A 34 48 4D 4D 45 39 00 00 00 40 00 00 43 53   DZ4HMME9...@..CS
  0x0030 30 36 20 20 20 20 54 53 30 34 30 30 4E 56 30 30   06    TS0400NV00
  0x0040 2D 38 44 32 31 52 36 36 20 20 20 20 20 20 20 20   -8D21R66        
  0x0050 20 20 20 20 20 20 20 20 20 20 20 20 20 20 10 80                 ..
  0x0060 00 40 00 2F 00 40 00 02 00 02 07 00 FF 3F 10 00   .@./.@.......?..
  0x0070 3F 00 10 FC FB 00 10 5C FF FF FF 0F 00 00 07 00   ?......\........
  0x0080 03 00 78 00 78 00 78 00 78 00 00 01 00 00 00 00   ..x.x.x.x.......
  0x0090 00 00 00 00 00 00 1F 00 0E 8D 46 00 CC 08 40 00   ..........F...@.
  0x00A0 F0 07 6D 00 6B 34 69 7D 73 61 69 34 69 BC 73 61   ..m.k4i}sai4i.sa
  0x00B0 7F 40 DD 00 DD 00 7F 80 FE FF 00 00 00 FE 00 10   .@..............
  0x00C0 00 00 00 00 10 27 00 00 B0 BE C0 D1 01 00 00 00   .....'..........
  0x00D0 00 00 00 00 03 60 00 00 00 50 00 C5 3F B2 F9 3A   .....`...P..?..:
  0x00E0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5E 40   ..............^@
  0x00F0 5C 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00   \@..............
[... 515 lines cut ...]
  0x00F0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x0100 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x0110 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x0120 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x0130 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x0140 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x0150 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x0160 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x0170 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x0180 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x0190 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x01A0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x01B0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x01C0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x01D0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x01E0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................
  0x01F0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00   ................

Read Log Ext DMA returning: SUCCESS

/dev/sg0 - ST4000VN008-2DR166 - ZDH4MM9E - ATA
Free Fall control feature is not supported on this device.`

Thanks in Advance for the help.
regards

xahmad, 2019-02-04:
Let me start by saying that the maintainers of this project are not part of any product team and since Seagate ships many products every year, we cannot comment on your specific question regarding the Ironwolf. You can look through the Ironwolf product manuals here to get more information: https://www.seagate.com/internal-hard-drives/hdd/ironwolf/

That being said, we can tell you how this code is working. 

Here is the line of code that checks if the Free-fall Control feature set is supported by any (even non-Seagate) ATA device. 
https://github.com/Seagate/opensea-operations/blob/dd7cf5ca7af50ef91e6a7400541ad38d3db41ee6/src/operations.c#L950

This code is based on industry standard ATA Command Spec. 
http://www.t13.org/Documents/UploadedDocuments/docs2013/d2161r5-ATAATAPI_Command_Set_-_3.pdf

Table 45, Word 119, Bit 5 to be exact. 

[image]

So, if this bit is not set, we will report this feature as not supported. 

If I squint my eyes through the output that you have provided, I don't think that bit is set. 

You can put a printf in the above mention code, recompile & see for yourself.

tuxd3v, 2019-02-04:
Hello xahmad,
Thanks a lot for the information provided, and the help on this!
You put me in the right track!

One more time, thanks a lot, I will look into it. :+1: 
Best Regards,
tux