Seagate/openSeaChest #15: Problems scanning Seagate IronWolf disk thorough PCIe controller

tuxd3v, 2019-03-02:
Hello xahmad,
I noticed that there have been some problems scanning disks attached to pcie controllers..
`
:~# openSeaChest_Basics --scan
==========================================================================================
openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
Copyright (c) 2014-2018 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
openSeaChest_Basics Version: 2.8.0-1_19_0 ARM
Build Date: Feb 28 2019
Today: Sat Mar  2 18:13:39 2019
==========================================================================================
open: Bad address
open failure
Error: 14 - Bad address
open: Bad address
open failure
Error: 14 - Bad address
`
I know that openSeachest tools offers options for "pass-through":
			ata - show only ATA (SATA) devices
			usb - show only USB devices
			scsi - show only SCSI (SAS) devices
			interfaceATA - show devices on an ATA interface
			interfaceUSB - show devices on a USB interface
			interfaceSCSI - show devices on a SCSI or SAS interface
			sd - show sd device handles
			sgtosd - show the sd and sg device handle mapping

Is there any option for PCIe controllers that work?
Nowadays Pcie controllers are very used due to their bandwidth and amount of drives we can put there..

Does you know how to surpasse this?
Thanks in Advance for your time.

xahmad, 2019-03-03:
@tuxd3v would you please clone the latest & test again? From your output it seems like you are a few versions behind.

tuxd3v, 2019-03-03:
Hello,
Thanks in advance for the tip.

I cloned the last stable release v19.01.31
I could clone the master repo  :)

Thanks in Advance,
Regards

xahmad, 2019-03-03:
@tuxd3v  please clone the latest on the develop branch & if it fixes your issues, please let me know.

tuxd3v, 2019-03-04:
Hello xahmad,
One more time, thanks for the help, on this.
I git cloned the develop branch, but no success with a  pcie->sata '[ASMedia ASM1061](http://www.asmedia.com.tw/eng/e_show_products.php?cate_index=166&item=118)' chipset.. maybe with the 'ASM1062' it could be possible has it advertize SMART..don't know..

1.1) First we tried:
`/opt/test/openSeaChest/Make/gcc/openseachest_exes# ./openSeaChest_Basics --scan --scanFlags interfaceSCSI interfaceATA interfaceUSB sd sgtosd ata scsi -V 4
==========================================================================================
openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
Copyright (c) 2014-2018 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
openSeaChest_Basics Version: 2.8.0-1_19_2 ARM
Build Date: Mar  3 2019
Today: Sun Mar  3 18:35:13 2019
==========================================================================================
Version Info for openSeaChest_Basics:
    Utility Version: 2.8.0
    opensea-common Version: 1.18.0
    opensea-transport Version: 1.19.2
    opensea-operations Version: 1.23.0
    Build Date: Mar  3 2019
    Compiled Architecture: ARM
    Detected Endianness: Little Endian
    Compiler Used: GCC
    Compiler Version: 6.3.0
    Operating System Type: Linux
    Operating System Version: 4.4.154-1132
    Operating System Name: Debian GNU/Linux 9 (stretch)
Bus error`

1.2) then with deep search:
`/opt/test/openSeaChest/Make/gcc/openseachest_exes/openSeaChest_Basics --Scan --scanFlags sd sgtosd ata scsi -V 4
==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2018 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 2.8.0-1_19_2 ARM
 Build Date: Mar  3 2019
 Today: Mon Mar  4 12:49:20 2019
==========================================================================================
Version Info for openSeaChest_Basics:
        Utility Version: 2.8.0
        opensea-common Version: 1.18.0
        opensea-transport Version: 1.19.2
        opensea-operations Version: 1.23.0
        Build Date: Mar  3 2019
        Compiled Architecture: ARM
        Detected Endianness: Little Endian
        Compiler Used: GCC
        Compiler Version: 6.3.0
        Operating System Type: Linux
        Operating System Version: 4.4.154-1132
        Operating System Name: Debian GNU/Linux 9 (stretch)
Bus error`

2.1) Then with a generic search:

`/opt/test/openSeaChest/Make/gcc/openseachest_exes/openSeaChest_Basics --scan --scanFlags sd sgtosd ata scsi -V 4
==========================================================================================
openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
Copyright (c) 2014-2018 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
openSeaChest_Basics Version: 2.8.0-1_19_2 ARM
Build Date: Mar  3 2019
Today: Mon Mar  4 12:49:20 2019
==========================================================================================
Version Info for openSeaChest_Basics:
       Utility Version: 2.8.0
       opensea-common Version: 1.18.0
       opensea-transport Version: 1.19.2
       opensea-operations Version: 1.23.0
       Build Date: Mar  3 2019
       Compiled Architecture: ARM
       Detected Endianness: Little Endian
       Compiler Used: GCC
       Compiler Version: 6.3.0
       Operating System Type: Linux
       Operating System Version: 4.4.154-1132
       Operating System Name: Debian GNU/Linux 9 (stretch)
Bus error`

2.2) Then with generic deep search:
`/opt/test/openSeaChest/Make/gcc/openseachest_exes/openSeaChest_Basics --Scan --scanFlags sd sgtosd ata scsi -V 4
==========================================================================================
openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
Copyright (c) 2014-2018 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
openSeaChest_Basics Version: 2.8.0-1_19_2 ARM
Build Date: Mar  3 2019
Today: Mon Mar  4 12:49:20 2019
==========================================================================================
Version Info for openSeaChest_Basics:
       Utility Version: 2.8.0
       opensea-common Version: 1.18.0
       opensea-transport Version: 1.19.2
       opensea-operations Version: 1.23.0
       Build Date: Mar  3 2019
       Compiled Architecture: ARM
       Detected Endianness: Little Endian
       Compiler Used: GCC
       Compiler Version: 6.3.0
       Operating System Type: Linux
       Operating System Version: 4.4.154-1132
       Operating System Name: Debian GNU/Linux 9 (stretch)
Bus error`

No way..
but via usb with '--scanflags interfaceUSB', on a usb-> sata 'ASMEDIA AS2105' chipset , I info data 
 about a disk ( but its another disk ...Toshiba, I haven't yet experimented with Seagate Ironwolfs )..
`

You need there a pcie->sata way like you have for 'interfaceUSB' like 'interfacePCIe'
 ;)

Thanks in Advance for your time,
Best regards

vonericsen, 2019-03-04:
Hi @tuxd3v,
I don't know why this chipset wouldn't show up in the scan, but we haven't seen this type of issue before.
We have tested various PCIe add-in cards from AHCI cards, SAS HBAs from various manufacturers, USB bridges, and even older add-in cards for PATA devices without issues before.
All of these cards have a driver in the kernel, and the SCSI block and SCSI generic drivers should provide handles to the devices attached to them (sd and sg).
We use SG_IO to talk to the drives since we are sending specific commands to them, and our scan doesn't care what interface it is on.
All the scanFlags do is filter the output to show only the specified things in the list. So when you give a scanFlag for _interfaceUSB_, it only shows devices attached via USB. If you give a scan flag for _ata_, then it will only show devices that we detected as ATA devices (PATA and SATA).
If you do a scan without any of the scan flags, then we look for all SCSI generic devices that we can find. If we can't find any of those, then we will look for SCSI block devices since the SCSI block driver can send SG_IOs most of the time.

If you have a part number for the add-in card you're using, we can try ordering one and see if we can debug it further.

The reason we don't have an interfacePCIe, is because there is nothing that distinguishes a PCIe add-in card from something built-in to the system. Depending on how we detect the low-level interface, it'll end up being called a SCSI interface, or an ATA interface.
We detect this by reading some of the sysfs to figure out the full path of the device we're talking to. For ATA, we look for "ata" in the path and for USB, we look for "usb" etc and anything we don't recognize we consider as SCSI.

If you do _ls /dev/sd*_ and _ls /dev/sg*_ do you get anything in that list?
If you do, try doing something like _openSeaChest_Basics -d </dev/s??> -i_ and see if that works or spits anything out. Adding _-v 4 | tee someFile.txt_ will save the results of the commands we attempt on that device which may help point to the issue. Make sure it is a little v as a big V is for the version information. If you use a big V, we may be skipping the scan to show the version information and exit.

jangrewe, 2019-04-29:
You may want to have a look here: https://github.com/Seagate/openSeaChest/issues/3

tuxd3v, 2019-04-30:
@jangrewe,
Thanks for your reply!
We detected that it had to do with the kernel not having the SCSI module driver compiled and present, at that time.. :(

Does you Succeed to query your disks even without the SCSI driver 'sg' loaded?
```lua
# Enable Power UP in Standby
# Its a persistent feature
# No Power Cycle required
${SEA_PATH}/openSeaChest_Configure --device /dev/sda --puisFeature enable
${SEA_PATH}/openSeaChest_Configure --device /dev/sdb --puisFeature enable
${SEA_PATH}/openSeaChest_Configure --device /dev/sdc --puisFeature enable

# Enable Low Current SpinUP
# Its a persistent Feature
# Requires a Power Cycle
${SEA_PATH}/openSeaChest_Configure --device /dev/sda --lowCurrentSpinup enable
${SEA_PATH}/openSeaChest_Configure --device /dev/sdb --lowCurrentSpinup enable
${SEA_PATH}/openSeaChest_Configure --device /dev/sdc --lowCurrentSpinup enable
```

vonericsen, 2019-07-24:
This seems to be a system related issue and there have not been any updates in a while. Closing.

closed 2019-07-24 by vonericsen