Seagate/openSeaChest #17: fatal error: format.h: No such file or directory

jangrewe, 2019-04-29:
When running `make release` in `Make/gcc` of commit `5715ef44078a1dfcc63220543717459b35eac782`, i get:
```
mkdir -p openseachest_exes
cc -Wall -c -std=gnu99 -O3 -I../../opensea-common/include -I../../opensea-transport/include -I../../opensea-transport/include/vendor -I../../include -I../../opensea-operations/include -DDISABLE_TCG_SUPPORT ../../utils/C/openSeaChest/openSeaChest_Erase.c -o ../../utils/C/openSeaChest/openSeaChest_Erase.o
../../utils/C/openSeaChest/openSeaChest_Erase.c:45:20: fatal error: format.h: No such file or directory
 #include "format.h"
                    ^
compilation terminated.
Makefile:245: recipe for target '../../utils/C/openSeaChest/openSeaChest_Erase.o' failed
make: *** [../../utils/C/openSeaChest/openSeaChest_Erase.o] Error 1
```

This can be fixed with:
```
cp ../../opensea-operations/include/format.h ../../include/
```

vonericsen, 2019-04-30:
It looks like there was an issue with the opensea-operations submodule being at a different point and a feature was merged in after some work on the format code so something got messed up.
I updated the submodules and the utilities and was able to build them all.
Can you please try pulling the develop branch of openSeaChest and see if that fixes the issue?
The commit I pushed for this is `bafb778c312aec0b3bf139d28f60df700ad00698`

closed 2019-07-23 by vonericsen

edrozenberg, 2020-03-11:
I just ran into the same issue and I'm using `develop` (`Your branch is up to date with 'origin/develop'.`)

```
[openSeaChest.git/Make/gcc]# make release
mkdir -p openseachest_exes
cc -Wall -c -std=gnu99 -O3 -I../../opensea-common/include -I../../opensea-transport/include -I../../opensea-transport/include/vendor -I../../include -I../../opensea-operations/include -DDISABLE_TCG_SUPPORT ../../utils/C/openSeaChest/openSeaChest_Erase.c -o ../../utils/C/openSeaChest/openSeaChest_Erase.o
../../utils/C/openSeaChest/openSeaChest_Erase.c:45:10: fatal error: format.h: No such file or directory
   45 | #include "format.h"
      |          ^~~~~~~~~~
compilation terminated.
make: *** [Makefile:274: ../../utils/C/openSeaChest/openSeaChest_Erase.o] Error 1

None of the prebuilt openSeaChest binaries run on my Slackware systems (unknown issue that I can't figure out even with strace or gdb so far) and I'm not able to build from source either, so I'm using the prebuilt SeaChest binaries for now.

vonericsen, 2020-03-11:
Hi @edrozenberg,

I cannot repeat this issue using the latest develop branch code in Slackware 14.2.
Please make sure all the submodules were initialized and pulled when you cloned the source.
I typically use `git clone <repo> --recursive` to ensure all the submodules are properly pulled.
If you didn't clone the submodules at the same time, you can run `git submodule update --init --recursive` to pull all the submodules in.
Other methods of updating submodules are in [this stack overflow link](https://stackoverflow.com/questions/1030169/easy-way-to-pull-latest-of-all-git-submodules) in case your version of git doesn't support these methods.

As for the openSeaChest binaries not working, that could be because the Linux they were build in is newer than the Slackware sources. When we build SeaChest for release, we use CentOS 6.x since it is older than most other Linux's and is usually able to run without issues in newer Linux systems. The openSeaChest binaries may have been built in something newer than CentOS 6.x and won't run in some systems. The "releases" posted to openSeaChest are generated and uploaded automatically by Travis CI today, so those come from whichever Linux Travis CI is using.

edrozenberg, 2020-03-11:
@vonericsen Thanks! I was missing the submodules. Need to get in the `--recursive` habbit which I never normally do.