Seagate/openSeaChest #19: make release fails in 19.06 release

akhilerm, 2019-07-01:
`make release` fails in Release-19.06, with the following error 
```
make: *** No rule to make target '../../src/seachest_util_options.o', needed by 'openSeaChest_NVMe'.  Stop.
```
Platform:
```
uname -a
Linux MayaData 4.15.0-52-generic #56-Ubuntu SMP Tue Jun 4 22:49:08 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

akhilerm, 2019-07-01:
Fixed in #20 

Will there be a new dot release with the fix?

vonericsen, 2019-07-05:
@akhilerm, there is no current plan for a new release since there hasn't been a change to the tools themselves.
I merged this fix into our master branch which is more stable and is supposed to contain release code.
I can add a tag if you would like to reference that. There is already one named "Hotfix-Makefile_Fix", but I can create one called something like "Release-19.06.01" if that would be better.

akhilerm, 2019-07-08:
@vonericsen That would be great. A tag like `Release-19.06.01` will be much cleaner if we reference from other repos. Can you proceed with creating that tag?

vonericsen, 2019-07-15:
@akhilerm I am sorry for the delay creating the tag, but it has been created and pushed!
The tag was created as: `Release-19.06.01` on openSeaChest.git master branch with the makefile fixes.

akhilerm, 2019-07-16:
Thanks @vonericsen .

closed 2019-07-16 by akhilerm