Seagate/openSeaChest #25: Solaris binary for openSeaChest_Security requested

achelon5, 2019-10-20:
Is it possible for you to rebuild The openSeaChest_Security binary on Solaris for me using the latest code so I can make use of the new features you have added regarding ATA Password management?

Unfortunately you have not yet shared the modules publicly for me to build this myself.

Achelon

vonericsen, 2019-10-22:
Hi @achelon5,

I will work on getting a recent build of SeaChest_Security for Solaris. What features are you looking for specifically for ATA security? We don't allow setting ATA Security passwords in the tools for a few reasons, but did add update some ATA security options for clearing passwords, setting freezelock, and viewing current ATA security status.

achelon5, 2019-10-22:
Hi,

For context, I had already modified a copy of your software to add the following commands:

```
        --securityDisable       [SeaChest | ASCIIPW] [user | master]
        --secureErase           [normal | enhanced]
        --securityFreeze
        --securityUnlock        [SeaChest | ASCIIPW] [user | master]
        --securitySet   [SeaChest | ASCIIPW] [user | master] [high | maximum] [master password identifier]
        --securityStatus
        --dcoFreeze
```
I had also coded the --securityStatus command to replicate the familiar hdparm style output. E.g.
/dev/rdsk/c3t0d0 - ST10000VN0004-1ZD101 - XXXXXXXX - ATA

```
Security:
        Master password identifier code = 65534 (0xFFFE)
                supported
        not     enabled
        not     locked
                frozen
        not     expired: security count
                supported: enhanced erase
        890 min for SECURITY ERASE UNIT. 890 min for ENHANCED SECURITY ERASE UNIT
```

Given that it appeared to me that you were integrating the majority of these into your code anyway I thought I'd prefer to use yours.

I am a software engineer myself and I understand your reticence in allowing users the ability to set ATA passwords, since this is not handled gracefully by many systems. For instance Solaris does not seem to allow an ATA password locked drive to be unlocked from within Solaris, erroneously claiming the locked drives "failed to power up" but I value the ability to use and test this feature since I would like to use it if I can get it working correctly.

Regards,
Achelon

vonericsen, 2019-10-22:
Hi @achelon5,
Thanks for the information!
I will work on getting a build of this tool up for you. 
Also, great information to know about what Solaris shows for ATA security locked drives.

achelon5, 2019-10-22:
Thanks, incidentally (appreciating the fact that you are not Seagate Support) is this likely to be a fault in the drive firmware or am I correct in surmising that the Operating System itself is misinterpreting the aborting of a command in the locked state as a drive failure?

vonericsen, 2019-10-22:
@achelon5, it is most likely the operating system not understanding how to identify the drive when it is locked.
Most operating systems that I've taken bus traces on usually send identify commands, read commands, and read some logs from the drive to figure out its capabilities and whether there are any filesystems present on it.
With ATA security in a locked state, most commands will be aborted. Any media access is definitely aborted, and other commands that would change something about the drive (such as trying to change the maxLBA or write to a log).

Identify, reading certain logs (such as the identify device data log), checking the power mode, changing from active to idle or standby, and ATA security commands to unlock or erase the drive are allowed. This is a short list from a quick glance at the ACS4 spec.

I'm not certain what Solaris does for discovery, but it may be trying to read the device, not paying attention to the ATA security status bits, and getting an abort and deciding that something is wrong with the drive. I've seen similar cases to this when putting ATA locked drives on some SAS HBAs that do similar device discovery.

rudock1, 2019-10-23:
Hi @achelon5 

As @vonericsen mentioned he has compiled a version of the closed source tool SeaChest_Security now available in our GitHub ToolBin at [https://github.com/Seagate/ToolBin/tree/master/SeaChest/Security/v2.0.1/Solaris](url)

Additionally, we are underway to introduce an openSeaChest_Security project.  I am not sure at this time when that one will arrive.

Good luck with your efforts!

closed 2019-10-23 by rudock1

vonericsen, 2020-01-15:
Hi @achelon5,
Today I pushed openSeaChest_Security as a new tool that can be built from the makefile that is checked in (as well as visual studio, mingw, etc).
By default, the ability to set passwords is disabled, but can be turned on with a #define in the openSeaChest_Security.c file.
This is to limit setting passwords since that has some consequences if the system is rebooted and causes issues in some OSs and behind some controllers.

achelon5, 2020-01-17:
Thanks, I'll check it out.

Achelon