Seagate/openSeaChest #28: Add option to change ready led status on USB Seagate drives e.g. Backup Plus

Blockguy24, 2019-10-26:
When you use SeaChest_Basics or SeaChest_Configure and use the `--readyLED` option, it says it isn't supported, even thought the Seagate Dashboard used to be able to change this setting, before Toolkit removed this option. My drive is a Backup Plus Slim 2TB, serial number: `NA9B5G8B`

rudock1, 2019-11-06:
Hi @Blockguy24  sorry for the delay.  As you described, the SeaChest version of the --readyLED option is currently only using SAS interface commands.  Can you let us know how and why you used the old Dashboard option on your USB device?  Would like to understand the use case... thanks!

Blockguy24, 2020-04-09:
I had to use this option because I had disabled the light before (for whatever reason), and I wanted to re-enable it, but no such option was available in the new Dashboard.