Seagate/openSeaChest #30: Unable to change powermode seagate expansion +

atokje, 2020-01-24:
Hi.
I have an issue with my new 2.5"HDD, where it doesn't automatically spin down or go in to sleep mode. I have tried installing openSeaChest with the latest git pull. unfortunately I'm unable to change the power settings on it, and get this error message:

/dev/sg0 - Expansion+ - NA*****             - SCSI
Failed to check if drive supports modifying power conditions!
ERROR: Could not change power mode settings.

It also seems to me that I don't get all the info from the HDD:

==========================================================================================
 openSeaChest_Basics - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2020 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Basics Version: 2.9.0-1_21_12 X86_64
 Build Date: Jan 25 2020
 Today: Sat Jan 25 00:42:20 2020
==========================================================================================

/dev/sg0 - Expansion+ - NA*****             - SCSI
	Vendor ID: Seagate 
	Model Number: Expansion+      
	Serial Number: NA******
	Firmware Revision: 9300
	World Wide Name: 5000000000000001
	Drive Capacity (TB/TiB): 5.00/4.55
	Temperature Data:
		Current Temperature (C): Not Reported
		Highest Temperature (C): Not Reported
		Lowest Temperature (C): Not Reported
	Power On Time: Not Reported
	Power On Hours: Not Reported
	MaxLBA: 9767541166
	Native MaxLBA: Not Reported
	Logical Sector Size (B): 512
	Physical Sector Size (B): 4096
	Sector Alignment: 0
	Rotation Rate (RPM): Not Reported
	Form Factor: Not Reported
	Last DST information:
		Not supported
	Long Drive Self Test Time: Not Supported
	Interface speed:
		Not Reported
	Annualized Workload Rate (TB/yr): Not Reported
	Total Bytes Read (B): Not Reported
	Total Bytes Written (B): Not Reported
	Encryption Support: Not Supported
	Cache Size (MiB): Not Reported
	Read Look-Ahead: Enabled
	Write Cache: Enabled
	SMART Status: Unknown or Not Supported
	ATA Security Information: Not Supported
	Firmware Download Support: Full, Segmented
	Specifications Supported:
		SPC-4
		SBC-3
		UAS
		SPC-4
	Features Supported:
		Power Conditions
		Informational Exceptions [Mode 0]
	Adapter Information:
		Vendor ID: 0BC2h
		Product ID: 2323h
		Revision: 0000h

Is this a bug within the openSeaChest, or is it just the lack of support from the HDD firmware?

vonericsen, 2020-01-27:
Hi @atokje,

If I understand this correctly, pass-through support on Seagate USB products with UAS support is blocked by the Linux kernel, so it is not possible to issue the necessary commands to the drive to change these settings.
You may be able to do this in another OS, like Windows though (although I know that is not a good solution either).
If you look at [this link from the smartmontools documentation](https://www.smartmontools.org/wiki/SAT-with-UAS-Linux), they have a lot of detail about the issue and how to work around it, which may make this work for you.

vonericsen, 2025-05-30:
I have added some information to our own Wiki about getting USB access to run these commands [here](https://github.com/Seagate/openSeaChest/wiki/Enabling-Access-To-Additional-Commands-In-Linux)