Seagate/openSeaChest #32: Unlocking a drive locked with Toolkit using openSeaChest?

john-connett, 2020-02-28:
I have some **Seagate Backup Plus Ultra Touch** drives that can be locked, with a password, using **Toolkit** on **Windows**.

It would be really useful to be able to unlock them on other platforms (**FreeBSD**, **Linux**, **OmniOS**, ...) using either a utility or the libraries from **openSeaChest**.

Are there any example scripts or code to do this?

vonericsen, 2020-03-02:
Hi @john-connett,
I am looking into what is required to do this with openSeaChest or what will need to be implemented to get this functionality.

vonericsen, 2020-03-16:
Hi @john-connett,
I haven't forgotten about this, but I do finally have some information about how the drive you mentioned is locked/unlocked.
It's using TCG Opal specification to do the locking on the SED drive that it is using.
We do not currently have open source TCG code, but we are starting an internal discussion about open sourcing what we have. I do not have a timeline for this right now, but I will keep this updated as much as I can.

john-connett, 2020-03-17:
On 16/03/2020 22:06, Tyler Erickson wrote:

Hi @john-connett<https://github.com/john-connett>,
I haven't forgotten about this, but I do finally have some information about how the drive you mentioned is locked/unlocked.
It's using TCG Opal specification to do the locking on the SED drive that it is using.
We do not currently have open source TCG code, but we are starting an internal discussion about open sourcing what we have. I do not have a timeline for this right now, but I will keep this updated as much as I can.

Many thanks. I had a quick look at 'Unlock Drive for Windows.exe' which is available from the drive when it is locked. It's a .NET Assembly (Framework 4.0) which appears to call a native dll with six functions with names starting with 'SecureDrive' such as 'SecureDriveUnlock'. My guess is that the equivalent of this native dll is what would be useful to support cross-platform compatibility.

An open source tool I have found for SED drives is sedutil-cli. Versions are available from Drive Trust Alliance (https://github.com/Drive-Trust-Alliance/sedutil) and SEDutil (https://sedutil.com/).

I look forward to your updates.

mynameismon, 2024-10-23:
@vonericsen given 4 years have passed, has there been any updates in the situation? I would greatly appreciate if there is a way to do the same now.

vonericsen, 2024-10-24:
Hi @mynameismon,

Unfortunately, I do not have any updates on this issue.
I still want to implement this, but we have not had the resources to get into this with other features and bugs and internal requirements we've had to work on.

I want to keep this open as I know there is interest in this capability and I would like to add it to openSeaChest.

mynameismon, 2024-10-24:
thanks for the update! If not this, is it possible for a workaround/temporary solution allowing me to access my drives?

vonericsen, 2024-10-30:
@mynameismon,

Sorry I didn't see your last reply!

The only suggestions I have to try are the `sedutil` mentioned above or if you are comfortable with coding, trying the Seagate project called [TCGStorageAPI](https://github.com/Seagate/TCGstorageAPI)

I'm not involved in that project, but I know it uses the same low-level libraries as openSeaChest to communicate with the drive, but I don't remember all the details on how it works. I know the purpose of this project was to help customers that wanted to do some of their own SED management have a starting place with some common functionality already implemented. I would imagine that unlocking a drive is already in this API somewhere.