Seagate/openSeaChest #33: Ridiculous number of compiler warnings building on a modern Linux system

joekhoobyar, 2020-03-06:
Hey, I'm sorry if the title seems "mean" - but it is, at least, apt.

When building openSeaChest, there are a _ton_ of compiler warnings.

Things like:

```
/usr/include/bits/stdio2.h:36:10: note: '__builtin___sprintf_chk' output between 2 and 21 bytes into a destination of size 19
   36 |   return __builtin___sprintf_chk (__s, __USE_FORTIFY_LEVEL - 1,
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   37 |       __bos (__s), __fmt, __va_arg_pack ());
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

And:

```
In function 'strncpy',
    inlined from 'main' at ../../utils/C/openSeaChest/openSeaChest_Logs.c:279:17:
/usr/include/bits/string_fortified.h:106:10: warning: '__builtin___strncpy_chk' specified bound depends on the length of the source argument [-Wstringop-overflow=]
  106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../../utils/C/openSeaChest/openSeaChest_Logs.c:28:
../../utils/C/openSeaChest/openSeaChest_Logs.c: In function 'main':
../../utils/C/openSeaChest/openSeaChest_Logs.c:279:62: note: length computed here
  279 |                 strncpy(MODEL_STRING_FLAG, optarg, M_Min(40, strlen(optarg)));
../../opensea-common/include/common.h:264:35: note: in definition of macro 'M_Min'
  264 |     #define  M_Min(a,b)    (((a)<(b))?(a):(b))
      |                                   ^
```

I can't be the only person seeing these warnings.

These warnings are usually a sign that the code uses unsafe C programming practices.

Is there a plan to get these cleaned up?

xahmad, 2020-03-06:
Can you post your gcc/clang version?

BTW, some of them are just laziness on our part, which should definitely be fixed, but others are challenge of being able to write code that works in Windows/Linux/FreeBSD and all the different compliers & compiler versions that are supported in each of these OSes.

joekhoobyar, 2020-03-06:
gcc version 9.2.0, everywhere I've compiled it (including a cross-compiler).

On my amd64 boxes:

```
$ gcc -v -c
Using built-in specs.
COLLECT_GCC=gcc
Target: x86_64-pc-linux-gnu
Configured with: /var/tmp/portage/sys-devel/gcc-9.2.0-r2/work/gcc-9.2.0/configure --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/gcc-bin/9.2.0 --includedir=/usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/include --datadir=/usr/share/gcc-data/x86_64-pc-linux-gnu/9.2.0 --mandir=/usr/share/gcc-data/x86_64-pc-linux-gnu/9.2.0/man --infodir=/usr/share/gcc-data/x86_64-pc-linux-gnu/9.2.0/info --with-gxx-include-dir=/usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/include/g++-v9 --with-python-dir=/share/gcc-data/x86_64-pc-linux-gnu/9.2.0/python --enable-languages=c,c++ --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --disable-nls --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo Hardened 9.2.0-r2 p3' --enable-esp --enable-libstdcxx-time --disable-libstdcxx-pch --enable-shared --enable-threads=posix --enable-__cxa_atexit --enable-clocale=gnu --enable-multilib --with-multilib-list=m32,m64 --disable-altivec --disable-fixed-point --enable-targets=all --enable-libgomp --disable-libmudflap --disable-libssp --disable-systemtap --enable-vtable-verify --disable-libquadmath --enable-lto --without-isl --enable-default-pie --enable-default-ssp
Thread model: posix
gcc version 9.2.0 (Gentoo Hardened 9.2.0-r2 p3) 
```

On my 32-bit arm boxes:

```
$ gcc -v -c
Using built-in specs.
COLLECT_GCC=gcc
Target: armv7l-unknown-linux-gnueabihf
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://github.com/archlinuxarm/PKGBUILDs/issues --enable-languages=c,c++,fortran,go,lto,objc,obj-c++,d --enable-shared --enable-threads=posix --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --disable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --host=armv7l-unknown-linux-gnueabihf --build=armv7l-unknown-linux-gnueabihf --with-arch=armv7-a --with-float=hard --with-fpu=vfpv3-d16 gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
gcc version 9.2.0 (GCC) 
```

And my cross-compiler for arm:

```
$ arm-hardfloat-linux-gnueabihf-gcc -v -c
Using built-in specs.
COLLECT_GCC=arm-hardfloat-linux-gnueabihf-gcc
Target: arm-hardfloat-linux-gnueabihf
Configured with: /var/tmp/portage/cross-arm-hardfloat-linux-gnueabihf/gcc-9.2.0-r2/work/gcc-9.2.0/configure --host=x86_64-pc-linux-gnu --target=arm-hardfloat-linux-gnueabihf --build=x86_64-pc-linux-gnu --prefix=/usr --bindir=/usr/x86_64-pc-linux-gnu/arm-hardfloat-linux-gnueabihf/gcc-bin/9.2.0 --includedir=/usr/lib/gcc/arm-hardfloat-linux-gnueabihf/9.2.0/include --datadir=/usr/share/gcc-data/arm-hardfloat-linux-gnueabihf/9.2.0 --mandir=/usr/share/gcc-data/arm-hardfloat-linux-gnueabihf/9.2.0/man --infodir=/usr/share/gcc-data/arm-hardfloat-linux-gnueabihf/9.2.0/info --with-gxx-include-dir=/usr/lib/gcc/arm-hardfloat-linux-gnueabihf/9.2.0/include/g++-v9 --with-python-dir=/share/gcc-data/arm-hardfloat-linux-gnueabihf/9.2.0/python --enable-languages=c,c++ --enable-obsolete --enable-secureplt --disable-werror --with-system-zlib --disable-nls --enable-checking=release --with-bugurl=https://bugs.gentoo.org/ --with-pkgversion='Gentoo Hardened 9.2.0-r2 p3' --enable-esp --enable-libstdcxx-time --disable-libstdcxx-pch --enable-poison-system-directories --with-sysroot=/usr/arm-hardfloat-linux-gnueabihf --disable-bootstrap --enable-__cxa_atexit --enable-clocale=gnu --disable-multilib --disable-altivec --disable-fixed-point --with-float=hard --enable-libgomp --disable-libmudflap --disable-libssp --disable-systemtap --enable-vtable-verify --disable-libquadmath --enable-lto --without-isl --enable-default-pie --enable-default-ssp
Thread model: posix
gcc version 9.2.0 (Gentoo Hardened 9.2.0-r2 p3) 
```

joekhoobyar, 2020-03-06:
I also opened #34 to fix missing MAKEFLAG on sub-make invocations.

vonericsen, 2020-03-06:
Thanks for reporting this @joekhoobyar 
I'm set up a newer system with GCC 9 and can see the large number of warnings.
It looks like some of these were introduced in GCC9. I will figure out how to clean these up.
I have been using GCC 7.4.0 which is why I haven't seen many of them before.

I'll also take a look at #34 today as well.

vonericsen, 2020-03-06:
Hi @joekhoobyar, 
I've pushed some changes to fix some of the warnings.
A majority of them are a warning for possible unaligned addresses when accessing packed/aligned structures.
I fixed most of those on the opensea-transport branch "Warning_Packed_Fixes", but I need to do more testing before I finish that branch and before I fix the remaining warnings that are still present.

joekhoobyar, 2020-03-14:
That's awesome @vonericsen - this will really help increase confidence in cross-compiled builds.  I am planning on getting a 32-bit arm version ready for a dependent project: `openebs`

vonericsen, 2020-04-13:
Hi @joekhoobyar,

I have finished cleaning up all the warnings I could find in GCC9. I also ran clang and fixed a few more that were found by clang as well.
All of the changes are pushed on develop now.
The only warning that I see remaining is from clang related to some #pragma message output we use to help understand which NVMe include file is being used in Linux. 
If you find others, please let us know! We do our best to keep free of warnings, but occasionally miss some.

closed 2020-04-13 by vonericsen