Seagate/openSeaChest #36: Can't upgrade firmware to ST2000DM001-1CH164

AlexDaciuk, 2020-04-18:
Hello

I'm trying to upgrade the firmware of my ST2000DM001-1CH164 from CC27 to CC29

Machine 

CPU : Ryzen 3700X
RAM : 16 GB DDR4 3200 Mhz
Motherboard: Asus X570 Gaming Plus
OS : Arch Linux
Kernel Version :  5.6.4 or 5.4.32-lts (tried with both)

SHA256 of the GBP2TBCC29.LOD file is:

`be1c6c79b8da5e2b7caa86f6aa86fd972ebcf88dbd627bc1e71aae487f8e941c`

Cannot compare to anything as Seagate does not provide any hash

OpenSeaChest Scan returns

```
root:Downloads/ # ./openSeaChest_Firmware --scan                                                                                                                                                                                                                              [23:47:33]
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2020 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 2.8.1-1_21_30 X86_64
 Build Date: Mar  9 2020
 Today: Fri Apr 17 23:47:52 2020
==========================================================================================
Vendor   Handle       Model Number            Serial Number          FwRev     
ATA      /dev/sda     ST2000DM001-1CH164      Z340EMCF               CC27      
ATA      /dev/sdb     Samsung SSD 850 PRO 256GB S251NXAG819278X        EXM04B6Q  
ATA      /dev/sdc     HGST HDN726040ALE614    K3GPBP0L               APGNW7JH  
NVMe     /dev/nvme0n1 Sabrent Rocket 4.0 500GB 036C079A181100340429   RKT401.1  

```

After running the command 

`openSeaChest_Firmware -d /dev/sda --downloadFW GBP2TBCC29.LOD`

I get the message 

```
root:Downloads/ # ./openSeaChest_Firmware -d /dev/sda --downloadFW GBP2TBCC29.LOD                                                                                                                                                                                             [23:48:30]
==========================================================================================
 openSeaChest_Firmware - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2020 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Firmware Version: 2.8.1-1_21_30 X86_64
 Build Date: Mar  9 2020
 Today: Fri Apr 17 23:49:28 2020
==========================================================================================

/dev/sda - ST2000DM001-1CH164 - Z340EMCF - ATA
.....................................................
Firmware Download successful
Firmware Download time (ms): 189.62
Average time/segment  (ms): 3.57
Activate Time          (ms): 125.28
New firmware version is CC27
```

So, it did not upgrade the firmware, CC27 is the actual firmware of the device

If I run the same command with verbosity 2 I get 

`
.Sending ATA Download Microcode, subcommand 0x3
Download Microcode returning: FAILURE
`

Full logs in here: [openseachest_logs.zip](https://github.com/Seagate/openSeaChest/files/4495941/openseachest_logs.zip)

All commands run as root

Can you help me?

Cheers

vonericsen, 2020-04-18:
Hi @AlexDaciuk,
Thank you for the logs.
What is the drive attached to (Motherboard, SATA controller, SAS controller, etc)? If it is the motherboard SATA ports, can you tell me if it is in AHCI or RAID mode?

AlexDaciuk, 2020-04-18:
Hi,

It is attached to the integrated SATA ports of the motherboard and it is in AHCI mode

`lspci | grep SATA` returns 

`SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 51)`

rudock1, 2020-04-21:
Hi @AlexDaciuk 

Matching firmware to a given product can sometimes be tricky due to various designs and configurations that Seagate ships. While your SHA matches what we found for the "2TB" file, it seems that is not the correct firmware file for your drive. Due to the age of the drive family I could not find an expert to ask.

Our closed source version, SeaChest_Firmware, has an additional feature and can utilize the .cfs configuration file that is included with the SATA firmware ZIP package. Hopefully, it will take care of matching the correct file to the drive. Place cfs and lod files in the same configuration as seen in the package (generally cfs and all lod files in the same directory).  We have binary versions here in the GitHub Seagate ToolBin: https://github.com/Seagate/ToolBin/tree/master/SeaChest/Firmware

If you prefer to continue using the open source version you have, then sending the 3TB lod file (MD5:A709F515AE74C7F28FC3A7C536EFFF10) may work.

One last note for customers upgrading their firmware: modern Seagate drives do a lot to protect themselves from flashing incorrect firmware (reference year of 2012 or so). Older products may not have the same level of protection. The SATA firmware update packages from Seagate.com that use cfs files will make sure that only appropriate firmware is downloaded to a given drive to prevent any bad firmware flashes that might occur. 

Always backup your data before using firmware in case something awful did go wrong during the upgrade process.

AlexDaciuk, 2020-04-21:
Ok, that make sense a little but the thing is that the firmware package I donwloaded is the one suggested by Seagate own searcher (https://apps1.seagate.com/downloads/request.html) when I put the serial number of the HDD, so, it's Seagate own web page returning a non matching firmware?

Weirdly enough, there is a [github wiki ](https://github.com/jandelgado/general/wiki/Firmware-update-of-Seagate-harddisk-using-Linux)about upgrading it with hdparm directly that uses the exactly same HDD

The model and serial number I use it's the one provided by `hdparm -I /dev/sda`

```
root:Firmware_Seagate/ # hdparm -I /dev/sda                                                                                                                           

/dev/sda:

ATA device, with non-removable media
        Model Number:        ST2000DM001-1CH164                      
        Serial Number:         Z340EMCF
        Firmware Revision:  CC27    
        Transport:                Serial, SATA 1.0a, SATA II Extensions, SATA Rev 2.5, SATA Rev 2.6, SATA Rev 3.0
Standards:
        Used: unknown (minor revision code 0x001f) 
        Supported: 9 8 7 6 5 
        Likely used: 9
```

The file that I downloaded has the name `Barracuda-GrenadaBP-CC29-CC49.zip` , I really don't know what GrenadaBP stands for, its sha256 sum is `8873c0a96e8c850ae6bf4970bad1ce146d679450e362a8046baec447dad6a749` , just for reference

The output from SeaChest it's the same that the one from OpenSeaChest when trying to upgrade the firmware, gonna try using the cfs file to see if it changes anything

I should email seagate to see if they can provide the correct firmware?

Cheers 
Alex

AlexDaciuk, 2020-04-21:
Ok, after using the cfs file, it upgraded to CC29

I downloaded latest SeaChest_Firmware from the repo you linked and used the CFS file, this is the output

```
root:firmware/ # ./SeaChest_Firmware_282_12130_64 -d /dev/sda --fwdlConfig GPCC2949.CFS                                                                                                                                                                                       [17:25:40]
==========================================================================================
 SeaChest_Firmware - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2020 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Firmware Version: 2.8.2-1_21_30 X86_64
 Build Date: Apr 13 2020
 Today: Tue Apr 21 17:26:15 2020
==========================================================================================

/dev/sda - ST2000DM001-1CH164 - Z340EMCF - ATA
Downloading file GBP3TBCC29.LOD
Error reading contents of firmware file!
.....................................................
Firmware Download successful!
Firmware Download time (s): 6.39
Average time/segment (ms): 120.56
Activate Time          (s): 6.33
New firmware version is CC29
Successfully validated firmware after download!
```

and --scan tell me that the firmware is at CC29, so, the upgrade was successful

```
root:firmware/ # ./SeaChest_Firmware_282_12130_64 --scan                                                                                                                                                                                                                      [17:26:21]
==========================================================================================
 SeaChest_Firmware - Seagate drive utilities - NVMe Enabled
 Copyright (c) 2014-2020 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 SeaChest_Firmware Version: 2.8.2-1_21_30 X86_64
 Build Date: Apr 13 2020
 Today: Tue Apr 21 17:26:30 2020
==========================================================================================
Vendor   Handle       Model Number            Serial Number          FwRev     
ATA      /dev/sda     ST2000DM001-1CH164      Z340EMCF               CC29  

```

So, thanks for the cfs suggestion, that worked out, still can't understand why it used the 3TB firmware on my 2TB unit

closed 2020-04-23 by AlexDaciuk