Seagate/openSeaChest #37: FreeBSD NVME support

darkfiberiru, 2020-05-18:
openSeaChest needs to be able to talk to FreeBSD nvme driver. As far as I'm aware just needs low level plumbing

darkfiberiru, 2020-05-18:
I assume this would be added here https://github.com/Seagate/opensea-transport/blob/develop/src/cam_helper.c and maybe rename cam_helper.c to freebsd_helper.c as there is a lower nvme layer that supports  a direct nvme implmentation and a implemntation of nvme through cam.

vonericsen, 2020-05-18:
Hi @darkfiberiru,
All it needs is the plumbing and some testing. We have been investigating it and know that is it possible to add and make work and have started some work internally for it.
It'll probably go into cam_helper.c for now, but we are thinking about how we want to better organize the files and file names in the future.
Thank you!

darkfiberiru, 2020-12-18:
This may be good starting/reference  point with compat layer work @bsdimp did on nvme-cli

https://github.com/bsdimp/nvme-cli/commits/freebsd-port

vonericsen, 2020-12-22:
Hi @darkfiberiru,
Thanks for the reference!

I chatted with @pranali-tirkhunde and she has been making good progress on this support. She is finalizing some testing to ensure it is fully functional, then we'll review the code and merge it to develop when it looks good.

darkfiberiru, 2020-12-22:
No problem thanks for the work on this. Were shipping toolset both on TrueNAS CORE/Enterprise based on freebsd and TrueNAS Scale based on linux so parity is nice.

bsdimp, 2020-12-23:
> I chatted with @pranali-tirkhunde and she has been making good progress on this support. She is finalizing some testing to ensure it is fully functional, then we'll review the code and merge it to develop when it looks good.

I'm around as well. I'd be happy to review anything, either in nvme-cli or in nvmecontrol. Chuck Tuffli @ctuffli also has done some work in this area...

pranali-tirkhunde, 2021-01-08:
Hi @darkfiberiru and @bsdimp,
We have checked in the latest code for adding NVMe support in FreeBSD. Please feel free to check that with the develop branch for openSeaChest and let us know if you face any issues.

vonericsen, 2021-04-28:
Closing this since we haven't heard anything and have found this to be working in our own internal testing.
We've also rolled this into release v21.04.27 and some notes can be found in #54. 
If anyone has any trouble with this, please either reopen this issue, or create a new one about what your issue is. 
Thanks!

closed 2021-04-28 by vonericsen

darkfiberiru, 2021-04-29:
Thank you for the work on this. I much appreciate the project