Seagate/openSeaChest #40: Make a CI/CD

tomidix, 2020-10-07:
Make a CI/CD so people don't have to go through the build process itself..
Having the tools available just to run is so much faster for us.. so many work hours would be saved..

vonericsen, 2020-10-16:
Hi @tomidix,
Thanks for the suggestion. We will look into adding a CI/CD for the Windows build and see what we can get going.

Kaos-Industries, 2020-12-29:
Would also be interested in this. A Windows binary would work out of the box using Cygwin - for which `hdparm` is no longer maintained and therefore an alternative is needed - but actually building the binary with the Cygwin toolchain would be a hassle if past experience is anything to go by.

xahmad, 2021-01-04:
We used to have Travis CI building Windows binaries on a release tag. Then it stopped working. I reached out to Travis support and at the time (2+ years ago) for all open source projects, they were not providing any debug support for Windows, so the automatic build was commented out from the yml file. 

@vonericsen maybe...check the git history on the travis.yml file to see if the old one with Windows would build now.

@Kaos-Industries have you looked at the releases posted here?
https://github.com/Seagate/ToolBin

vonericsen, 2021-03-04:
Wanted to provide an update. Yesterday I added Github actions for CI on Windows and Linux, so those are now running.
I'm still working my way through setting them up to trigger posting release builds automatically.

I need to investigate how Travis and Cirrus CI are setup and whether they are still needed or not. CirrusCI will likely stay to provide FreeBSD CI, but I will try to get it updated to have the same capabilities to build releases too.

vonericsen, 2022-04-14:
I am closing this issue now as there is a lot of active CI including generating artifacts. If there are additional requests to enhance the CI for openSeaChest, please open a new issue describing what needs to be done.

closed 2022-04-14 by vonericsen