Seagate/openSeaChest #41: ST16000VN001 EPCfeature disable not supported

Deepcuts, 2020-10-15:
Hello,

I have 4 Seagate Ironwolf NAS 16 TB ST16000VN001 drives.
I would like to change the EPCfeature but for some reason Seachest reports that these drives do not support EPC

```
SeaChest_PowerControl_201_12130_64.exe -d PD2 --EPCfeature disable
EPC Feature is not supported by this device \\.\PhysicalDrive2.
```
[image]

Nevertheless, -i shows that EPC is enabled
[image]

I have opened a ticket with Seagate but after about 5 replies I am convinced the people replying have absolutely no connection to Seagate.

vonericsen, 2020-10-16:
Hi @Deepcuts,
Thanks for the report. This looks like a bug in detecting the EPC feature when trying to disable it. 
We may or may not already have this fixed in openSeaChest, but I will need to test and confirm it first.

If you have the ability to compile and test openSeaChest for this, that would be great, but if not, we'll look into it to make sure this is corrected before our next release.

Deepcuts, 2020-10-20:
Hi @vonericsen ,

I was able to compile on linux.
Ran the newly compiled version and was able to disable EPC on all four drives.
No more click freakin' clack all the time after a small idle period. Ordering a Voodoo doll now for whomever came up with this EPC enabled by default on performance oriented drives.
[image]

Shutdown for good measure.
Back in Windows and -i shows EPC is not longer enabled.
[image]

Thank you for pointing me in the right direction.

closed 2020-10-20 by Deepcuts

vonericsen, 2020-10-26:
Hi @Deepcuts 
Thanks for the update! Glad that was able to work for you!
openSeaChest and SeaChest share the same code for this operation, so it will be fixed when we roll out a new build of SeaChest.