Seagate/openSeaChest #42: data cmplt err -75 & -71

drzraf, 2020-10-23:
On a brand new WD green SSD (SATA, 480GB), plugged via [this enclosure](http://www.fideco-it.com/copy_prod_details_081448_2378373.html), any openSeaChest operation triggers the following warning:

```
sd 3:0:0:0: [sg3] tag#28 data cmplt err -75 uas-tag 1 inflight: CMD 
sd 3:0:0:0: [sg3] tag#28 CDB: Report luns a0 00 00 00 00 00 00 00 00 08 00 00
sd 3:0:0:0: [sg3] tag#29 data cmplt err -71 uas-tag 1 inflight: CMD 
sd 3:0:0:0: [sg3] tag#29 CDB: Inquiry 12 01 00 00 60 00
```

vonericsen, 2020-10-26:
Hi @drzraf ,

This is a USB adapter issue where it is not handling the Report LUNs command properly. I wrote and pushed a workaround a few minutes ago to avoid sending this command which should help.

If you are able to run the following command and report the output, it will help make sure we are able to handle any other issues it may have. Please run both of these as one option can occasionally limit the output due to the nature of how some adapters respond.
`openSeaChest_PassthroughTest -d /dev/sg? --runPTTest --ptDriveHint ata --ptTypeHint sat --enableHangCmdsTest all `
`openSeaChest_PassthroughTest -d /dev/sg? --runPTTest --ptDriveHint ata --ptTypeHint sat`

What this tool does is it runs through a lot of known issues and checks to see which may be present in the adapter so that workarounds in the code can be applied for it. It would be best to unplug and plug it back in before testing to make sure it is in a "good" state and not hanging onto issues from previously running it. When I wrote this tool, I wrote it to be data-safe, so it should not affect any data you may have written to it.
If you can provide the full output from the tool in here, I will add it to the code's internal "database" that looks up and works around known problems that some adapters experience.

I have requested ordering one to be able to do further testing for additional issues, but it will take some time before I'm able to sit and test it and provide more support via code changes.

drzraf, 2020-10-29:
```
Version Info for openSeaChest_PassthroughTest:
	Utility Version: 0.3.1
	opensea-common Version: 1.19.11
	opensea-transport Version: 1.21.30
	opensea-operations Version: 1.24.21
	Build Date: Mar  9 2020
	Compiled Architecture: X86_64
	Detected Endianness: Little Endian
	Compiler Used: GCC
	Compiler Version: 7.5.0
	Operating System Type: Linux
	Operating System Version: 5.4.0-52
	Operating System Name: Ubuntu 20.04.1 LTS
```

[pttest.txt](https://github.com/Seagate/openSeaChest/files/5455633/pttest.txt)
[pttestEnableHangCmds.txt](https://github.com/Seagate/openSeaChest/files/5455634/pttestEnableHangCmds.txt)

vonericsen, 2020-10-29:
Hi @drzraf,
Thanks for the files! I went through them and added what the tool was able to detect to opensea-transport to help improve support for it.
If you are able to compile the latest develop branch, you should get this change.

vonericsen, 2021-03-18:
The release [v21.03.01](https://github.com/Seagate/openSeaChest/releases/tag/v21.03.01) includes these changes for this device.

Please test it and if this fixes it for you, please close this issue.

vonericsen, 2022-02-10:
Closing due to inactivity.

closed 2022-02-10 by vonericsen