Seagate/openSeaChest #43: Incorrect exit code returned when permissions error occurs with --onlySeagate flag

natanlidukhover, 2020-11-06:
When running a -i on all drives (-d all) without proper permissions (i.e. user not in disk group and/or not run as root/sudo), the drive information is not accessed and the exit code returned is 3. However, when including the --onlySeagate flag, the drive information is similarly not accessed, and a permissions error is displayed, but the exit code returned is 0, indicating a successful operation.

The same bug occurs with a --modelMatch ST flag as well. The issue also occurs on a machine with all Seagate drives. I have included a few images of the issue occurring below, where the user is neither a superuser nor in the "disk" group.

[image]
[image]

vonericsen, 2020-11-10:
The reason this does not return any errors with `--onlySeagate` or `--modelMatch` is because those devices will be entirely skipped if they do not have the necessary information for the tool to make those comparisons. Then, since none of the options are actually run on an device, no errors are ever detected to trigger a different exit code.

I will need to think about the best way to handle this.

vonericsen, 2025-05-30:
Fixed in [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)

closed 2025-05-30 by vonericsen