Seagate/openSeaChest #48: Not able to wipe/sanitize a Capacity Enterprise 2TB drive...please help

MrM40, 2021-02-27:
Result from **--deviceInfo**:
`ATA Security Information: Supported, Enabled, Locked
Features Supported:
                Sanitize
                Security [Enabled]`

Result from **--ataSecurityInfo**:
`Security State: 4
        Enabled: True
        Locked: True
        Frozen: False
        Password Attempts Exceeded: False
Master Password Capability: High
Master Password Identifier: 65534 (may be set to manufacture master password)
Enhanced Erase Time Estimate:  3 hours 16 minutes
Security Erase Time Estimate:  3 hours 16 minutes
All user data is encrypted: False
Restricted Sanitize Overrides ATA Security: False
SAT security protocol supported: False`

Result from **--sanitize info**:
`Sanitize
        The following sanitize commands are supported:
                Overwrite       --sanitize overwrite`

Result from **--sanitize overwrite**:
`Sanitize
Starting Sanitize Overwrite Erase
Sanitize command failed!`

Result from **--ataSecureErase normal**:
`ATA Security Erase
Starting ATA Security Erase using User password:  "SeaChest" (User)

        Current Time: Sun Feb 28 00:27:34 2021
        Drive reported completion time:  3 hours 16 minutes from now.
        Estimated completion Time : Sun Feb 28 03:43:34 2021

        Please DO NOT remove power to the drive during the erase
        as this will leave it in an uninitialized state with the password set.
        If the power is removed, rerun this test with your utility.
        Upon erase completion, the password is automatically cleared.

        ATA Security erase failed to complete after

        WARNING!!! The drive is in a security state where clearing the password is not possible!
        Please power cycle the drive and try clearing the password upon powerup.
        Erase password that was used was:  "SeaChest" (User)`

What am I doing wrong?

vonericsen, 2021-03-01:
Hi @MrM40,

According to the ataSecurityInfo, your drive is locked. Until it is unlocked, or provided a matching password for the erase, it cannot perform any of these actions as defined in the ATA & ACS specifications for the ATA security feature.

You can attempt to provide the password to it using `--ataSecPassword yourPasswordHere` when performing the erase, but depending on how it was set and by which BIOS set it, it may be encoded in some other form that these tools will not know. Some BIOS's will send the ASCII password to the drive, others will run some proprietary encoding method for this.

Your full command line to attempt using the password of your choice or that the drive knows would be
`openSeaChest_Erase -d <yourDeviceHandle> --ataSecurityErase normal --ataSecPassword <yourPasswordHere> --confirm <confirmation here> `

If you can disable ATA security from the original machine with the original password, that will remove the need to specify a password and either sanitize or ATA security erase should be able to complete at that point. This is typically defined as a boot password in most BIOSs if you see it labelled as that.

MrM40, 2021-03-02:
Thanks for you help :-)

I assume you mean `--ataSecureErase` and not `ataSecurityErase` ;-)

I can't make it work.

The disks doesn't have SED or any other build in encryption. SuperMicro board they have been used on doesn't even support specifying a password in bios for a disk, so I pretty sure none has ever been specified.

Tried to unlock the drive:
`openSeaChest_Security --device /dev/sg1 --unlockATASec`
`openSeaChest_Security --device /dev/sg1 --ataSecPassword "" --unlockATASec`
`openSeaChest_Security --device /dev/sg1 --ataSecPassword  SeaChest --unlockATASec`
All give the same result:

>    /dev/sg1 - ST2000NM0105-1YY104 - ZC2264A5 - ATA
>    Attempting to unlock security with password =  (password is empty) (User)
>    Unable to unlock drive with password =  (password is empty) (User)
>    Failed to unlock ATA security.

vonericsen, 2021-03-02:
Hi @MrM40,

Try using "PASS" either with openSeaChest or hdparm. That appears to be setting the password to that.
Let me know if you need any other assistance. 😃

MrM40, 2021-03-02:
Yes, it worked flawlessly :-)
My mistake....thanks for your support :-)

vonericsen, 2021-03-02:
That's great! Glad to hear that solved it for you!
Please close this issue when you are ready, or I can close it in a day or so in case you have anything else pop-up that is related to this.

closed 2021-03-02 by MrM40