Seagate/openSeaChest #51: Device ST10000DM0004. "EPC Feature is not supported by this device".

EyeAndTea, 2021-03-28:
OS: Windows 7 64bit.
Device: ST10000DM0004
Tools: OpenSeaChest_PowerControl (64bit), OpenSeaChest_PowerControl(32bit)

Using OpenSeaChest_PowerControl to set idle times, or to completely disable EPC, does not work, with messages indicating that the operation is not supported on the device. I tried the arguments:
- --idle_a 20
- --EPCfeature disable

According to the manual of the device, and to the best of my understanding of it, these operations are supported by the device. The following setting does work:
- --showEPCSettings 

Modifying "Idle A" is my primary objective.

vonericsen, 2021-03-29:
Hi @EyeAndTea,

I need a little more information to investigate this issue.
Can you tell me a little about the system configuration? Is the drive attached to a motherboard or to an HBA or to a USB adapter?

Also, can you run the `-i` option and share that output? Also, share it with `-v 4` as well? You can attach it as a text file to this issue.
The two commands will look something like this:
1. `OpenSeaChest_PowerControl.exe -d PD? -i > driveInfo.txt`
2. `OpenSeaChest_PowerControl.exe -d PD? -i -v4 > driveInfoVerbose.txt`

EyeAndTea, 2021-03-29:
The device is attached to the motherboard by a sata port, via a passive connection from an internal drive bay made available by the computer case. 

Please find below the requested files.

[driveInfo.txt](https://github.com/Seagate/openSeaChest/files/6223200/driveInfo.txt)
[driveInfoVerbose.txt](https://github.com/Seagate/openSeaChest/files/6223201/driveInfoVerbose.txt)

vonericsen, 2021-03-30:
Hi @EyeAndTea,

Thanks for the logs. There is definitely a bug somewhere, but I haven't figured out where yet. The drive does seem to report EPC feature support, so these commands should work.

Can you provide the verbose output when trying the options you tried?
Ex: `OpenSeaChest_PowerControl.exe -d PD? --EPCfeature disable -v4 > driveInfoVerbose.txt`

Also, can you try running `--idle_a disable`? And the verbose output from that?

EyeAndTea, 2021-03-31:
Thank you for your continued attention. Please find attached the requested data.

[driveInfoVerbose __ --EPCfeature.txt](https://github.com/Seagate/openSeaChest/files/6236639/driveInfoVerbose.__.--EPCfeature.txt)
[driveInfoVerbose __ --idle_a.txt](https://github.com/Seagate/openSeaChest/files/6236640/driveInfoVerbose.__.--idle_a.txt)

vonericsen, 2021-03-31:
Thanks for the additional information @EyeAndTea!

I see what the problem is, but it is a strange one I haven't seen before.
The basic commands to identify the drive and get information about it work, but sending the commands to change the EPC feature result in saying "Invalid command operation code" which is a nonsense error since the same operation code was used for the first commands We are sending SCSI style SAT commands to the driver right now so opcodes will all be the same, which is why this looks like a bogus error. This can be because the driver is blocking it purposely or due to a bug since I have seen similar things in the past.

Can you send me some screenshots from the device manager?
I want to see what the controller is reporting as and the driver version so I can see if we have similar hardware I can sit and debug with.
Here are examples of what I would like to see:
[image]
In this first image, it will show the controller/adapter in one of the two red sections. I know you have some sort of AMD adapter since that is what the vendor and product IDs showed in the `-i` output, but that is not yet capable of displaying the driver and driver version.

[image]
This second image is after right clicking and selecting properties for the adapter, then selecting the driver tab.

[image]

This last image is from clicking on "Driver Details" which should be the last thing I want to look at to make sure we know exactly the driver running.

It is possible an updated driver fixes this, but I would like this information first so I can at least save some information about things that may be potentially problematic somewhere...I have a few internal docs for this right now that I want to put somewhere in the openSeaChest documentation at some point.

Also, I don't want to waste too much of your time debugging this, so if you are able to use a live CD or live USB to boot something like Ubuntu, you should be able to download and use the Linux version of openSeaChest to make your EPC changes. Linux drivers generally don't have some of the same weirdness like the Windows versions typically do for things like this. I will use whatever info you can share to investigate this further once I can get the right hardware and driver to debug it with myself.

EyeAndTea, 2021-03-31:
Here is a list of my IDE ATA/ATAPI controllers:
- AMD SATA Controller
- ATA Channel 0
- ATA Channel 1
- Standard AHCI 1.0 Serial ATA Controller

Please find attached more details about the AMD Controller. As for the other drivers, they are all from Microsoft. The last controller in the list might be referring to an ASMedia controller, which I do not use for the hard drives.

My storage controllers are all virtual on the other hand.

I tried to compile the tools before on the Linux SystemRescue distribution, but I prefer windows because I was bitten before by the persistence of these setting. If things fail to proceed, I shall certainly be trying again, and perhaps again on an Ubuntu distribution to avoid compilation.

[info.zip](https://github.com/Seagate/openSeaChest/files/6238660/info.zip)

vonericsen, 2021-04-14:
Hi @EyeAndTea,
I've been working to get in contact with AMD to try and figure this out, but I need some more information.
Can you tell me more about the system hardware, like the AMD chipset/platform this is on?
If you are able to provide specific motherboard or other information, that would also be helpful.
If you are not sure, could you export the system information?
You can get this by hitting the start button and typing "system information" then you can export it with File->Export, or File->Save.

EyeAndTea, 2021-04-15:
I am happy to hear from you.

The chipset is AMD 990FX. The motherboard is ASUS Crosshair V Formula. More information can be found at: [link](https://www.guru3d.com/articles_pages/asus_crosshair_v_formula_review,3.html). The operating system is Windows 7 64bit.

To alleviate much suspicion relevant to debugging, I have confirmed this to be a problem with the Windows setup by compiling your tools on the Linux SystemRescue distribution and trying to make changes. (However, please keep this bug report open.)

vonericsen, 2021-04-15:
Thanks for the information @EyeAndTea!
I will share this with AMD so we can figure out what it going on with this setup.

Deepcuts, 2021-09-24:
No clue if this has been fixed for you, but check https://github.com/Seagate/openSeaChest/issues/41

vonericsen, 2021-09-24:
This issue is a little different as it seems to be the HBA driver blocking the EPC commands. #41 was a bug in the code itself, which is not responsible for this issue based on the verbose output that has been shared earlier.

I have not heard from AMD on how to resolve this. I have left this open as I think there is something still worth debugging before this is closed, but I have not been able to track down hardware that matches for some testing. My personal system at home is AMD, but the driver is much newer and doesn't seem to have this issue. I have tested some other AMD systems we have at the office, but they are all running Win10 and also have a newer driver installed and don't seem to have this issue.

My experience with different systems and drivers is that sometimes commands are blocked by design (for example Microsoft blocks Sanitize, ATA security unless in PE mode, libATA in Linux blocks TCG commands by default). Sometimes there is a way around this with another method of issuing the command, but this is not always possible or available.

One thing I would like to try is to issue the commands as ATA passthrough instead of SCSI passthrough. It is possible that this may be handled differently by the driver and may get around this, but it might still be filtered the same way. I can share what changes to the code are needed for testing this as there is not a built-in way to switch this from a compiled version of the tool today.

Last thing I would like to note is if this system is setup as "AHCI" or "SATA" mode in the BIOS, you can *probably* switch to the Microsoft AHCI driver that comes with Windows safely. If it is in RAID mode, this is much more complicated to change (you need to modify the registry otherwise you get BSODs when you boot). I have done this with systems in our office quite a few times to test Intel drivers vs built-in drivers, but have not tested it with AMD before, so it may be different...if you attempt this, make sure you have a system restore point set and a fresh system backup just in case.

EyeAndTea, 2021-12-21:
Thank you very much for the update.

If you already have the code changes, and only if you already have them, please do post them. I would like to keep the code changes for my reference. However, if you do not have the code changes already, then I do not wish to set you to work by a false promise.

(As an affirmation, I write that I still strongly like this bug report to be kept open. Per empathy, I also state that there would be no hard feeling if you choose to close it.)