Seagate/openSeaChest #53: Making the openSeaChest EFI binaries

achelon5, 2021-04-26:
Hi,

Apologies for the beginner questions, but could you give me some clues about setting up an appropriate build environment for building the EFI binaries for this project? A recommendation for an appropriate distro, distro version and supporting libraries that would mimic closely your own would be helpful.

Regards,
Richard

vonericsen, 2021-04-26:
Hi @achelon5,

I will work on fully documenting this for you, but I won't be able to do this until next week.
There are some "gotchas" to share about this too, both building and running the tools from our experience testing them.

vonericsen, 2021-05-07:
Hi @achelon5,

This week was way busier than I expected. I will get to this as soon as I can, hopefully next week. Sorry for the delay.

vonericsen, 2021-05-14:
Hi @achelon5,

I added documentation inside of BUILDING.md to discuss how to build the EFI tools.
Please let me know if this helps, or if more detail is necessary.
There are some known problems for UEFI based tools but I'm not sure where the best place to document those is yet. These issues are part of the reason the UEFI build is not being actively maintained. 

The biggest reason we stopped the UEFI build and development was lack of system compatibility. In our lab testing a lot of the systems we tested were missing different passthrough drivers, or the drivers appeared broken. Sometimes the screen output was super slow and eventually the systems would hang too. We did have varying degrees of success loading our own ATA or NVMe passthrough drivers, but some systems refused to load these or refused to unload the drivers that they included.

When I recently revisited this, I was in process of submitting the changes I made to edk2-libc when I noticed that the main edk2 repo removed support for unsafe string functions. This breaks the libc project which hasn't seen changes to deal with this....I'm unsure of the edk2-libc project's future. Once I get some time to dig in and figure out what I did wrong last time I tried submitting changes, I will submit them again, but for the process as of now, you need to pull an older stable edk2 tag and the forked edk2-libc repo like I have documented.

vonericsen, 2023-03-01:
Closing this due to inactivity.
We have not been maintaining the efi binaries lately due to so many system compatibility problems we ran into when we were testing them. I am happy to update that build process if the build is failing and the documentation if anyone sees this and needs help. Please feel free to reopen this issue and I will take another look at it.

closed 2023-03-01 by vonericsen