Seagate/openSeaChest #55: Cannot download nor update firmware at all

helpfirmy, 2021-06-09:
Hi . Anybody else suffer from this problem ? ST91000460NS . What am I not doing ?

vonericsen, 2021-06-09:
Hi @helpfirmy,
Can you share the command line you are using? And which OS you are attempting the update in?

Also, if you capture the verbose output to a file that can also help. 
Ex: `openSeaChest_Firmware -d /dev/driveHandle --downloadFW -v 3 > someFile.txt`

TheOddball, 2022-01-23:
This appears to be a dead issue, but I've been scratching my head with this problem for awhile.

I bought some IBM branded ST1600FM0013 drives (1.6Tb SED SSDs) on eBay for really cheap, but they're formatted in 528 bytes. I have successfully reformatted some HGST drives to 512 with sg_format, but these drives lock out the ability to change block size. Seeing this is most likely a firmware issue, I have resorted to changing the firmware back to the OEM Seagate firmware. I've tried sg_write_buffer and openSeaChestFirmware to no avail. Putting this here to see if the devs have any idea, since you guys have more experience with Seagate stuff than I do.

I've gotten past Step 1: Revert the SED. Also; the drive appears to be accepting the firmware, just preventing me from flashing it.

Here's a `-v 4` output of a *STILL SED LOCKED* SSD [SED.txt](https://github.com/Seagate/openSeaChest/files/7919674/SED.txt)
And also a `-v 4` output of an unlocked SSD [Unlocked.txt](https://github.com/Seagate/openSeaChest/files/7919673/Unlocked.txt)

I'm using `Nytro1200_2-KohoSSD-STD-000A.LOD` to flash them, with command `./SeaChest_Firmware_x86_64-redhat-linux -d /dev/sg55 --downloadFW ../Nytro1200_2-KohoSSD-STD-000A.LOD --fwdlSegSize 4k -v 4`

vonericsen, 2022-01-24:
HI @TheOddball,

Seagate drives do not allow switching between OEM and customer firmwares or visa-versa.

It is possible that if the drive was SED locked, that was the reason you were unable to format to a different sector size. Also, a lot of times IBM drives ship with PI 2 formatting, so it may also be that needs to be part of the format command as well.

If you use `openSeaChest_Format -d /your/device --formatUnit 512 --protectionType 0 --confirm ...` would be how to remove PI and go to a "standard" sector size. (Add `--poll` to see progress until fully complete).
I know customer firmware can have specific requirements about what is or is not allowed, but it may be worth trying again since you did the revertSP. That will remove any security locks and put the drive back to "factory" state which would mean no active security and no commands would be blocked by the security feature of the drive.
If this works, you will need to do revertSP before the format for the others too.

TheOddball, 2022-01-24:
Unfortunately, the IBM firmware does not allow you to change the sector size..
[Here's a `-v 4` output of the format command you suggested](https://github.com/Seagate/openSeaChest/files/7928453/format.txt)

vonericsen, 2022-01-24:
Thanks for the verbose output.
This is very similar case to what I looked up on Seagate/ToolBin#22 with another product with IBM firmware.
I checked with the firmware lead and they said it is fairly common that customer unique code have specific set sector sizes (among other requirements).
I'm going to add a warning/note in the `--showSupportedFormats` output to help with this. Unfortunately we do not have a way to figure this out automatically right now, but I'm trying to figure out if there is a better way to detect this for the future.
Sorry I do not have a way to get to the format you need.

TheOddball, 2022-01-24:
That's highly unfortunate. Do you think I could send you an email about this situation? I have some questions that don't exactly relate to this project.

vonericsen, 2022-01-24:
Would you be willing to open a discussion, either here or on Seagate/ToolBin?
Discussions can be useful for questions that may not directly relate to the software project and can be useful to helping others with similar issues/questions about drives, among other things. I enabled them on both projects to be a little more open/free than the issues which tend to relate to something more specific about the software.

vonericsen, 2023-03-01:
I am closing this because we now have additional warnings that customer unique code may not work the same as Seagate's OEM code due to the requirements from those customers.
It is also not possible to switch from a customer code to OEM code or visa-versa on Seagate products.
Please reopen this or create new issues and I am happy to look into them and help you as best I can whether you have OEM code or a customer unique code.

closed 2023-03-01 by vonericsen