Seagate/openSeaChest #57: Do not assume that Linux is using LGPL GNU libc

xbjfk, 2021-07-12:
https://github.com/Seagate/openSeaChest/blob/7e4e2d15fb951a5e326a9b213ab9f96a08685bbf/src/EULA.c#L415-L417
Linux does not necessarily use GNU libc. For example, MIT-license [musl](https://www.musl-libc.org/) could be used instead. I propose to check if `__GLIBC__` is defined instead of `__linux__`.

vonericsen, 2021-07-12:
Hi @xbjfk,
Thanks for creating this issue. I've been meaning to update and investigate compiling with musl for a while now.

Using the glibc macro seems reasonable. I will do some additional research and see what looks like the best solution since the tool should display the MIT license for musl when it is used instead of glibc.

vonericsen, 2021-07-14:
Hi @xbjfk,

That seems like a good reason to only check glibc. I have seen the musl macro discussions in the past, so it was going to be some kind of `#if glibc ... #else...` approach anyways. I was searching to see if there were any other likely libc's that would be encountered...it looks like it is not likely that anything else will come up...and if it does, that can be a new issue 😄

When the linux macro check was first added, it was because one of our customers used the `strings` tool to search for `lgpl` and `gpl` and were not wanting to find those licenses anywhere in the FreeBSD build, so the solution will likely keep the Linux macro for now and add a second layer below for detecting libs to preserve this functionality so that the MIT license isn't shown when it isn't used in one of the other OSs that openSeaChest supports.

Something like:
```
#if defined (__linux__)
     #if defined (__glibc__)
           //show lgpl
     #else      
          //show mit for musl
     #endif     
#endif
```

I'm also considering a way to add a preproccessor definition, such as `LIBC_MIT` or `LIBC_LGPL` to force a specific license to allow for setting the correct license in other odd builds where autodetect looks off.

I'm happy to hear other ideas, but want to make sure that the licenses shown in each build of openSeaChest is reasonably accurate. I have a contact in Seagate I can verify legal questions with if I need to since I'm not a legal expert either 😄

xbjfk, 2021-07-14:
I personally do not think that assuming that non-glibc linux systems are using musl. I'd say just to print nothing if compiled with linux but not glibc, because the binary releases are using glibc, and I do think that it is the responsibility of the person compiling the code to ensure they fall under the libc's license

vonericsen, 2021-07-15:
Hi @xbjfk,

I did some testing of the commit I pushed yesterday in a few machines that I have and an Alpine linux docker container and it works.
A preprocessor flag can be added to the makefile for musl (-DUSING_MUSL_LIBC), but others are not defined at this time. Other libc support can be added in the future as needed.

Let me know if this solves this, or if you think I need to tweak it some more first.

xbjfk, 2021-07-16:
I think is the best solution, Thank you!

closed 2021-07-16 by xbjfk

vonericsen, 2021-07-16:
Excellent 😄 

I merged it into our develop branch this morning.
The last thing I added yesterday allows for making this even easier:
`make release USING_MUSL=1`