Seagate/openSeaChest #58: build failure: undeclared identifier 'BLOCK_SIZE'

swills, 2021-08-30:
On FreeBSD, I see build failure:

cc -O2 -pipe -Werror=date-time -march=broadwell  -Wall -c -std=gnu99 -fstack-protector-strong -fno-strict-aliasing  -fdebug-prefix-map=/wrkdirs/usr/ports/sysutils/openseachest/work/openSeaChest-21.06.21=. -O3 -I../../opensea-common/include -I../../opensea-transport/include -I../../opensea-transport/include/vendor -I../../include -I../../opensea-operations/include -DDISABLE_TCG_SUPPORT ../../utils/C/openSeaChest/openSeaChest_NVMe.c -o ../../utils/C/openSeaChest/openSeaChest_NVMe.o
../../utils/C/openSeaChest/openSeaChest_NVMe.c:1279:33: error: use of undeclared identifier 'BLOCK_SIZE'
                uint32_t size = BLOCK_SIZE;
                                ^
../../utils/C/openSeaChest/openSeaChest_NVMe.c:1300:31: error: use of undeclared identifier 'BLOCK_SIZE'
                    offset += BLOCK_SIZE;
                              ^
../../utils/C/openSeaChest/openSeaChest_NVMe.c:1314:49: error: use of undeclared identifier 'BLOCK_SIZE'
                            fullSize = offset + BLOCK_SIZE * teleHdr->teleDataArea1;
                                                ^
../../utils/C/openSeaChest/openSeaChest_NVMe.c:1319:49: error: use of undeclared identifier 'BLOCK_SIZE'
                            fullSize = offset + BLOCK_SIZE * teleHdr->teleDataArea2;
                                                ^
../../utils/C/openSeaChest/openSeaChest_NVMe.c:1324:49: error: use of undeclared identifier 'BLOCK_SIZE'
                            fullSize = offset + BLOCK_SIZE * teleHdr->teleDataArea3;
                                                ^
../../utils/C/openSeaChest/openSeaChest_NVMe.c:1344:43: error: use of undeclared identifier 'BLOCK_SIZE'
                                offset += BLOCK_SIZE;
                                          ^
../../utils/C/openSeaChest/openSeaChest_NVMe.c:1350:157: error: use of undeclared identifier 'BLOCK_SIZE'
                                        printf("Error: Could not retrieve Log Page %" PRIu8 " for offset %" PRIu64 "\n", GET_TELEMETRY_IDENTIFIER, offset - BLOCK_SIZE);
                                                                                                                                                            ^
../../utils/C/openSeaChest/openSeaChest_NVMe.c:1382:47: error: use of undeclared identifier 'BLOCK_SIZE'
                                    offset += BLOCK_SIZE;
                                              ^
../../utils/C/openSeaChest/openSeaChest_NVMe.c:1388:161: error: use of undeclared identifier 'BLOCK_SIZE'
                                            printf("Error: Could not retrieve Log Page %" PRIu8 " for offset %" PRIu64 "\n", GET_TELEMETRY_IDENTIFIER, offset - BLOCK_SIZE);
                                                                                                                                                                ^
9 errors generated.

swills, 2021-08-30:
Perhaps related to: https://github.com/Seagate/openSeaChest/commit/f4c7627f50032b3dcda3ee4f5db464e35207ded3

vonericsen, 2021-08-30:
Hi @swills,
This does seem like it could be related, but I would only expect this error if the submodules were pulled at develop rather than the referenced commit when the release tag was created.

I will pull this into a FreeBSD machine and verify this behavior.

swills, 2021-08-30:
Ah, you're right, that's exactly what I did... Thanks for the clue! Sorry for the noise... Closing.

closed 2021-08-30 by swills

vonericsen, 2021-08-30:
No problem. Glad it was simple 😄