Seagate/openSeaChest #59: openSeaChest_Configure fail to set SAS Mode Page

ThunderEX, 2021-09-19:
Hi,
When I use openSeaChest_Configure to set SAS Mode Page, for example WCE on page 08h:
```Shell
[root@localhost ~]# ./openSeaChest_Configure -d /dev/sdp --showSCSIMP 08

/dev/sg15 - ST1200MM0129 - WFK49Q0G0000K94575NM - SCSI

===========================================================
 Page 8h Caching
 Current Values
===========================================================
88 12 14 00 FF FF 00 00 FF FF FF FF 91 20 00 00 00 00 00 00

[root@localhost ~]# ./openSeaChest_Configure -d /dev/sdp --setSCSIMP 08:2:2:1=0

/dev/sg15 - ST1200MM0129 - WFK49Q0G0000K94575NM - SCSI
*** Error in `/root/seachest_exes_gcc/SeaChest_Configure_x86_64-redhat-linux': free(): invalid next size (fast): 0x000000000255e010 ***
```
But it works with `sdparm`:
```Shell
[root@localhost ~]# sdparm --page=08h --hex /dev/sdp
Caching (SBC) [0x8] mode page:
    Current:
 00     88 12 14 00 ff ff 00 00  ff ff ff ff 91 20 00 00
 10     00 00 00 00
    Changeable:
 00     88 12 a5 00 00 00 ff ff  ff ff 00 00 21 00 00 00
 10     00 00 00 00
    Default:
 00     88 12 10 00 ff ff 00 00  ff ff ff ff 90 20 00 00
 10     00 00 00 00
    Saved:
 00     88 12 14 00 ff ff 00 00  ff ff ff ff 91 20 00 00
 10     00 00 00 00
[root@localhost ~]# sdparm --page=08h --set=2:2:1=0 --save /dev/sdp
    /dev/sdp: SEAGATE   ST1200MM0129      C006
[root@localhost ~]# sdparm --page=08h --hex /dev/sdp
Caching (SBC) [0x8] mode page:
    Current:
 00     88 12 10 00 ff ff 00 00  ff ff ff ff 91 20 00 00
 10     00 00 00 00
    Changeable:
 00     88 12 a5 00 00 00 ff ff  ff ff 00 00 21 00 00 00
 10     00 00 00 00
    Default:
 00     88 12 10 00 ff ff 00 00  ff ff ff ff 90 20 00 00
 10     00 00 00 00
    Saved:
 00     88 12 10 00 ff ff 00 00  ff ff ff ff 91 20 00 00
 10     00 00 00 00
```
Could you please have a look? Thanks!

vonericsen, 2021-09-20:
Hi @ThunderEX,
We will take a look and figure out what happened. 

Can you try it with a file as the input as well?
Your file would be a text file (`wceMP.txt`) that looks like this (change any field as you see fit):
`88 12 14 00 ff ff 00 00  ff ff ff ff 91 20 00 00`

The openSeaChest command would be `openSeaChest_Configure -d /dev/yourDeviceHandle --setSCSIMP file=wceMP.txt`

ThunderEX, 2021-09-21:
Hi @vonericsen 
It works:
```Shell
[root@localhost tmp]# echo "88 12 14 00 ff ff 00 00 ff ff ff ff 91 20 00 00" > wceMP.txt
[root@localhost tmp]# ~/dev/openSeaChest/Make/gcc/openseachest_exes/openSeaChest_Configure -d /dev/sdb --setSCSIMP file=wceMP.txt
==========================================================================================
 openSeaChest_Configure - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2021 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Configure Version: 2.0.2-2_2_4 X86_64
 Build Date: Sep 19 2021
 Today: Tue Sep 21 10:54:47 2021        User: root
==========================================================================================

/dev/sg2 - ST1200MM0129 - WFK47SPX0000E9454C89 - SCSI
Successfully set SCSI mode page!

[root@localhost tmp]# sdparm --page=08h --hex /dev/sdb
Caching (SBC) [0x8] mode page:
    Current:
 00     88 12 14 00 ff ff 00 00  ff ff ff ff 91 20 00 00
 10     00 00 00 00
    Changeable:
 00     88 12 a5 00 00 00 ff ff  ff ff 00 00 21 00 00 00
 10     00 00 00 00
    Default:
 00     88 12 10 00 ff ff 00 00  ff ff ff ff 90 20 00 00
 10     00 00 00 00
    Saved:
 00     88 12 14 00 ff ff 00 00  ff ff ff ff 91 20 00 00
 10     00 00 00 00
```

vonericsen, 2021-09-21:
@ThunderEX,
Please reopen this issue if you see any other problems, but the commit I pushed resolved the problem in my testing.

Thanks for reporting this issue!

closed 2021-09-21 by vonericsen

ThunderEX, 2021-09-22:
It works! Thanks@vonericsen !

commit 27fa22ad (2021-09-21): bug: Fixing a bug in the --setSCSIMP option that caused a crash

Fixing a bug that would crash openSeaChest_Configure inside the --setSCSIMP option.
The issue was that the function scsi_Set_Mode_Page takes a buffer that should contain ONLY the mode page data, no headers or data block descriptors, but in the case of providing the byte offset, bit, value to make a change, it was providing the header which caused a mismatch in what data was supposed to be sent to the drive and caused a heap corruption.

The solution was to take into account the differences in how this data was read from the funct [... cut]