Seagate/openSeaChest #62: Win32 binary

Beliathal, 2021-11-26:
Win32 binary release won't run on Windows 2003 R2 Server (non valid Win32 app error), it's either a 64 bit app or is not compatible with legacy OSes?

vonericsen, 2021-11-29:
Hi @Beliathal,
Windows server 2003 (and Windows XP) or older are not supported by the builds we make.
The oldest OS we expect to work is Vista or Server 2008.
As I recall, the format of Windows executables changed with Vista which is why you see that kind of message regardless of the executable being 32bit or 64bit.

The code _might_ work if compiled for these old systems, but there will be definite loss of functionality due to bug in the ATA passthrough in these old operating systems. I do not know if there are issues with SCSI passthrough in these old systems...I will have to check with some other people, but ATA passthrough definitely does not work with transfers greater than 512B (it has to do with when the interrupt between data blocks occurs and XP stops at the first one regardless of your request).
These were some of the reasons we did not support these older OSs when we first created this toolset internally.

The last time I tried setting up the xp compatibility compiler, it required a lot more changes than I wanted to make to get it to compile that may impact the functionality on modern systems and I did not think that tradeoff was worth it at that time.
The mingw/msys2 compile process might work better for this. 
Unfortunately I don't even have a system I could test this on at this point.

Beliathal, 2021-11-29:
Thank you so much for replying and clearing this up

closed 2021-11-29 by Beliathal