Seagate/openSeaChest #67: Proprietary SeaChest bootable does not allow erasing non-Seagate disks

Callistemon, 2022-01-14:
I have a Western Digital Black WD6003FZBX that fell-apart after 5 weeks, since Western Digital has become a scam company. It cannot be recognized by any tool in Windows, Linux, or FreeBSD. It does appear in the firmware settings of a computer with Legacy BIOS. SeaTools Bootable with the proprietary version of SeaChest is the only thing able to recognize the WD6003FZBX besides the BIOS on that computer, but it cannot erase the HDD because it is non-Seagate. The WD Black is connected directly with SATA.

Is there a flag to bypass this? Why is this restriction in place? I don't care if it's bricked by not being compatible with Western Digital's proprietary secrets, because it is already bricked and absolutely nothing else is able to recognize it anyway. It needs to be formatted before sending it in for warranty, since it has terabytes of personal files.

Callistemon, 2022-01-14:
This repository is for the open-source version, but the pre-sales contact is not relevant, and the technical support contact requires a Seagate serial number. If this issue is fixed in the open-source version, despite being originally experienced in the proprietary version, that would suffice (provided the open-source version doesn't require `make` to create a bootable device).

vonericsen, 2022-01-14:
Hi @Callistemon,

I'm sorry to hear you are having bad luck and going through this frustration! I have been through something similar before and it very upsetting...I'll do my best to help you with what I can.

The closed source SeaChest restricts certain option to only run on Seagate drives and does not have a method to bypass that check.
The openSeaChest tools have no such restrictions, other than device capabilities (if a specific feature is supported or not by the device).
The restrictions have been around since SeaTools for Windows was developed and were due to concerns about supporting non-Seagate products. I have sent this issue to the group where we have discussed removing this check to talk about this again ahead of our upcoming release (not yet scheduled).

This repo does not have a bootable USB for openSeaChest in it, but If you have already created a SeaChest bootable USB, you can copy the openSeaChest files onto it and run them. 

Once you have booted the USB (Tinycore linux), you can `cd` to the USB's folder path where the files were copied, which will be something like this: `/mnt/sdc1`. You may have multiple directories similar to this depending on what other storage devices are also attached to the  system. 
If you do `ls` of each of these you can find it by seeing which has various SeaChest pdf and txt help files on it (this is what I did), but you can also use the `blkid` command to look for which device has the `LABEL="SEABOOT"` as well...just note that `blkid` uses `/dev/sdc1` in the output and you will need to cd to `/mnt/sdc1` or whatever the USB shows up as.

If you already extracted the .tar.gz, you may need to set the execute bit (the fat32 partitioning does not preserve this) with `sudo chmod +x ./openSeaChest*`. I did not in my testing, but I have needed this in the past.
Now you should be able to run the tools from this directory using `./` before each tool name...tab completion should work to save typing the full name and helps verify that the system knows the tools are there and can be executed.
At this point you should be able to run whatever operation you want from the tools provided the drive supports them and is responding correctly.

I want to also provide some caution when working with a troublesome or sick drive like it sounds you have. Sometimes the drive will not be able to write because they are in such bad shape, but sometimes they will work, and sometimes the drive will partially work...it depends on the failure type it has had. 

Let me know if there is something else I can do to help.

Callistemon, 2022-01-15:
Thank you so much for your fast replies and thorough understanding. I have been really impressed with Seagate lately and have every reason to switch. Turns out the reason the WD Black is recognized by SeaChest Bootable is simply because I connected it with direct SATA while displaying the F12 boot menu. Previously, it was never possible to recognize it when the WD Black was inserted after logging into Linux. It wasn't possible to connect the WD Black before starting Linux, as it prevents the BIOS from continuing past the POST. Once it was recognized in Linux, it was still impossible to start a secure erase or even benchmark because of Input/Output Errors. HDDSuperClone was able to start shredding it at 512 KB/s and has cleared 4 gigabytes so far. If it doesn't speed up, I might try a secure erase with openSeaChest, however, even hdparm wasn't able to start one after Linux recognized it.

I am completely done with Western Digital, which has become a scam company that produces trash nothing like their products from years ago, and have respect for both the developer of HDDSuperClone and Seagate for assisting with this bait-and-switch scam disaster. I also have a new SanDisk memory card reader that no longer functions, and another WD Black with I/O Device Errors. There has been a large increase in complaints about I/O Device Errors with Western Digital products, while the negative reviews on the latest Seagate models have significantly reduced since 2018. Western Digital has switched to packaging HDDs in cheap Styrofoam, which is impossible to recycle and is very poor at cushioning. Seagate uses mostly paper-based material for packaging HDDs.

Western Digital is too cheap and negligent to fix severe vulnerabilities in My Cloud OS 3, and will instead leave customers vulnerable for several more months before crippling functionality with a ransom to purchase a new WD My Cloud. Each one of your quality responses adds a level of reassurance that vulnerabilities in Seagate products will be quickly fixed with competence. I apologize for purchasing Western Digital. That was totally wrong and based on outdated information.

vonericsen, 2022-01-19:
@Callistemon were you able to get your erase to complete? Or need anything else to run openSeaChest to erase it?

I brought up the filter for Seagate drives and after some discussion, it is planned to be less restrictive in our next release (TBD), so it will be possible to erase non-Seagate drives in the future. Thanks for bringing this use-case to our attention!

Callistemon, 2022-01-19:
It has had over 512 GB cleared by HDDSuperClone, which is working at 1550 KB/s. If it continues increasing at the current rate, as long as the WD Black doesn't further crumble, it will finish in about 9 days. It is impossibly successful so far, but very slow.

`hdparm --sanitize-block-erase` and `hdparm sanitize-overwrite` are unable to start because of an Input/Output Error. Would `openSeaChest_Erase --sanitize overwrite`, `openSeaChest_Erase --ataSecureErase enhanced` or `openSeaChest_Erase --overwrite` be any different? If those work, would they be any faster than HDDSuperClone Erase Destination?

vonericsen, 2022-01-19:
I think it will depend on exactly what the input/output error is.

The drive could be struggling and having lots of retries slowing it down, no matter which method is used, and if it has run out of room to automatically reallocate defects, then sanitize or ATA secure erase could report similar failures if the data cannot be erased.
Sanitize and ATA security enhanced erase could be faster since they are erases entirely run by the drive firmware (including writing to original locations of reallocated sectors).
Both Sanitize and ATA secure erase will have to start over at the beginning of the drive when they run though.
You will also need to make sure the drive supports those features. (`openSeaChest_Erase -d /dev/sg? --showEraseSupport`)

The overwrite option in openSeaChest can be given a specific starting, and even ending, location to overwrite, but I suspect that it will be about the same as the current tool is running. Most tools to erase work similar to this overwrite option and issue command after command to perform the erase.

Callistemon, 2022-01-20:
After 5 weeks, it had a Current Pending Sector Count of 8 and an Uncorrectable Sector Count of 1, and failed the short self-test with Error Code 7. The pending sectors increased to 16, and then it started clashing and grinding doing nothing. Although anything above 0 is unacceptable for those attributes, that doesn't seem like enough to run out of spares. SMART attributes cannot be read even by the same HDDSuperClone, and temperature cannot be retrieved with `hdparm -H /dev/sdb` even though the command supported this model before it fell-apart.

vonericsen, 2022-08-02:
The USB bootable version of SeaChest was updated on seagate.com yesterday to a new version that includes the changes to remove restrictions on non-Seagate drives like the commit I pushed for openSeaChest earlier.
This update took way longer than I wanted it to, but it is finally done.
I'm closing this since this issue is now resolved, but please feel free to open more issues or reopen this if you think it is necessary to revisit this.

closed 2022-08-02 by vonericsen