Seagate/openSeaChest #68: Activating && or || depending on short DST result with different Error Levels for failing disks and empty slots

Callistemon, 2022-01-22:
I want to start automatic short self-tests every week, and launch a warning message with `&&` or `||` if there are errors. At first, I was looking at `smartctl`, but that always returns an exit code of 0 as soon as the test begins, even if it completes with errors. It appeared `openSeaChest` provided this functionality, as there is a long list of exit codes including 3 for Operation Failure, but only `--shortGeneric` waits to send an exit code until finished, while `--shortDST` always activates `&&` after starting. There should be a `--shortDST` option that waits to send an exit code until completion.

Both `--shortDST` and `--shortGeneric` activate `||` if a disk with the specified number is not connected. It's not practical to have failure warnings appear when a disk doesn't happen to be available at a specified number (such as PD4, which may be empty most of the time). I understand `||` is activated by anything above 0, so it will be activated even if the Error Code for failing and unavailable disks is different, but `echo %ErrorLevel%` shows that 3 is the Error Code for an empty slot, so it is the same Error Code as when disks fail with errors, assuming Error Code 3 is even used for failed tests.

vonericsen, 2022-01-24:
Short generic is a read test so it has to wait until completion since the software is managing the entire thing.
Short DST is a drive managed test and the software starts it and can optionally poll it for progress. If you run `--shortDST --poll` it will wait until the test has completed and then return the exit code based off of the full operation.

I don't think openSeaChest has a way to detect a difference between "drive not found" and "failure". I'll look into it and see if that is something that can be added.

Callistemon, 2022-01-31:
> Short DST is a drive managed test and the software starts it and can optionally poll it for progress. If you run --shortDST --poll it will wait until the test has completed and then return the exit code based off of the full operation.

Great! That's exactly what I wanted.

> I don't think openSeaChest has a way to detect a difference between "drive not found" and "failure". I'll look into it and see if that is something that can be added.

I just looked at the list of Error Levels again, and an unavailable handle is supposed to be Exit Code 2. However, it is 3, which is for operation failure (in this case, short DST failure).

vonericsen, 2025-05-30:
Fixed in [v25.05](https://github.com/Seagate/openSeaChest/releases/tag/v25.05)

closed 2025-05-30 by vonericsen