Seagate/openSeaChest #69: Pull upstream change of wingetopt

ThunderEX, 2022-02-05:
Hi,

When I tried to build on MSVC with meson, it comes up with a linker warning that both debug and release runtime are linked: `warning LNK4098: defaultlib 'MSVCRTD' conflicts with use of other libs; use /NODEFAULTLIB:library`. This could also be seen in your GitHub Actions: https://github.com/Seagate/openSeaChest/runs/4744648956?check_suite_focus=true

I guess it may due to release / debug build type is not correctly passed from Meson to CMake. So I made a workaround to meson.build like:
```
if target_machine.system() == 'windows'
  if not c.has_header('getopt.h')
    cmake = import('cmake')

    opt_var = cmake.subproject_options()
    if get_option('debug')
      opt_var.add_cmake_defines({'CMAKE_BUILD_TYPE': 'Debug'})
    else
      opt_var.add_cmake_defines({'CMAKE_BUILD_TYPE': 'Release'})
    endif
    wingetopt = cmake.subproject('wingetopt', options: opt_var)
    wingetopt_dep = wingetopt.dependency('wingetopt')
    os_deps += [wingetopt_dep]
  endif
  windows = import('windows')
  resources = windows.compile_resources('openSeaChest.rc')
  common_sources += [resources]
  add_project_arguments('-D_CRT_SECURE_NO_WARNINGS', language : 'c')
endif
```
And it works.

Furthermore, I could see that, the upstream wingetopt has updated recently to support Meson: https://github.com/alex85k/wingetopt/pull/3

So I would suggest to pull upstream changes of wingetopt, and thus all subprojects will be all Meson, and also resolve the problem above by the way.

vonericsen, 2022-02-07:
Hi @ThunderEX,

Thank you for this information!
We will work on pulling in those upstream wingetopt changes since that seems like a great update to pull in!

pranali-tirkhunde, 2022-02-09:
Hi @vonericsen,
I have added the latest changes from the upstream branch into Seagate/wingetopt branch. Can you please verify once?

closed 2022-02-10 by ThunderEX