Seagate/openSeaChest #76: openSeaChest_Logs: FARM data in stdout

yaarith, 2022-03-16:
It would be nice if we can pipe the FARM data from openSeaChest_Logs to openSeaChest_LogParser and not go through a file on disk (i.e. `openSeaChest_Logs --farm -o - | openSeaChest_LogParser ...`).
Currently, the openSeaChest_Logs saves the data to a file. That takes a little logic to find what the file name is (need the serial number and the time the log was generated), it's more automation friendly to allow piping.

vonericsen, 2022-03-16:
Hi @yaarith,
The current method is due to how we've interacted with different customers in the past and had binaries shared directly back to Seagate. This has been changing over time and we've begun parsing more in the log parser. They are separate processes due to requirements we've had from data centers in the past.
I think this is another good use-case to support though.
I'll check our project schedule to see where we can fit this in.

yaarith, 2022-03-24:
Thanks for clarifying, @vonericsen.
When can we expect this feature?

Rungthida, 2022-03-30:
Hi @yaarith,
It would take a few weeks.

Rungthida, 2022-05-26:
Hello @dmick,
We've started looking into it but it has been delayed due to other projects that we're working on in parallel. Hopefully, this should be ready for testing in a couple weeks.

Rungthida, 2022-08-24:
Hello @yaarith & @dmick,
I'm so sorry it took so long. We have had a lot of internal customer request. However, the new code has been pushed to the develop branch for both openSeaChest_Logs and openSeaChest_LogsParser. We also added a bash script for Linux in"example" folder. 

Example:
./openSeaChest_Logs -d /dev/sdb --farm --logMode **pipe** | ./openSeaChest_LogParser_x86_64 --inputLog **fromPipe** --logType farmLog --printType json --outputLog testLog.jsn

yaarith, 2022-08-25:
Thanks for the update, @Rungthida.
I do not see these commits in either repos under the `develop` branch, when were they merged?

vonericsen, 2022-08-25:
Hi @yaarith,
I have merged @Rungthida's code into develop. We had a miscommunication, but it is now on develop in both openSeaChest and openSeaChest_LogParser.
Let us know if you have any problem using this.

vonericsen, 2023-03-01:
Closing this as the code has been in develop and has now been released in https://github.com/Seagate/openSeaChest/releases/tag/v23.03
Please feel free to reopen this issue or create a new issue if you need any other help!

closed 2023-03-01 by vonericsen