Seagate/openSeaChest #85: Feature Read Look-Ahead detection incorrect for SCSI drives

Jeetandra, 2022-05-14:
Read Look-Ahead detection is tied to detecting if DRA bit (cache mode-page) is changeable. If DRA (Disable Read Ahead) is changeable, openSeaChest incorrectly reports Read Look-Ahead is not supported.

Following logic needs correction:
```
drive_info.c
diff --git a/src/drive_info.c b/src/drive_info.c`
index 2cb972e..1327c7d 100644
--- a/src/drive_info.c
+++ b/src/drive_info.c
@@ -3954,8 +3954,9 @@ int get_SCSI_Drive_Information(tDevice *device, ptrDriveInformationSAS_Sata driv
              }
              if (pageRead)
              {
+                                          // changeable implies feature supported.
 driveInfo->writeCacheSupported = cachingPage[headerLength + 2] & BIT2 ? true : false;
-                                          driveInfo->readLookAheadSupported = cachingPage[headerLength + 12] & BIT5 ? false : true;
+                                          driveInfo->readLookAheadSupported = cachingPage[headerLength + 12] & BIT5 ? true : false;
              }
      }
}
```

vonericsen, 2022-05-16:
@Jeetandra,
Thanks for reporting this!
I have pushed a fix for this on the develop branch. 

Please reopen this if you think more changes are required.

closed 2022-05-16 by vonericsen

Jeetandra, 2022-05-17:
thanks! @vonericsen. Fix looks good.