Seagate/openSeaChest #87: This project can send a usb-PCIE command  ?such as a write command

MagaretJi, 2022-07-05:
[image]

vonericsen, 2022-07-05:
Hi @MagaretJi,
Assuming you mean to communicate with a M.2 NVMe drive on an adapter, this project should be able to do that.
USB uses SCSI block device commands to communicate for reads and writes, so SCSI read16 or SCSI write16 should be able to read and write data. The scsi to nvme translation whitepaper from nvmexpress.org is usually what is implemented, but some translations may be missing in some cases (depends on the chip and firmware level of support provided). Generally reads, writes, inquiry, request sense, and test unit ready SCSI translations will be implemented as those are needed for basic communication with the OS. This software can communicate with all of these commands.

For more specific NVMe commands there is some support for some adapters from ASMedia and JMicron to passthrough specific NVMe commands, but the level of support depends on the chip being used in this case.

vonericsen, 2023-03-01:
I am closing this due to inactivity and no response after my last comment.
Please feel free to reopen this or create a new issue if you need help with anything else!

closed 2023-03-01 by vonericsen