Seagate/openSeaChest #88: Please remove retired manuals from release packages

MrM40, 2022-07-21:
E.g. release v22.06, package "openSeaChest-refs-tags-v22.06-linux-x86_64.tar.xz" the included manual "openSeaChest_PowerControl.8" contain old commands e.g. "--changePower"  which is retired.

It's better not to include manuals at all if they are not correct/not updated.

Wrong/incorrect data is worse than no data at all.

vonericsen, 2022-07-21:
Hi @MrM40,

Thanks for reporting this!
I pushed a commit to update them. I have been working on some bug fixes to this release that are nearly wrapped up so this will be pulled into that update.

vonericsen, 2022-07-29:
I've made the release v22.07 which includes new man pages in it.
If these look good to you, please either close this or let me know and I will close this issue.

vonericsen, 2023-03-01:
This issue was corrected in the previous release and I made sure to generate new manual pages for the new https://github.com/Seagate/openSeaChest/releases/tag/v23.03 release as well.
I'm closing this since it is resolved, but please feel free to reopen this if you run into any problems or create a new issue if you need help with something else!

closed 2023-03-01 by vonericsen