Seagate/openSeaChest #89: Encryption & PSID

Xanth0k1d, 2022-08-31:
I bought a few st1000lm035 hard drives, I noticed that they all have PSIDs, but with deviceInfo, I realised that encryption is not supported.
Why is this so? Can I switch to a firmware supporting encryption or are these drives unable to support encryption?(If so, then why do they all have PSID?)

vonericsen, 2022-09-01:
Hi @Xanth0k1d,

All Seagate drives come with PSIDs for a many years now as all Seagate products come with SD&D (Secure Downloads and Diagnostics) enabled on all firmware's to provide security better security for every drive made. This blocks access to the internal diagnostics outside of Seagate and adds more validation to make sure any firmware is authentic from Seagate so that it is not possible to flash a non-Seagate firmware to the drive.

This document has some more information about the [SD&D feature](https://www.seagategov.com/seagategov-content/product/maximize-security-lock-down-hard-drive-firmware-with-seagate-secure-download&diagnostics.pdf).

Xanth0k1d, 2022-09-02:
Hi @vonericsen ,
Thank you for the information. 
The Seagate brochure for that family of drives also indicates that ST1000LM035 is the base model and not an SED.
Can it be said that the PSID on the drive label had nothing to do with drive content encryption?

vonericsen, 2022-09-07:
> Can it be said that the PSID on the drive label had nothing to do with drive content encryption?

For Seagate drives, this is a correct statement. I'm not sure if this would apply to other vendors.

closed 2022-09-07 by Xanth0k1d