Seagate/openSeaChest #90: Concurrent Operations on Multiple Drives

Xanth0k1d, 2022-09-03:
Hi,
With the following command
Seachest_xxx -d device1 -d device 2 
It is possible to conduct the same operation on 2 drives in the sequence
Is there a way to conduct the same operation for both devices concurrently?

vonericsen, 2022-09-07:
Hi @Xanth0k1d,

There is not currently support for doing this in parallel from the command line like this.
When we've been asked about this in the past, we have recommended using different scripting methods to handle the parallel execution.
I'm not sure if that documentation is in the repo right now, but I have been wanting to expand documentation anyways and can look into adding it somewhere.

The methods we've recommended vary depending on the OS and shell a bit, but we've used python, xargs, and gnu parallel among others to accomplish this.

Here are a couple examples:

**GNU Parallel**
`sudo parallel -k 'time ./openSeaChest_SMART -d /dev/sg{} --shortDST --poll' ::: 0 1 2 3 `

**Xargs**
`/dev/sg0 /dev/sg1 /dev/sg2 /dev/sg3 | xargs -t -n 1 -P0 sudo ./openSeaChest_SMART --shortDST --poll -d `

These are among the shortest examples to make this work in a Linux shell today (you may need to install these tools though).

rudock1, 2022-09-07:
Hi Tyler, I've been meaning to say hi!  

Just a quick note to mention the concurrent testing examples folder we have over in the ToolBin
github  Seagate/ToolBin/SeaChest/parallel_testing/

thx
Billy

Xanth0k1d, 2022-09-07:
Thanks, was thinking of using scripts as well!

closed 2022-09-07 by Xanth0k1d

vonericsen, 2022-09-07:
> Hi Tyler, I've been meaning to say hi!
> 
> Just a quick note to mention the concurrent testing examples folder we have over in the ToolBin github Seagate/ToolBin/SeaChest/parallel_testing/
> 
> thx Billy

It's good to hear from you Billy!
Thank you for pointing this out. I forgot that there were some excellent readme's on this in that repo!

Here is a more direct link in case someone needs it quicker:
https://github.com/Seagate/ToolBin/tree/master/SeaChest/parallel_testing