Seagate/openSeaChest #91: Why not using static compilation for Windows builds?

jeremywaguet, 2022-10-08:
Released Windows binaries are dynamically linked to the VC runtime. Why not compile it statically so it can easily be integrated into the Windows PE environment without having to recompile from sources? An unzipped DLL build is approximately 9Mb in size, compared to 11.8Mb for a static build (using VC2022). I suppose this SeaChest utility suite is used mainly in preboot environments, right?

vonericsen, 2022-10-10:
Hi @jeremywaguet,

We can look into having the CI compile statically for the Windows build.
The closed source SeaChest is built and released as static builds if you would be ok using those while we look into changes to the CI on this project.

The tools are used in both a standard Windows environment and the preboot/recovery environments. PE/recovery do have some advantages such as less interruption from other software and fewer restrictions on the commands that can be sent. The subprojects are also used by SeaTools for a standard Windows environment as well.

Thanks for bringing this to our attention!

jeremywaguet, 2022-10-10:
I tried the closed source first (available from https://support.seagate.com/seachest/SeaChestUtilities.zip) and could NOT run this in the PE env. Then I used old good DepencyWalker program to catch that closed source SeaChest has dependency on 'VCRUNTIME140.DLL' (vc redist) and so that (as of today) it is no more released as static exe. Hopefully this opensource project can easy be compiled statically.

vonericsen, 2022-10-10:
Hmm...sounds like there was a packaging error.
Just to confirm, you tried the `Windows\Win64` folder in the package?

We also post builds to the Toolbin repo here on github. I was the one who committed them, so there should be a static package here:
https://github.com/Seagate/ToolBin

vonericsen, 2022-10-27:
This should be fixed for the next release we create.
The meson build job was the one creating this package and I found a flag that can be added to tell msvc to statically link and it appears to work as expected.
I check it with a dependency tool (https://github.com/lucasg/Dependencies) and it matched the output from the static-release target in our visual studio projects which I have previously used for WinPE tools. I tried using dependency walker, but it kept hanging on me for some unknown reason, but this other tool worked great.

vonericsen, 2023-03-01:
This has been fixed in https://github.com/Seagate/openSeaChest/releases/tag/v23.03

Please reopen this issue if you have any trouble. I am happy to look into this some more.

closed 2023-03-01 by vonericsen