Seagate/openSeaChest #95: Cannot set PROTECT=0

JustinLoecher, 2023-01-12:
Good afternoon, I am trying to figure out how to disable this setting/set it to 0.  I have tried the commands in openSeaChest, but they just continue to show me the info below.  
 
 
root@truenas[~]# openSeaChest_FormatUnit -d /dev/sg0 --protectionType 0
==========================================================================================
openSeaChest_Format - openSeaChest drive utilities - NVMe Enabled
Copyright (c) 2014-2022 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
openSeaChest_Format Version: 2.2.1-2_2_1 X86_64
Build Date: Sep 26 2022
Today: Wed Jan 11 09:54:00 2023        User: root
==========================================================================================
Usage
=====
         openSeaChest_Format [-d <sg_device>] {arguments} {options}
 
Examples
========
        openSeaChest_Format --scan
        openSeaChest_Format -d /dev/sg? -i
 
Return codes
============
        Generic/Common exit codes
        0 = No Error Found
        1 = Error in command line options
        2 = Invalid Device Handle or Missing Device Handle
        3 = Operation Failure
        4 = Operation not supported
        5 = Operation Aborted
        6 = File Path Not Found
        7 = Cannot Open File
        8 = File Already Exists
        9 = Need Elevated Privileges
        Anything else = unknown error
 
 
Utility Options
===============
        --echoCommandLine
        --enableLegacyUSBPassthrough
        --forceATA
        --forceATADMA   (SATA Only)
        --forceATAPIO   (SATA Only)
        --forceATAUDMA  (SATA Only)
        --forceSCSI
        -h, --help
        --license
        --modelMatch [model Number]
        --onlyFW [firmware revision]
        --onlySeagate
        -q, --quiet
        -v [0-4], --verbose [0 | 1 | 2 | 3 | 4]
        -V, --version
 
Utility Arguments
=================
        -d, --device deviceHandle
        --displayLBA [LBA]
        -F, --scanFlags [option list]
        -i, --deviceInfo
        --poll
        --progress [format | nvmformat | depop | repop]
        -s, --scan
        -S, --Scan
        --SATInfo
        --testUnitReady
        --depopulateMaxLBA [requested MaxLBA]
        --showPhysicalElementStatus
        --showSupportedFormats
 
        SAS Only:
        =========
        --showFormatStatusLog (SAS Only)
 
Data Destructive Commands
=========================
        --pattern [repeat:asciinospaces | random | increment:startValue | file:filename]
        --removePhysicalElement [element #]
        --repopulateElements
        --setSectorSize [new sector size]
 
        SAS Only:
        =========
        --disableCertification
        --disablePrimaryList
        --discardGList
        --disableImmediateResponse
        --formatMaxLBA [ new max LBA ]
        --protectionIntervalExponent [ exponent value ]
        --protectionType [ 0 | 1 | 2 | 3 ]
        --fastFormat [fast format mode] (SAS Only) (SBC4 required)
        --formatUnit [current | new sector size]    (SAS Only)
        --securityInitialize
        --stopOnListError
 
        NVMe Only:
        =========
        --nvmFmtMetadataSet [ xlba | separate ] (NVMe Only)
        --nvmFmtMS [ # of bytes for metadata ]  (NVMe Only)
        --nvmFmtNSID [all | current]    (NVMe Only)
        --nvmFmtPI [ 0 | 1 | 2 | 3 ]    (NVMe Only)
        --nvmFmtPIL [ beginning | end ] (NVMe Only)
        --nvmFmtSecErase [none | user | crypto] (NVMe Only)
        --nvmFormat [current | format # | sector size]  (NVMe Only)
 
 
I have verified with the protection is not disabling.  I have gone ahead and updated the firmware on one of the drives from K003 to K004 just to verify that was not an issue.  
 

root@truenas[~]# time sg_format -v /dev/sg0
    SEAGATE   ST8000NM0065      K004   peripheral_type: disk [0x0]
      PROTECT=1
      << supports protection information>>
      Unit serial number: ZA1BQ0LD0000C843MBXU
      LU name: 5000c500a64050cf
    mode sense(10) cdb: [5a 00 01 00 00 00 00 00 fc 00]
Mode Sense (block descriptor) data, prior to changes:
  Number of blocks=1953506646 [0x74702556]
  Block size=4096 [0x1000]
    read capacity(10) cdb: [25 00 00 00 00 00 00 00 00 00]
Read Capacity (10) results:
   Number of logical blocks=1953506646
   Logical block size=4096 bytes
No changes made. To format use '--format'. To resize use '--resize'
sg_format -v /dev/sg0  0.00s user 0.00s system 10% cpu 0.023 total

vonericsen, 2023-01-12:
Hi @JustinLoecher,

The `--protectionType` option is used with a format command to reformat the drive.
To disable the protection information you would need to run the format command like this:
`openSeaChest_FormatUnit -d <handle> --formatUnit current --protectionType 0 --poll --confirm ...`
This will take many hours to complete as the drive rewrites its formatting.

Before you do that though, the `PROTECT = 1` part of the sg_format output appears to come from a standard inquiry response saying the drive supports protection information, not necessarily that it is enabled. [I reviewed this from the code I found here to find this.](https://github.com/hreinecke/sg3_utils/blob/627b12990c899934dfb34bedd349408884d8f61b/src/sg_format.c#L905) There is not a way to tell the drive to stop reporting this bit since it is only a support bit.

Part of the `-i` output for a drive in openSeaChest will dump what protection types are supported, and if one is enabled, which is enabled for types 1, 2, or 3 that are supported. Ex: `openSeaChest_FormatUnit -d <handle> -i`

You would see something like `Protection Type 1 [Enabled]` under the `Features Supported` section if the drive is currently formatted with type 1 protection. If a given protection is supported, but not enabled you would see `Protection Type 1` to indicate support without it being enabled.
Protection Type 0 is "no protection" which is not shown at all right now in the output. If you think this is confusing and should be part of the output, please let me know and I will add it.

We've recently pushed better examples to the help, but that has not been released yet, but will show similar examples to the one I've provided in here.

JustinLoecher, 2023-01-12:
I did do -i as well, I have posted it below.  So being as enabled is not there next to either of them that would imply it is turned off?  

`Features Supported:
                Protection Type 1
                Protection Type 2
                Application Client Logging
                Self Test
                Automatic Write Reassignment [Enabled]
                Automatic Read Reassignment [Enabled]
                EPC
                Informational Exceptions [Mode 6]
                Translate Address
                Rebuild Assist
                Seagate In Drive Diagnostics (IDD)
                Format Unit
                Sanitize`

JustinLoecher, 2023-01-12:
Thank you so much for the explanation.  That makes complete sense,I just did not know.

vonericsen, 2023-01-12:
@JustinLoecher no problem, happy to help and share knowledge 😄 
If you think this is good and your question is answered, feel free to close the issue, or keep it open if you think you may have more questions related to this.

closed 2023-01-12 by JustinLoecher