Seagate/openSeaChest #96: `GLIBC_2.28' not found on Synology DSM 7.1.1-42962 Update 3

MilesTEG1, 2023-01-13:
Hello,
I'm trying to run this tool on my Synology NAS (DS920+, under DSM 7.1.1-42962 Update 3).
But I get this error:
```bash
root@Syno-DS920Plus:/volume1/homes/Miles/openSeaChest-refs-tags-v22.07-linux-x86_64# ./openSeaChest_FormatUnit --help
./openSeaChest_FormatUnit: /lib64/libc.so.6: version `GLIBC_2.28' not found (required by ./openSeaChest_FormatUnit)
```
And I don't know how to fix this...
There isn't an apt get command on the NAS to update the GLibc version...
My version is:
```bash
ll -a  /lib64/libc.*
lrwxrwxrwx 1 root root 12 Sep  5 17:52 /lib64/libc.so.6 -> libc-2.26.so
```

So, is-ti possible somebody made a version compatible with libc-2.26.so ?
Thanks in advance.
Miles

vonericsen, 2023-01-13:
Hi @MilesTEG1,

Linux binary compatibility is tricky to accomplish, and it's usually due to varying glibc versions between Linux's.
So the build you are using came from the Github action using the latest version of Ubuntu with a newer glibc.

This is something that should be solved with the next release we publish as we've received a pull request (#94) that will build packages on CentOS7 which uses an older version.
I have also pushed some new cross-compile builds that use musl and cut the dependency problem out entirely.

I've attached the latest build artifact from the musl build and "many linux" build if you would like to try either of them and provide some feedback about whether it worked or not on your NAS.
[openSeaChest-develop-linux-x86_64-portable.tar.zip](https://github.com/Seagate/openSeaChest/files/10415794/openSeaChest-develop-linux-x86_64-portable.tar.zip)
Please note that this should be considered an alpha level release at this time since it has not had thorough testing yet.

MilesTEG1, 2023-01-14:
hello @vonericsen 
Both versions you provide in the archive file joined with your message works on my Synology !
Thank you very much.

My goal using this tool is to set 4Kn for my new WD Red Plus 4To harddrive directly into the NAS. Because if I had to use my windows computer, I only have an USB3 dock...
Your last comment:
> Please note that this should be considered an alpha level release at this time since it has not had thorough testing yet.
isn't rassuring 😨 
Is it relatively safe to use it ?

Thanks again for this version 😃

MilesTEG1, 2023-01-14:
I'm not sure I can use the tool...
The `--showSupportedFormats` option on the new drive doesn't show 4Kn possibility on it...

```bash
root@Syno-DS920Plus:/volume1/homes/Miles/openSeaChest-dev/openSeaChest-dev-portable# ./openSeaChest_FormatUnit -d /dev/sg2 --showSupportedFormats
==========================================================================================
 openSeaChest_Format - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2023 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Format Version: 2.6.2-4_0_2 X86_64
 Build Date: Jan 12 2023
 Today: Sat Jan 14 16:30:52 2023        User: root
==========================================================================================

/dev/sg2 - WDC WD40EFPX-68C6CN0 - WD-XXXXXXXXXXXX - 81.00A81 - ATA

WARNING: Customer unique firmware may have specific requirements that
         restrict sector sizes on some products. It may not be possible to format/
         fast format to common sizes like 4K or 512B due to these customer requirements.

Supported Logical Block Sizes and Protection Types:
---------------------------------------------------
  * - current device format
PI Key:
  Y - protection type supported at specified block size
  N - protection type not supported at specified block size
  ? - unable to determine support for protection type at specified block size
Relative performance key:
  N/A - relative performance not available.
  Best
  Better
  Good
  Degraded
--------------------------------------------------------------------------------
 Logical Block Size  PI-0  PI-1  PI-2  PI-3  Relative Performance  Metadata Size
--------------------------------------------------------------------------------
*               512     Y     N     N     N                   N/A            N/A
--------------------------------------------------------------------------------
```

Noamly, as the WD doc says:
[image]
Advanced Format means 4Kn support, right ?

MilesTEG1, 2023-01-18:
Hello @vonericsen 
> If you do not see the `Set Sector Configuration` feature listed, then you may need to contact WD to see if there is a way to change the sector size.
That's it : it doesn't show this feature...
Maybe with WD support, but, it may took too long to have an answer... I already replace my fault drive with this one, so I forget the idea to set 4Kn on this drive.
Thanks for your help 😃

closed 2023-01-23 by vonericsen