Seagate/openSeaChest #97: How to disable FDE (Full DIsk Encryption) ?

Zibri, 2023-01-22:
Any idea?

vonericsen, 2023-01-24:
Hi @Zibri,

There is no way to disable the encryption when data is written to the medium as that part of the firmware or hardware performing the encryption is always active for self encrypting drives (SED).
The different security features may offer ways to disable certain parts of the feature, but that doesn't stop the encryption when writing to the medium (platters/NAND).

If your drive supports TCG Opal, there is a feature where you can disable "SID" which is essentially disabling the ability to set a password on the drive. This disables using any other part of TCG opal unless you perform a revertSP to restore all settings to factory defaults using the PSID from the label.

For TCG Enterprise, this feature doesn't really exist, but there is something similar that can prevent setting a password on the drive, but this may be a Seagate unique feature implemented by Seagate's firmware.

There are also "ISE" or "Instant Secure Erase" drives that have full disk encryption, but do not use passwords or the other security features offered by TCG Opal or TCG Enterprise. As these do not have a password, there is no additional configuration available.

For ATA security, there is the freezelock command which blocks using any part of the feature until the drive power cycles. Some motherboards will issue this automatically on startup, and I believe Windows does this too (unless it's a modern version of Windows PE).
Be aware that drives with ATA security and TCG features have interactions between them. So if you enable ATA security (set a password), that same password can be used with certain TCG opal features. I do not remember all of the interactions between the features though, but ATA security is much more limited in capabilities than what is offered through TCG.
Not all devices with ATA security encrypt the medium though. `openSeaChest_Security -d <handle> --ataSecurityInfo` will read the ATA security info as well as a bit that the drive can set if it uses full disk encryption and report it there.

At this time openSeaChest does not have the capabilities to make these changes to the TCG features, but it is present in the closed source SeaChest tools available from https://www.seagate.com/support/software/seachest/
These options are limited to Seagate drives though.

Zibri, 2023-01-26:
Question: is there a way to SET the KEY directly ... AFAIK it's only possible to change it using erase/sanitize crypto ext.

vonericsen, 2023-01-26:
@Zibri,
I had to ask one of our security engineers about this, but the short answer is you cannot specify a value for the key with anything from the standards. You can use something like the genkey, revert, or revertSP method to request the drive generate a new key, but that is all that you can do. Sanitize crypto erase does this as well.
This doc is about Seagate secure drives, but may be able to provide you more information: https://csrc.nist.gov/CSRC/media/projects/cryptographic-module-validation-program/documents/security-policies/140sp3252.pdf

As for the "handy store" this sounds a lot like the TCG Opal spec's "shadow MBR" which allows putting a piece of software in this location to allow launching and unlocking the drive with a password. It can be used by plugging in a drive and accessing it, or it also allows an OS to write a small bootloader to place here and perform the unlock. Microsoft Bitlocker does this in hardware encryption mode.

Zibri, 2023-01-27:
According to the [drive manual](https://documents.westerndigital.com/content/dam/doc-library/en_us/assets/public/western-digital/product/data-center-drives/ultrastar-dc-hc500-series/product-manual-ultrastar-dc-hc530-sata-oem-spec.pdf)
it also might be some HPA.. I don't know.