Seagate/openSeaChest #98: Well done! openSeaChest just destroyed my 14 TB Western Digital drive

Zibri, 2023-01-22:
This is the command I ran:

openSeaChest_FormatUnit.exe -d PD2 --securityInitialize --setSectorSize 512 --confirm this-may-erase-data

everything went "well" and after power cycling the drive was unresponsive.

Thanks a lot.

Zibri, 2023-01-23:
Update: it turns out it messed up the controller. probably disabled SMART on the drive.
With a different USB/SATA controller the drive works.
How can I fix this mess??
Controller is JMICRON (wd my book) 
I reflashed the controller with generic firmware and the drive works but smart is unresponsive and freezes anything using smart,
please HELP.

vonericsen, 2023-01-24:
Hi @Zibri,

I'm sorry to hear that this happened with your drive. We are discussing internally about what we can do to help resolve this issue.
In the meantime, I did find a few threads mentioning a tool that WD has which can do fast format. They may have a way to help you recover the drive from that tool as well:

- https://www.truenas.com/community/threads/how-do-i-change-a-hard-drives-logical-sector-size-from-512-bytes-to-4096-bytes.103556/page-2
- https://www.truenas.com/community/threads/troubleshooting-disk-format-warnings-in-bluefin.106051/#post-729956
- https://superuser.com/questions/1576894/how-to-convert-the-western-digital-ultrastar-dc-hc530-14tb-hdd-from-512e-to-4
- https://forums.servethehome.com/index.php?threads/please-help-convert-hgst-he10-and-he12-to-4kn.27668/
- https://community.wd.com/t/how-to-format-wdbwlg0040hbk-from-512e-to-4kn/222085/9

The last piece of advice I can give is to reach out to WD's support and see if they can help you with any other recovery method that they may have.

Zibri, 2023-01-24:
After some more checking this is the situation:

the HDD works and now has 4096 bytes per sector.
Reflashed the USB/SATA bridge with a gebneric Jmicron firmware and linux has no problems with the drive but windows does.
What I miss:
1) original firmware for JMS579 and NVRAM
2) I am pretty sure that customized firmware expects something that is not anymore there.

weirdly, with the "generic" JMicron firmware, smartctl works on linux but freezes windows (or any application using smart).

What would be great is some utility from WD to:
1) setup the drive for their modified JMicron firmware to "like"
2) reflash the JMS579 with the original 1031 firmware.

Zibri, 2023-01-24:
About WD, I contacted them 2 days ago. Still no answer.

vonericsen, 2023-01-24:
Hi @Zibri,

Thanks for the update. To make sure I'm helping you as best I can, can you please describe a bit more about what you are trying to accomplish with your drive?
For example, is this supposed to be a USB backup drive that is formatted to 4KN that you are using, or if you are just formatting this drive before it goes into a NAS, or it's a new boot or data drive for some new system, etc. Is it being used on Windows, Linux, FreeBSD, etc, or moved between them, etc.

We do have a warning in the help output that not all USB adapters will work properly with a 4KN sector sized drive, so from your last update that may explain some of the behavior you are seeing. If you connect the drive directly to the motherboard or an HBA, it may work without an issue.  It is strange that it works on Linux, but not on Windows, but I have seen many strange behaviors with USB adapters over the years I've been at Seagate. Usually when I've seen these strange things, I try connecting the drive directly to the motherboard to rule out adapter issues, which may be helpful in further troubleshooting what is happening with the drive right now.

Zibri, 2023-01-24:
Long story:
Setup: WD My Book 14 TB (USB).
OS: windows and linux.

I was studying the WD controller and started reverse engineering commands and so on.
I compiled a full list and everything was working.
I also was trying to find a way to put the controller in a DUMB mode so that it could work with any drive and didn't encrypt/decrypt it on the fly as it used to do.
Everything was fine thus far.

Then I issued a few sanitize commands (while testing) and tried to change the blocksize to 4096 (native).
I succeeded and sh*t started hitting the fan...
I had all sorts of errors.
But by sound I could hear the drive initializing properly, so I was 100% sure it was a software error.
Since where I am I don't have a desktop I had to be creative because my other USB/SATA controller did not support an HDD this big.
So I used a JMICRON flasher program, shorting out pins 1 and 2 of the 25X20 winbond chip and backup up the firmware (without the ram because of the short) 
Then I flashed a generic jmicron firmware, and I found out, as I supposed, that the drive was still working, even if SMART was freezing both windows both linux (like the controller was not answering).
Finally I reformatted the drive with (logical) 512 bytes/block and FINALLY (a few minutes ago) the drive works "perfectly" with the unbranded firmware. SMART works again both in windows both in linux.

Now: the problem is that if I flash back the original FW (version 1031), that firmware expects the NVRAM to be set somehow and I don't have it.
Flashing it without the nvram makes it identify as "1031 25A7" instead of "1031 25ED".
Moreover: if I connect the drive, I would expect to see it as encrypted, instead all I get is an "hardware error" from the controller.
Probably WD firmware expects 2 things:
1) nvram data
2) a sector on the disk probably in the service area of some kind (speculation)

Anyway, with the generic jmicron firmware the drive seems healthy (probably I would get the same result by connecting it in SATA directly to a desktop)

Here is the log:
```
=== START OF INFORMATION SECTION ===
Device Model:     WDC WD140EDGZ-11B1PA0
Serial Number:    ***5**6*
LU WWN Device Id: 5 000cca 299c26d99
Firmware Version: 85.00A85
User Capacity:    14,000,519,643,136 bytes [14.0 TB]
Sector Sizes:     512 bytes logical, 4096 bytes physical
Rotation Rate:    7200 rpm
Form Factor:      3.5 inches
Device is:        Not in smartctl database 7.3/5319
ATA Version is:   ACS-2, ATA8-ACS T13/1699-D revision 4
SATA Version is:  SATA 3.2, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Tue Jan 24 20:48:14 2023 EST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x00) Offline data collection activity
                                        was never started.
                                        Auto Offline Data Collection: Disabled.
Self-test execution status:      (   0) The previous self-test routine completed
                                        without error or no self-test has ever
                                        been run.
Total time to complete Offline
data collection:                (  101) seconds.
Offline data collection
capabilities:                    (0x5b) SMART execute Offline immediate.
                                        Auto Offline data collection on/off support.
                                        Suspend Offline collection upon new
                                        command.
                                        Offline surface scan supported.
                                        Self-test supported.
                                        No Conveyance Self-test supported.
                                        Selective Self-test supported.
SMART capabilities:            (0x0003) Saves SMART data before entering
                                        power-saving mode.
                                        Supports SMART auto save timer.
Error logging capability:        (0x01) Error logging supported.
[... 26 lines cut ...]
194 Temperature_Celsius     0x0002   039   039   000    Old_age   Always       -       40 (Min/Max 23/47)
196 Reallocated_Event_Count 0x0032   100   100   000    Old_age   Always       -       0
197 Current_Pending_Sector  0x0022   100   100   000    Old_age   Always       -       0
198 Offline_Uncorrectable   0x0008   100   100   000    Old_age   Offline      -       0
199 UDMA_CRC_Error_Count    0x000a   100   100   000    Old_age   Always       -       0

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
Num  Test_Description    Status                  Remaining  LifeTime(hours)  LBA_of_first_error
# 1  Short offline       Completed without error       00%        44         -
# 2  Short offline       Completed without error       00%         0         -

SMART Selective self-test log data structure revision number 1
 SPAN  MIN_LBA  MAX_LBA  CURRENT_TEST_STATUS
    1        0        0  Not_testing
    2        0        0  Not_testing
    3        0        0  Not_testing
    4        0        0  Not_testing
    5        0        0  Not_testing
Selective self-test flags (0x0):
  After scanning selected spans, do NOT read-scan remainder of disk.
If Selective self-test is pending on power-up, resume after 0 minute delay.
```

Zibri, 2023-01-24:
By the way the format took some time but it works:

wdckit format /dev/sdb -b 512 --fastformat   # for 512
wdckit format /dev/sdb -b 4096 --fastformat # for 4096

[wdckit 2.15 can be found here](https://files.hddguru.com/download/Software/Western%20Digital/?order=rating&direction=1)

Zibri, 2023-01-26:
@vonericsen there is something you can do for me. The drive is fine but the enclosure is now unbranded.
I need the original firmware (which must be asked to WD).
I need a full flash (probably 256k).
If they won't provide the firmware then I just need that someone who own a My Book (25ED 1031) could physically dump the WINBOND spi flash. (JMS579 from JMICRON not the ASMEDIA version)

vonericsen, 2023-01-27:
Hi @Zibri, 
I'm happy to hear that your drive is working again!

I'm not entirely sure what WD's tool would have done differently than ours, but I'm still reviewing our code to make sure it matches the standards to make sure this is unlikely to happen in the future.

Unfortunately we don't have any of those WD drives or know where to find their firmware, so we don't know how to help you find it. Asking WD would be your best bet, but their tech support may take some time to track that down too.

The only other advice I have for recovering the USB adapter's original firmware would be seeing if someone on a technical forum is able to dump it for you. Maybe even try one of the numerous reddit communities that are full of people tinkering with their devices to see if anyone on there can help you find it.

Zibri, 2023-01-28:
@vonericsen no need to review your code. It was a mix of ignorance (mine) and horrible software (WD) that caused this.
By setting MAX LBA and then doing a secure erase (no matter with which utility) I probably deleted some HPA/DCO zone containing something that the enclosure expects. By reflashing the enclosure I also clearly lost the original NVRAM.
Anyway all this lead me to do a deep study of the JMICRON controller and I found a very stable a new firmware and configured it properly (not like the ones around the net) for use in USB or UASP mode and getting it's name and serial from the drive.
I found out that the JMS579 present in MyBook drives is actually very good and supports very high bandwidths even with SSDs...
Now I am just curious if someone could dump the W25X20 present in a recent myboot 1031 25ED.
And if possible to dump the HPA/DCO of their drive. Those both things will allow me to understand what went wrong and how to repair things if anyone needs.
I am going to publish a full report on my findings.

vonericsen, 2023-03-01:
We have released https://github.com/Seagate/openSeaChest/releases/tag/v23.03 which now contains a lot more warnings about `--setSectorSize` to help warn about potential for issues when it is used and originally led to this issue being created.
I have also implemented a better automatic recovery for Seagate drives using a vendor unique command, but cannot do anything else to assist with issues or recovery of WD's products since I do not have documentation on what I can do for them.

I'm closing this for now, but please feel free to reopen if you think it is necessary or create new issues if you run into anything else that I may be able to assist you with!

closed 2023-03-01 by vonericsen

Zibri, 2023-03-17:
My WD drive is made by HITACHI.. if that helps.
Still dead in the water unless someone can find a way to factory reset it.
It's this "family" https://documents.westerndigital.com/content/dam/doc-library/en_us/assets/public/western-digital/product/data-center-drives/ultrastar-dc-hc500-series/product-manual-ultrastar-dc-hc530-sata-oem-spec.pdf
Branded as WD140EDGZ