Seagate/openSeaChest #99: openSeaChest_Firmware.exe - Crash

aureyoboss, 2023-01-26:
The “openSeaChest_Firmware.exe” application crashes systematically when I Scan (--scan) :

```
Détecteur d'erreurs 1813004394593752413, type 4
Nom d’événement : APPCRASH
Réponse : Non disponible
ID de CAB : 0

Signature du problème : 
P1 : openSeaChest_Firmware.exe
P2 : 0.0.0.0
P3 : 62e06bfd
P4 : openSeaChest_Firmware.exe
P5 : 0.0.0.0
P6 : 62e06bfd
P7 : c0000005
P8 : 000000000003e354
P9 : 
P10 : 
```

```
Nom de l’application défaillante openSeaChest_Firmware.exe, version : 0.0.0.0, horodatage : 0x62e06bfd
Nom du module défaillant : openSeaChest_Firmware.exe, version : 0.0.0.0, horodatage : 0x62e06bfd
Code d’exception : 0xc0000005
Décalage d’erreur : 0x000000000003e354
ID du processus défaillant : 0x0x914
Heure de début de l’application défaillante : 0x0x1D93170485C924B
Chemin d’accès de l’application défaillante : D:\openSeaChest_Firmware.exe
Chemin d’accès du module défaillant: D:\openSeaChest_Firmware.exe
ID de rapport : 22106732-383b-406e-9c08-24e43ce6364a
Nom complet du package défaillant : 
ID de l’application relative au package défaillant : 
```

Same problem with last version of "SeaChest_Firmware.exe" :

```
Détecteur d'erreurs 2008509029050638161, type 4
Nom d’événement : APPCRASH
Réponse : Non disponible
ID de CAB : 0

Signature du problème : 
P1 : SeaChest_Firmware.exe
P2 : 3.1.4.0
P3 : 62e01ae0
P4 : SeaChest_Firmware.exe
P5 : 3.1.4.0
P6 : 62e01ae0
P7 : c0000005
P8 : 00000000000161dd
P9 : 
P10 : 
```

```
Nom de l’application défaillante SeaChest_Firmware.exe, version : 3.1.4.0, horodatage : 0x62e01ae0
Nom du module défaillant : SeaChest_Firmware.exe, version : 3.1.4.0, horodatage : 0x62e01ae0
Code d’exception : 0xc0000005
Décalage d’erreur : 0x00000000000161dd
ID du processus défaillant : 0x0x125C
Heure de début de l’application défaillante : 0x0x1D9316F7BE15918
Chemin d’accès de l’application défaillante : D:\SeaChest_Firmware.exe
Chemin d’accès du module défaillant: D:\SeaChest_Firmware.exe
ID de rapport : 139f511e-6a6a-44de-a335-d522ca57ba3c
Nom complet du package défaillant : 
ID de l’application relative au package défaillant : 
```

My config :
Windows 11 (up to date) ;
Seagate FireCuda 530 ZP500GM30023 (Intel RST VMD Controller 467F) ;
Use CMD with admin rights.

Thanks.

vonericsen, 2023-01-26:
Hi @aureyoboss,
We may have already fixed this issue from some internal reports I've gotten that are very similar to this, but I have not been able to repeat it myself so I am unable to confirm. I did use various debugging tools to track down a few potential causes and fix those issues.
Can you try this attached build from the CI's latest develop branch build and let me know if it corrects this crash on your system when you try the scan?
[openSeaChest-develop-win-x64_13_01_2023.zip](https://github.com/Seagate/openSeaChest/files/10513290/openSeaChest-develop-win-x64_13_01_2023.zip)

aureyoboss, 2023-01-27:
Hi @vonericsen,
Thank you very much for the quick response.
However, I still have the same problem: the application crashes after the main message...

```
Détecteur d'erreurs 1622080093318134267, type 4
Nom d’événement : APPCRASH
Réponse : Non disponible
ID de CAB : 0

Signature du problème : 
P1 : openSeaChest_Firmware.exe
P2 : 0.0.0.0
P3 : 63c1ede9
P4 : openSeaChest_Firmware.exe
P5 : 0.0.0.0
P6 : 63c1ede9
P7 : c0000005
P8 : 00000000000403d4
P9 : 
P10 : 
```

```
Nom de l’application défaillante openSeaChest_Firmware.exe, version : 0.0.0.0, horodatage : 0x63c1ede9
Nom du module défaillant : openSeaChest_Firmware.exe, version : 0.0.0.0, horodatage : 0x63c1ede9
Code d’exception : 0xc0000005
Décalage d’erreur : 0x00000000000403d4
ID du processus défaillant : 0x0x2A80
Heure de début de l’application défaillante : 0x0x1D93224525AC95F
Chemin d’accès de l’application défaillante : D:\FireCuda530-SSD-NVMe-SU6SM003\command line tools\SeaChest\Windows\Win64\openSeaChest_Firmware.exe
Chemin d’accès du module défaillant: D:\FireCuda530-SSD-NVMe-SU6SM003\command line tools\SeaChest\Windows\Win64\openSeaChest_Firmware.exe
ID de rapport : 998b021d-6ed7-4f26-aaf2-1774d2b945e7
Nom complet du package défaillant : 
ID de l’application relative au package défaillant : 
```
[image]

vonericsen, 2023-01-27:
@aureyoboss,
Darn, I guess I have not found this bug already.

Can you tell me a bit more information about your system?

How many storage drives do you have attached? Include any NVMe, SATA, SAS, USB, RAID, flash drives, SD cards, etc.

Can you share a screenshot of the device manager under the disk drives section and the storage controllers section?

Here is an example from one of my systems:
[image]

Some of the issues I had previously fixed would only show up with multiple devices which is why this may help me track this down a bit more.

If you manually specify the device for something like the `-i` option, does that work?
Ex: `openSeaChest_Firmware.exe -d PD0 -i`
Windows starts these handles at zero and counts up for each device in the system, so the next drive would most likely be `PD1`, then `PD2` and so on. Please let me know if this works too, or if this also causes a crash like this one.

Once I get this info I will share a debug build that will output a lot more information that it will help me track this down even further.

aureyoboss, 2023-01-27:
@vonericsen,
Hello,

Here is the screenshot of my devices:
[image]

The following `openSeaChest_Firmware.exe -d PD0 -i` command crashes...

On the other hand, by launching the following command line : `openSeaChest_Firmware.exe -d PD1 -i`. I see my second hard drive (WDC WD10EZEX-00BN5A0).

vonericsen, 2023-01-27:
@aureyoboss,
Interesting that it crashes for PD0, but not PD1. That makes me think it's an issue specific to NVMe...that should help me figure out where to look a bit closer.
I will work on a build to generate debug messages that can help me figure out where the crash is. I will share it here once I think it has everything I might need to find this bug.

vonericsen, 2023-01-28:
@aureyoboss,

As I was debugging my own system and testing my additional debug output, I did find another place that could cause a crash and pushed a commit to fix that.
I've incorporated it into this debug version to see if that fix helps or if there is still another bug somewhere else.

When you run this version it will output a lot of information to the screen as it scans and opens devices that could help me determine where the crash happened.
Can you save this output to a file and share it with me?
I would like to see output from the following commands and I've included the redirects to a file to keep it easier too:

1. `openSeaChest_Firmware -s > normalScan.txt`
2. `openSeaChest_Firmware -s -v4 > verboseScan.txt`
3. `openSeaChest_Firmware -d PD0 -i > devInfo.txt`
4. `openSeaChest_Firmware -d PD0 -i -v4 > verboseInfo.txt`

[OpenSeaChest_Firmware_Debug_1_27_01_2023.zip](https://github.com/Seagate/openSeaChest/files/10524023/OpenSeaChest_Firmware_Debug_1_27_01_2023.zip)

While I'm hoping the issue I found is the real cause, but the additional output from these commands can help me be sure I've found all the places where this kind of bug might be coming from.
Thanks for your help and patience as I'm debugging this!

aureyoboss, 2023-01-28:
@vonericsen,

Unable to download file... False positive? 
[image]

aureyoboss, 2023-01-28:
@vonericsen,

I just disabled Windows antivirus.

Here are the results. Hope they will help you.
Thank you again for your work!

[normalScan.txt](https://github.com/Seagate/openSeaChest/files/10526690/normalScan.txt)
[verboseInfo.txt](https://github.com/Seagate/openSeaChest/files/10526691/verboseInfo.txt)
[devInfo.txt](https://github.com/Seagate/openSeaChest/files/10526692/devInfo.txt)
[verboseScan.txt](https://github.com/Seagate/openSeaChest/files/10526693/verboseScan.txt)

vonericsen, 2023-01-28:
@aureyoboss,

We have seen false positives reported by some anti-virus software for some of our tools in the past and this would be a false positive...I built it from the source code and ran it myself so it should be free of anything sinister 😃 

Thanks for the logs! That helped me track down what is most likely the cause of the crash.
Here is one more debug version to test just to make sure there isn't any other issues lurking.
It will output the same debug messages as the last one, but hopefully it will complete those operations without crashing this time.
If it crashes again, can you run the same commands and share the logs again? If it does not crash then this issue should be solved and we will need to release a new build soon.

[OpenSeaChest_Firmware_Debug_2_28_01_2023.zip](https://github.com/Seagate/openSeaChest/files/10528365/OpenSeaChest_Firmware_Debug_2_28_01_2023.zip)

aureyoboss, 2023-01-29:
@vonericsen,

You're great! It no longer crashes. Thank you very much for this teamwork (even if I didn't do much).

So, will I be able to update my Seagate FireCuda 530 ZP500GM30023 (7VP00X08) firmware from version SU6SM001 to SU6SM003 without risk?

If I understand correctly, here are the commands to use:
```
1. openSeaChest_Firmware -d PD0 --downloadFW FireCuda530_SU6SM003.bin
2. openSeaChest_Firmware -d PD0 --activateFW
```

Is it advisable to backup the hard drive before doing this?

Have a good day.

vonericsen, 2023-01-30:
@aureyoboss,

Thanks for the logs! It is super important to us to fix issues that cause crashes like this as it is not always easy to duplicate them, so the logs are super helpful when we cannot duplicate it. I suspect I was unable to duplicate it because while I have similar hardware, I do not have the exact same setup so it wasn't going down this path in the code on my system like it does for your system.

At this point, a firmware update should be possible for you with this debug version (you will just have some extra output on the screen when it runs right now).

Those are the correct commands to perform the update. You can combine the options into one single command line run if you want as well:
`openSeaChest_Firmware -d <handle> --downloadFW myfile.bin --activateFW`
This will send the firmware to the drive and activate it all in one go  if you prefer to do it all at once.

> Is it advisable to backup the hard drive before doing this? Can these commands erase data from the disk?

I would always advise a backup of your data just to be safe. Whether it is a firmware update for a drive, a firmware update for a motherboard, or some other significant change like enabling or disabling bitlocker (as an example). 
I've worked on firmware update tools for Seagate for many years and firmware updates will not touch user data unless there is documentation or other warnings provided with the update that mention this. Updates that touch user-data are extremely rare. I only see that kind of update on engineering samples. 
There are qualification processes in place to detect issues or user data changes and to make sure the update works in as many configurations as our labs are able to test. 
Sometimes weird or unexplained things happen though, so it is best to have a backup in case something were to go wrong during the update (power loss at the worst possible time for example).

For what it is worth, I recently ran this exact update during a test in the last week or two and did not see any issues regarding the data stored on the drive.

aureyoboss, 2023-02-01:
Thanks very much @vonericsen !

closed 2023-02-01 by aureyoboss