Seagate/openSeaChest #100: Device wildcard results in invalid argument errors

Jolly-Pirate, 2023-01-31:
Running the command `openSeaChest_PowerControl` for example, it cites a few examples such as `openSeaChest_PowerControl -d /dev/sg? --showEPCSettings`, however, running any command with a wildcard doesn't work properly, it only shows an output for the `/dev/sg0` device, but not the rest. The same problem happens with the SeaChest tools from Seagate's website.
```
$ sudo openSeaChest_PowerControl -d /dev/sg? --showEPCSettings
==========================================================================================
 openSeaChest_PowerControl - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2022 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_PowerControl Version: 3.1.10-3_2_1 X86_64
 Build Date: Jul 26 2022
 Today: Tue Jan 31 10:43:21 2023        User: root
==========================================================================================
Invalid argument: /dev/sg1
Invalid argument: /dev/sg2
Invalid argument: /dev/sg3
Invalid argument: /dev/sg4
Invalid argument: /dev/sg5

/dev/sg0 - M4-CT064M4SSD2 - 00000000111802001090 - 070H - ATA
Showing EPC Settings not supported on this device.
```

vonericsen, 2023-01-31:
Hi @Jolly-Pirate,

If you want to run a command on all detected devices use `-d all` instead. 
So doing `openSeaChest_PowerControl -d all --showEPCSettings` will do what you are trying right now.

The `?` is meant to be a "fill in your number" representation. If you think there is a better way to put this in the help that is more intuitive, please let me know and I will make that change.

Jolly-Pirate, 2023-01-31:
Oh, the `?` seemed misleading to me, it looked like a wildcard. Usually parameters like that are expressed as `<id>` for example. Perhaps the documentation should be corrected like so.
`openSeaChest_PowerControl -d /dev/sg<id> --showEPCSettings`

vonericsen, 2023-01-31:
@Jolly-Pirate,

Thanks for the feedback!
I've pushed an update for the missing `-d all` help information. I will look at the `<id>` idea next.
I'll leave this open until I've finished that work and pushed an update for that as well.

vonericsen, 2023-03-01:
Release https://github.com/Seagate/openSeaChest/releases/tag/v23.03 contains the help updates I mentioned in my last comment.
I'm leaving this open for now to look at the <id> suggestion as I think we will work on that for the next release of openSeaChest.

vonericsen, 2023-03-27:
I implemented the `<id>` as `<#>` in the handle examples since this seemed to make the most sense for now since the handles being used today/are expected (or recommended by openSeaChest in the case of Linux sd vs sg) all use numbers.

This is included in [v23.03.1](https://github.com/Seagate/openSeaChest/releases/tag/v23.03.1) that was just released.
I'm closing this since I believe all issues have been addressed, but please feel free to reopen this if I need to look at anything else or create a new issue as you see fit.

closed 2023-03-27 by vonericsen