Seagate/openSeaChest #101: ST4000DX001 how to 4kn ?

infr-automation, 2023-02-10:
Drive Model:                                                                    Seagate ST4000DX001-1CE168
Drive Firmware Revision:                                                        CC44

SeaChest_Lite --device /dev/sg... --setSectorSize 4096 --confirm ... 

didn't work. Not supported. ?

b-mc, 2023-02-10:
Have u checked what formats the drive supports?
`openSeachest_FormatUnit -d /dev/sgX --showSupportedFormats`

For example:
```
--------------------------------------------------------------------------------
 Logical Block Size  PI-0  PI-1  PI-2  PI-3  Relative Performance  Metadata Size
--------------------------------------------------------------------------------
                512     Y     N     N     N                   N/A            N/A
*              4096     Y     N     N     N                   N/A            N/A
--------------------------------------------------------------------------------
```

infr-automation, 2023-02-10:
Shows only 512 with this check.

vonericsen, 2023-03-01:
Hi @infr-automation,
I'm sorry I missed this issue when you created it.
Not all products support the ability to change sector sizes. The firmware must support this capability to use that option.
@b-mc's suggestion is a good way to check whether this is supported on SATA drives or not.

Another way is to check the output of `-i` in the `Features Supported` section of the output.
For SATA drives it will say "Set Sector Configuration"
For SAS drives it will say "Fast Format"

You can also check my answer on this discussion for more details: https://github.com/Seagate/ToolBin/discussions/18

Please feel free to reopen this issue if you need more information on this or feel free to create a new issue for anything else you run into. We will do our best to help you!

closed 2023-03-01 by vonericsen

infr-automation, 2023-03-03:
My research to 4kn leads me to believe it will be possible to do conversion from 512e to 4kn models by sending custom code by service tools or by let's say the more robust LSI controllers that can actually pass the command.

I haven't seen a 4kn sshd in  like this Sgt PDFs so it's possible the function to work in 4kn is not available on the firmware.

However what if the firmware is replaced with a 4kn firmware, all metadata lost, and the 4kn "resurfacing" is initated then to recreate metadata to sync the PCB with the platters?

Then it should work?

vonericsen, 2023-03-03:
@infr-automation,
Seagate drives do not allow switching firmware like this to change the sector size.