Seagate/openSeaChest #102: Sanitize Overwrite Started - Device Removed - Sanitize Overwrite Aborted - How to proceed

FurkanGozukara, 2023-02-27:
Hello. I am using SeaTools 5.1.176 Windows version

[image]

I have purchased 3x ST8000VX0022 so I wanted to test them before start actually using them

So I have started Sanitize Overwrite on all disks

After like 2%, first disk got removed from the computer entirely. Like shown below

Then other 2 disks are also got removed like at 10%

[image]

I still didn't restart or closed the computer. How should I proceed? All 3 brand new disks are now invisible. Nothing displayed in their ports as shown below

[image]

So here my questions

What happens if I shut down PC and open back?

How can I see the status? It has been 8 hours since Sanitize Overwrite started

If I restart, does it resume wherever it left? 

Thank you

vonericsen, 2023-02-27:
Hi @FurkanGozukara,

The sanitize feature will resume after power loss where it left off. So if the device was unplugged from power, it will resume sanitizing itself as soon as it gets power again based on the last checkpoint the firmware made during the erase as soon as it spins up again. Same thing for any reset on the bus during this time, it will continue sanitizing where it left off until the sanitize has completely erased the entire drive and all previously reallocated user-data sectors. There is no way to stop a sanitize once it has been started; no power removal, reset, or command can stop it once it has been started. The drive must complete the erase or hit a failure condition, but I have not seen a failure up to this point on any drive I have tested.

Erasing an HDD takes about 1.5-2 hours per terabyte. So a rough estimate would be approximately 16 hours to erase these drives.
The best thing to do is just leave them powered on, even if they show disconnected in other software or even the BIOS as the erase will eventually finish and return them to a "normal" state again that most host software is familiar with.
I have even tested this by starting it and removing both SATA power and data cables, then connecting it only to power on my desk and letting it sit for the day, then bringing it back to the original system I started the erase on and it was ready for use after the erase finished overnight.

If you wanted to try openSeaChest_Erase, you might be able to get a progress indication out of them, but it will depend on what the OS/driver/HBA has done (if it removed the handle associated with the device).
These utilities are CLI and must be run through an Admin mode powershell or cmd.exe window in Windows or with root/sudo capabilities in Linux/FreeBSD/etc.
If you do `openSeaChest_Erase -s` and the drives are listed in that output, you can do `openSeaChest_Erase -d <handle> --progress sanitize` and that will report the drive's progress indication for sanitize.

If these drives "disappeared" on their own while SeaTools was running, this could be because the driver, the OS, or the HBA it was connected to while it was sanitizing is trying other management operations or monitoring of its own and does not know how to handle the state the drive enters during a sanitize. This state is a limited set of commands that can return responses at this time while it is erasing. It could be counting on reading something that is now aborting the command instead of a normal response, so it thinks something has gone wrong and stopped attempting to communicate with the drive.

I have found that sometimes under Windows, a device may disappear because something did not understand how to handle it, but if you boot up a Linux on the same system it may have a better ability to continue communicating with it in these other states. So if you really need a progress indicator, maybe try the same commands I mentioned earlier under Linux. Even a live-boot of something like Ubuntu would be sufficient for this.

FurkanGozukara, 2023-02-27:
@vonericsen  excellent answer

unfortunately command didnt list the drives probably I need a restart

[image]

May I ask another question?

The disks have this label on them

[image]

However their model and firmware were displayed like below on seatools and crystaldiskmark info

What do you think about this?

[image]

[image]

vonericsen, 2023-02-27:
@FurkanGozukara,
I'm investigating what might cause this and need to reach out to some others inside Seagate. Thank you for this question. I will update when I have something else to share.

vonericsen, 2023-02-27:
@FurkanGozukara,

After discussion with some internal sources, it appears that the pictured drive you shared is not a valid Seagate product.
We also verified this with the code on verify.seagate.com with the information from the label.

Can you share where you purchased these drives so Seagate can continue to investigate how this happened?

We will do our best to continue supporting you with this tool, but since we cannot verify that this is an authentic product, we cannot verify the drive's health and cannot guarantee its ability to store data either.

However, Seagate is extremely interested that you get closure on this. Please keep us in the loop on this post with anything you find or any concerns you have.

vonericsen, 2023-02-27:
Hi @FurkanGozukara,

Thank you for sharing that information! I have passed it along to the correct people inside Seagate.

The best way to verify a drive is authentic is through the website verify.seagate.com and follow the instructions on there.
This uses either scanning a QR code from the label or entering some text from the label (right below the QR code and above the verify.seagate.com text) to perform the verification.

I learned that you also reached out over email. I have passed that email address you used on to one of my contacts in Seagate to reach out to you directly as well to help some more with the verification.

FurkanGozukara, 2023-02-28:
@vonericsen it has been 22 hours. I did a computer shut down and open again

1 disk is returned back to the available disks

but 2 disks are still displayed like this what should I do?

should i clear meta data? or something else?

[image]

[image]

vonericsen, 2023-02-28:
Hi @FurkanGozukara,

My first question is, were these drives configured as part of a RAID when you first started all of this?
The metadata clearing I would imagine may help tell the system to stop trying to find these drives for a RAID.
When a RAID is created, the controller writes some metadata to each disk to help track which one is part of which RAID and which stripe it is holding for the RAID, etc.
Sanitizing would remove all of this, but the controller may have stored that info somewhere else too. 
If it was in a RAID and you delete the RAID from the controller that might help.

> this is the info of the available disk and I see a lot of Number Of Reallocated Logical Sectors that means this disk is gone?

Not necessarily dead, but it is also not a good sign that this drive is healthy and I would say this is more evidence that this drive is not as new as the listing led you to believe.

I would try to return these drives to try and get new ones that are authentic.

FurkanGozukara, 2023-02-28:
> Hi @FurkanGozukara,
>
> My first question is, were these drives configured as part of a RAID when you first started all of this? The metadata clearing I would imagine may help tell the system to stop [...]

yes i will return back but first i need to return them back to normal

clear metadata fails

i didn't set them raid 

the disks are not visible in administrative tools

what can I do? 

I tried to make raid on bios and it failed too

[image]

FurkanGozukara, 2023-02-28:
ok i did disable raid mode from motherboard

i see that sanitize is still going on after 48 hours wth :/

[image]

[image]

FurkanGozukara, 2023-03-01:
over 9 hours sanitize is almost never progressing :/

[image]

vonericsen, 2023-03-01:
Hi @FurkanGozukara,

Sorry that this is taking so long. These drives are clearly not new or working well if they are taking this much extra time. A brand new out of box drive would not take anywhere near this long to complete this overwrite.
These drives are likely encountering a lot of internal errors while performing the overwrite so it is doing a lot of retries to complete the overwrite. Each time it has to retry it causes additional delay.

I have seen in the past that requests for progress that are too frequent from software can interrupt error recovery when the drive recalculates the progress, then it has to restart that error recovery process again. That may be what is happening and causing some additional delays. I can look into some changes that may help with alleviate this issue for future updates to SeaTools and to SeaChest/openSeaChest as well.

As it seems leaving SeaTools open is causing some additional delay, I would leave it closed and check for progress about once every 30 minutes as this will cause a lot less for the drive to process and let the firmware do what it needs to without as many interruptions.
If you want to further reduce delays you could unplug the SATA interface cable and leave only the power cable connected to the drive. It will continue erasing without interruption at this point and may be even faster. Windows may send some occasional commands that may delay it, but probably not as significant as what you have been seeing so far.

FurkanGozukara, 2023-03-01:
@vonericsen ty so much for answers

I would like to ask few more questions if you reply I would appreciate very much

All disks are normal now

Could you evaluate their smart attributes and tell me how well they are looking for?

One of the disk has low health I will send it back. For those 2 others, the company may cause me problems since they look normal. so could you check values and tell me are they looking healthy?

For example what is  Read Error Rate and how to evaluate it?

it is displayed as current 66 worst 64 threshold 44 raw 00000036C2D1

what is Seek Error Rate rate ?

current 61 worst 60 threshold 45 raw 00000013225F

what is Hardware ECC recovered?

current 9 worst 8 threshold 0 raw 00000036C2D1

[image]

[image]

[image]

# results of --smartAttributes analyzed

```
C:\Users\King\Downloads\openSeaChest-refs-tags-v22.07-win-x64>openSeaChest_Info -d pd2 --smartAttributes analyzed
==========================================================================================
 openSeaChest_Info - openSeaChest drive utilities - NVMe Enabled
 Copyright (c) 2014-2022 Seagate Technology LLC and/or its Affiliates, All Rights Reserved
 openSeaChest_Info Version: 2.1.2-3_2_1 X86_64
 Build Date: Jul 26 2022
 Today: Thu Mar  2 02:42:59 2023        User: King (admin)
==========================================================================================

\\.\PhysicalDrive2 - ST8000NM0105 - ZA13HAR5 - G00A - ATA
1 - Read Error Rate
        Attribute Type(s):
                Pre-fail/warranty. Indicates a cause of known impending failure.
                Online Data Collection. Updates as the drive runs.
                Performance. Degredation of this attribute will affect performance.
                Error Rate. Attribute tracks and error rate.
        Current (Nominal) Value: 66
        Worst Ever Value:        64
        Threshold:               44
        Number Of Sector Reads: 3588817
        Number Of Read Errors: 0
3 - Spin Up Time
        Attribute Type(s):
                Pre-fail/warranty. Indicates a cause of known impending failure.
                Online Data Collection. Updates as the drive runs.
        Current (Nominal) Value: 93
        Worst Ever Value:        90
        Threshold set to always passing
4 - Start/Stop Count
        Attribute Type(s):
                Online Data Collection. Updates as the drive runs.
                Event Count. Attribute is a counter.
                Self-Preserving. Saves between power cycles.
        Current (Nominal) Value: 100
        Worst Ever Value:        100
        Threshold:               20
        Spin Up Count: 19
5 - Retired Sectors Count
        Attribute Type(s):
[... 1098 lines cut ...]

---General Errors Statistics---
 Number Of Reported Uncorrectable Errors                      N/A              0
 Number Of Resets Between Command Acceptance and Completion   N/A              0

---Temperature Statistics---
 Current Temperature                                          N/A              36 C
 Average Short Term Temperature                               N/A              Invalid
 Average Long Term Temperature                                N/A              Invalid
 Highest Temperature                                          N/A              49 C
 Lowest Temperature                                           N/A              29 C
 Highest Average Short Term Temperature                       N/A              Invalid
 Lowest Average Short Term Temperature                        N/A              Invalid
 Highest Average Long Term Temperature                        N/A              Invalid
 Lowest Average Long Term Temperature                         N/A              Invalid
 Time In Over Temperature                                     N/A              0 minutes
 Specified Maximum Operating Temperature                      N/A              60 C
 Time In Under Temperature                                    N/A              0 minutes
 Specified Minimum Operating Temperature                      N/A              5 C

---Transport Statistics---
 Number Of Hardware Resets                                    N/A              19
 Number Of ASR Events                                         N/A              1
 Number Of Interface CRC Errors                               N/A              0
```

vonericsen, 2023-03-02:
Hi @FurkanGozukara,

> Could you evaluate their smart attributes and tell me how well they are looking for?

I cannot really assist with analysis of drive data like this, especially since we have already determined these drives are not authentic Seagate products. I'm also not a failure analysis engineer so I cannot do much as far as analysis goes as that is more specialized than the knowledge that I have.
A couple easy things you can do are SMART check and a short diagnostic self-test (DST)  on the drives. These are pretty straight forward in terms of saying "it is good" or "it is bad".
ex: `openSeaChest_SMART -d handle --smartCheck --shortDST --poll`

My full response for the discussion you started (#103) will go into this a bit more once it has been reviewed by our failure analysis team for accuracy.

If you run into trouble returning these to where you bought them, you could try Seagate's official support to see if they are able to help you further.

Alex000Lexx, 2024-03-29:
> Hi @FurkanGozukara,
>
> The sanitize feature will resume after power loss where it left off. So if the device was unplugged from power, it will resume sanitizing itself as soon as it gets power again [...]

Thank you!!!!! you are a lifesaver!!! I had lost my mind, as I have been trying to understand why I was experiencing what looked like disk corruption / detection issues on relatively new IronWolf Pro drives. I ended up leveraging SeaChest (after failing to get SeaTools Linux to stop at a command prompt). Anyhow, I managed to get one of the drives to recognize on a system where it's running in legacy ata mode and once SeaTools X ran, I tried a variety of commands and then tried Sanitize in the Advanced menu and voila! the drive indicated it was at 14.87%.

Now that I know that these things will just continue to Sanitize without input from me, I can get some much needed sleep! I had the indication that things would continue, but knowing the drives simply need to remain powered on, and they will continue on their merry way is a huge relief, I seriously thought I had bricked 2 IronWolf Pro 16tb's and 2 18Tbs... 

Thanks for this information, it's amazing how much nonsense is on the internet!! now I know what this looks like.

Cheers!

vonericsen, 2024-03-29:
@Alex000Lexx,

I'm glad to hear this worked for you as well!
I'm working on some other updates to sanitize currently, and I was just thinking about adding a message to openSeaChest to say something like 

>If Sanitize appears to be taking much longer than expected, the system or HBA may not be handling the sanitize in progress state correctly. If this happens the next best thing to do is leave the drive powered on with the interface cable (SATA/SAS HBA cable) disconnected. Sanitize will continue running while the drive is powered on and the system will no longer be able to interrupt/slow down the sanitize operation.

If you think this message is good or could use additional information, let me know and I'll work on adding it in.

Alex000Lexx, 2024-03-29:
> @Alex000Lexx,
>
> I'm glad to hear this worked for you as well! I'm working on some other updates to sanitize currently, and I was just thinking about adding a message to openSeaChest to say something [...]

Hi vonericsen, 

I think it's a fantastic idea, it would make a big difference to anyone else attempting to use the sanitize command. What I had originally attempted was to use --poll to see if I could keep an eye on things but that didn't quite work out.

One other user lens here is that I was surprised that once I ran the sanitize command, I was able to start targeting more drives. I originally had the perspective that the command was a very quick wipe (duh on my part) thus ran it on 4 drives, thinking that when I came back to the shell prompt I was done (LOL).

I think a real nice to have would be in addition to the message you are indicating above is to say... "Note - Sanitize may take 1-2 hours per TB (then you read the drive size and give a notional estimate), and while I do appreciate these aren't newbie tools, a "are you sure?" along with the extra warning of time would take this tool out of the park. 

Thanks for the reply and for asking for feedback! I can't wait for these drives to return to normal state.

Cheers.