Seagate/openSeaChest #107: openSeaChest_PowerControl.exe --standby 1200000 instant spindown

dabugen, 2023-03-11:
Hello,

thanks for the great software, finally some alternative to the old stuff like HDPARM which also supports EPC. So I´ve been trying a few things with openSeaChest_PowerControl.exe on Windows Server 2022. Almost everything seems to work fine, but as soon as I use the "--standby" to specify a 20 minutes sleep/standby/spin-down time for my old (non-EPC) hard drives (full command line, in this case, is "openSeaChest_PowerControl.exe -d all --standby 1200000"), all drives spin-down instantly directly afterward. After spinning them up again by provoking disc activity in Windows, they then again spin down after 20 minutes, so at least this seems to be set all correct. However, why would it spin them down instantly after issuing the command? This happens everytime I use the --standby flag with whatever value and doesn´t happen with any of the EPC idle modes nor standby_y or standby_z. That seems to be a bug to me as HDPARM (which also sets the legacy standby time) doesn´t show this behavior.

Thanks for looking into this :-)

dabugen, 2023-03-12:
I´ve tested it extensively today and it´s pretty clear, each time the "--standby" switch is used with a timer value, the drive is put to instant standby after it´s been set, regardless of the timer value. Clearly a bug in my opinion.

vonericsen, 2023-03-13:
@dabugen,

Thanks for reporting this.

One funky thing about the old ATA standby timer is it can be set by either a standby command or a idle command according to the spec.
The openSeaChest code is using the standby command to do this right now, but I will investigate switching to the idle command instead, or at least something to detect the drive's state and choose the command based on that...like if already in standby change using standby, otherwise use the idle command to set it.

The reason the standby command is used to set this timer is because I used the SAT (SCSI to ATA translation) spec as a reference which uses the standby command on non-EPC drives to configure this timer. There may be a reason for the spec to have chosen the standby command instead of idle; like some drives did not set the timer properly from idle or something else. I will see if I can find out more and whether there was some other issue the spec committee was working around, at the very least so I can be aware of some other funky behavior that could happen using idle instead.

vonericsen, 2023-03-13:
@dabugen,

I have pushed a fix for this on our opensea-operations submodule. 
After checking with the spec committee representatives there doesn't seem to be a case for not using idle to set the timer on old drives.
So now the behavior will be: take the timer value, convert it with the SAT spec formula, then if not in standby, use the idle command to set the timer (which will keep the drive spinning, but may cause it to move to idle from active). If it is in standby already, it will set the timer using the standby command so it will not cause the drive to move to a spinning state either.
There is no way to keep the drive in "active" mode with these old timers, so idle is the closest I can get with this, but will prevent the drive from spinning up and down immediately when using this option.

dabugen, 2023-03-14:
Very interesting read, thanks for the explanation and fix. I´ll test it as soon as there is a release with the commit for Windows. Thanks again for taking the time to look into this, explain it, and fix it, it´s much appreciated!

vonericsen, 2023-03-14:
Hmm, I can click the green checkmark from my commit to get to the CI and download the artifact (but it is not the most intuitive), but IDK if others can do this or not....

So I downloaded it and am attaching the Windows build here so you can try it out. Let me know if you run into any issues with it 😃 
[openSeaChest-develop-win-x64.zip](https://github.com/Seagate/openSeaChest/files/10974691/openSeaChest-develop-win-x64.zip)

I do not have a release timeline yet since we just did one recently, but if we do an update (v23.03.01 for example) I would include this commit into that build. I figure it is best to share this artifact for now until we decide when to do a new build.

dabugen, 2023-03-17:
Thanks a lot, that´s appreciated! I´ll test it soon :-)

vonericsen, 2023-03-27:
I have included this change as part of [v23.03.1](https://github.com/Seagate/openSeaChest/releases/tag/v23.03.1).

I'm closing this for now, but please feel free to reopen this issue if you need me to look at this some more. Also feel free to create a new issue for anything else you find.

closed 2023-03-27 by vonericsen

dabugen, 2023-03-29:
Perfect, thanks a lot, it works without issues now.